espressif/esp-adf #232: ESP DevKit V1 Board with Speech Recognition

thuyducati, 2019-08-13:
Hello everyone.

Is it possible to use ESP DevKit V1 Board to recognize voice???
(ESP DevKit V1 Board: https://docs.zerynth.com/latest/official/board.zerynth.doit_esp32/docs/index.html)

I am tried to use make -j4 with my ESP DevKit V1 Board to recognize voice and I have the error like this:

$ make -j4
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from C:/msys32/home/MyASUS/esp/esp-idf/requirements.txt are satisfied.
App "play_mp3" version: v2.0-beta1-31-g76ef618-dirty
CC build/adf_utils/json_utils.o
CC build/app_trace/heap_trace_tohost.o
Python requirements from C:/msys32/home/MyASUS/esp/esp-idf/requirements.txt are satisfied.
CC build/app_trace/app_trace.o
CC build/adf_utils/cloud_services/aws_sig_v4_signing.o
App "play_mp3" version: v2.0-beta1-31-g76ef618-dirty
CC build/app_update/esp_app_desc.o
CC build/app_trace/host_file_io.o
CC build/adf_utils/cloud_services/baidu_access_token.o
CC build/app_update/esp_ota_ops.o
CC build/app_trace/app_trace_util.o
AR build/adf_utils/libadf_utils.a
CXX build/asio/asio/asio/src/asio.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_update/libapp_update.a
CC build/audio_board/lyrat_v4_3/board_pins_config.o
AR build/app_trace/libapp_trace.a
CC build/audio_hal/audio_hal.o
In file included from C:/msys32/home/MyASUS/esp/esp-adf/components/audio_pipeline/include/audio_event_iface.h:30,
                 from C:/msys32/home/MyASUS/esp/esp-adf/components/esp_peripherals/include/esp_peripherals.h:29,
                 from C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board.h:31,
                 from C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_pins_config.c:28:
C:/msys32/home/MyASUS/esp/esp-idf/components/esp32/include/rom/queue.h:1:2: warning: #warning rom/queue.h is deprecated, please use sys/queue.h instead [-Wcpp]
 #warning rom/queue.h is deprecated, please use sys/queue.h instead
  ^~~~~~~
In file included from C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board.h:29,
                 from C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_pins_config.c:28:
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'get_input_set_id':
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_def.h:33:35: error: 'TOUCH_PAD_NUM9' undeclared (first use in this function)
 #define BUTTON_SET_ID             TOUCH_PAD_NUM9
                                   ^~~~~~~~~~~~~~
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_pins_config.c:164:12: note: in expansion of macro 'BUTTON_SET_ID'
     return BUTTON_SET_ID;
            ^~~~~~~~~~~~~
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_def.h:33:35: note: each undeclared identifier is reported only once for each function it appears in
 #define BUTTON_SET_ID             TOUCH_PAD_NUM9
                                   ^~~~~~~~~~~~~~
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_pins_config.c:164:12: note: in expansion of macro 'BUTTON_SET_ID'
     return BUTTON_SET_ID;
            ^~~~~~~~~~~~~
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'get_input_play_id':
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_def.h:34:35: error: 'TOUCH_PAD_NUM8' undeclared (first use in this function)
 #define BUTTON_PLAY_ID            TOUCH_PAD_NUM8
                                   ^~~~~~~~~~~~~~
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_pins_config.c:169:12: note: in expansion of macro 'BUTTON_PLAY_ID'
     return BUTTON_PLAY_ID;
            ^~~~~~~~~~~~~~
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'get_input_volup_id':
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_def.h:35:35: error: 'TOUCH_PAD_NUM7' undeclared (first use in this function)
 #define BUTTON_VOLUP_ID           TOUCH_PAD_NUM7
                                   ^~~~~~~~~~~~~~
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_pins_config.c:174:12: note: in expansion of macro 'BUTTON_VOLUP_ID'
     return BUTTON_VOLUP_ID;
            ^~~~~~~~~~~~~~~
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_pins_config.c: In function 'get_input_voldown_id':
C:/msys32/home/MyASUS/esp/esp-adf/components/audio_board/lyrat_v4_3/board_def.h:36:35: error: 'TOUCH_PAD_NUM4' undeclared (first use in this function)
 #define BUTTON_VOLDOWN_ID         TOUCH_PAD_NUM4
[... 101 lines cut ...]
AR build/bootloader/efuse/libefuse.a
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/soc/src/soc_include_legacy_warn.o
CC build/bootloader/soc/src/memory_layout_utils.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/esp32/flash_encrypt.o
CC build/bootloader/soc/src/hal/spi_hal.o
CC build/bootloader/soc/src/hal/spi_hal_iram.o
CC build/bootloader/bootloader_support/src/esp32/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp32/secure_boot.o
CC build/bootloader/soc/src/hal/spi_flash_hal.o
CC build/bootloader/soc/src/hal/spi_flash_hal_iram.o
CC build/bootloader/bootloader_support/src/esp32/secure_boot_signatures.o
CC build/bootloader/soc/src/hal/spi_slave_hal.o
CC build/bootloader/soc/src/hal/spi_slave_hal_iram.o
AR build/bootloader/bootloader_support/libbootloader_support.a
AR build/bootloader/soc/libsoc.a
LD build/bootloader/bootloader.elf
esptool.py v2.7-dev

I really appreciate any help.
Thank you so much!

jason-mao, 2019-08-13:
@thuyducati  
> Is it possible to use ESP DevKit V1 Board connect with Sound Sensor LM393 to recognize voice?

It's theoretically possible.

Could you provide your toolchain version and IDF version when you got your build error?

thuyducati, 2019-08-13:
My toolchain version is esp32-2019r1
and 
My IDF version is v4.0-dev-1443-g39f090a4f

closed 2019-11-03 by thuyducati