espressif/esp-adf #233: esp-adf errors due to missing files

Patsjemoe, 2019-08-13:
I restarted completely with attached document setup of toolchain.
It refers to creating a shortcut that launches cmd and runs export.bat.
If yuu look into file missingexportbat, wich is a cmd screen, it shows that this IDF installation that I followed , does not include the export.bat file. Also other files are missing.
How to solve this ?
When compiling play_mp3, I receive errors, showing missing files (see attached files). How to solve this ?

[CMakeError.log](https://github.com/espressif/esp-adf/files/3496159/CMakeError.log)
[CMakeOutput.log](https://github.com/espressif/esp-adf/files/3496160/CMakeOutput.log)
[missingexportbat.pdf](https://github.com/espressif/esp-adf/files/3496163/missingexportbat.pdf)
[Standard Setup of Toolchain for Windows.pdf](https://github.com/espressif/esp-adf/files/3496164/Standard.Setup.of.Toolchain.for.Windows.pdf)

Regarsds
Ludo

jason-mao, 2019-08-13:
@Patsjemoe Could you help provide build log ?

Patsjemoe, 2019-08-13:
Hi,
with pleasure, but can you tell me where I can find this/which file in
which directory ?
regards
ludo

jason-mao, 2019-08-13:
When you running `idf.py build`, there are lot of characters print on your terminal, please copy all the characters to file and send me. Please run `idf.py clean` before building.

Patsjemoe, 2019-08-13:
Hi,
attached, requested file
best regards
Ludo

Patsjemoe, 2019-08-16:
using cmd with toolchain 2.0 istalled , and using esp-idf v3.2.2 seems to create missing files....have done new install, and only using MINGW32 = OK.

closed 2019-08-16 by Patsjemoe