espressif/esp-adf #234: esp-adf-arduino when will be relesed?

MichaelDu9226, 2019-08-13:
I am wonder esp-adf-arduino when will be relesed. Does the development team have a schedule on it?Thanks.

timotoots, 2019-08-13:
Yes, would be great development!

don41382, 2019-12-29:
Any update on this? Are there any plans?

MichaelDu9226, 2019-12-29:
我注意到ESP-ADF已经有micro python的库了，想问一下Arduino的库是不是也快了

jason-mao, 2019-12-30:
Hi guys, I'm sorry to inform you that the ADF-Arduino will be delayed. We will postpone the first version of arduino  to the first quarter of 2020.

timotoots, 2020-01-02:
Thank you @jason-mao! Could you share some features planned for the first version?

RealCorebb, 2020-02-18:
Just new to be here,i am glad to hear that.

robertpoll, 2020-02-24:
hi @jason-mao Would be great if you had an update on this as we're now quite a way through 1Q20. I'm looking to start some audio development soon so an update would be really helpful, and if you have an idea of release date that would be great.

timotoots, 2020-06-04:
Hello, @jason-mao. Is there an update about release schedule?

Releece, 2020-12-04:
Hi @jason-mao 
We are now one year past your decision to postpone the Arduino-ADF, any information you can give us?
Is this still scheduled to be released at all?

jason-mao, 2021-02-19:
Hi guys, I am very sorry for late to reply . Due to a task adjustment, we have suspended this feature. But the good news is that  it will be restarted this year. We will update the progress on time. Thank you.

MichaelDu9226, 2021-09-10:
@jason-mao Hi Json Any update?
We just want to play wav or MP3 based on auduo codec like ES8388 in arduino platform.

Releece, 2022-11-04:
Please let us know if the development of the feature has been restarted or if the plan is still to restart it in the future?
The ESP32 module is capable of amazing things with the ESP-ADF. I have installed many examples and it gave me a big smile to see this functionality appear for the ESP32, something I never imagined.

The problem is the IDF & ADF code is very very hard to learn and really above my level. It's just not attractive to learn a code with so any archaic sounding names.

I think you have a whole market opportunity for the hundreds of thousands of thinkerers, if you release the Arduino-ADF. Especially for home-assistant integration, it would be fantastic to control my home by voice command, and many others will go for that I'm sure.

Please drop us a message, even if it's negative, or you have no idea, we would surely appreciate it! And thanks for developping such a great chip and all the work you do around it.

Best regards,
Jori

breakstring, 2023-07-27:
4 years....
╮(╯▽╰)╭

Releece, 2023-07-29:
Maybe we should just try learning the ESP-IDF & ADF, and share code (examples) collectively :p It's so long ago I don't even really remember how awfull the language is. I don't even understand anymore, they're IDF's, so did they invent a whole new language for it as well, or did they just write a whole bunch of functions and parameters that had such off-puting names I stopped attempting to understand it after a few weeks, after having built and tried all their pre-provided example projects, and now misremember it as a language because it's parameters/functions were so long they made up 90% of the characters :p

I stil have the ESP32 Lyra-T (IIRC the name) lying around here, it's certainly the most exciting board I've ever seen. Would be nice if I could put it to actual use! Back then voice-assistant feature was only available for Baidu, but Google-assistent was next on their roadmap. Can you imagine running that on an ESP32? :D

vegasje, 2024-07-30:
Is this still a planned feature?