espressif/esp-adf #235: pipeline_bt_source examples unable to connect to Bluetooth.

uched41, 2019-08-14:
I have been trying to tweak the [pipeline_bt_source example](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_bt_source) with no success. The ESP32 recognizes the Sink devices and initializes connection but fails to connect.

### Steps to reproduce
- Clone current esp-adf master.
- Copy pipeline_bt_source example.
- Build and flash.

### Environment
Ubuntu 18.04
ESP32 devKitC v1

warwound, 2019-08-28:
I had an identical result when trying the pipeline_bt_source example earlier this afternoon.
Trying to pair my LyraT v4.3 to a Google Mini and to a Philips BT25.
The serial output showed that the LyraT detected each bluetooth speaker (which was in pairing mode), fetched the speaker info but then disconnected.

I just tried the example again so i could capture the serial output and post it here.
With the same Google Mini the example worked first time - my Google Mini played the streaming mp3 albeit not at all smoothly - I shall debug that tomorrow.
The example did, however, fail to connect to my Philips BT25 speaker and here's the serial output:

> I (52057) BLUETOOTH_SERVICE: Discovery started.
> I (56177) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (56177) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (56177) BLUETOOTH_SERVICE: --RSSI: -55
> I (56717) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (56717) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (56717) BLUETOOTH_SERVICE: --RSSI: -66
> I (57177) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (57177) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (57177) BLUETOOTH_SERVICE: --RSSI: -54
> I (61287) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (61287) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (61287) BLUETOOTH_SERVICE: --RSSI: -53
> I (64877) BLUETOOTH_SERVICE: Device discovery failed, continue to discover...
> I (69527) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (69527) BLUETOOTH_SERVICE: Device discovery failed, continue to discover...
> I (69537) BLUETOOTH_SERVICE: Discovery started.
> I (74087) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (74087) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (74087) BLUETOOTH_SERVICE: --RSSI: -51
> I (74177) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (74177) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (74177) BLUETOOTH_SERVICE: --RSSI: -57
> I (74447) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (74447) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (74447) BLUETOOTH_SERVICE: --RSSI: -55
> I (75007) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (75007) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (75007) BLUETOOTH_SERVICE: --RSSI: -54
> I (75277) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (75277) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (75277) BLUETOOTH_SERVICE: --RSSI: -58
> I (75757) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (75767) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (75767) BLUETOOTH_SERVICE: --RSSI: -57
> I (76397) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (76407) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (76407) BLUETOOTH_SERVICE: --RSSI: -60
> I (77047) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (77047) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (77047) BLUETOOTH_SERVICE: --RSSI: -65
> I (82187) BLUETOOTH_SERVICE: Scanned device: fc:58:fa:4e:0c:3c
> I (82187) BLUETOOTH_SERVICE: --Class of Device: 0x260404
> I (82187) BLUETOOTH_SERVICE: --RSSI: -43
> I (82337) BLUETOOTH_SERVICE: Device discovery failed, continue to discover...

I'm running Windows 10 Pro.

HTH.

esp-xiaodong, 2019-08-29:
Hi @uched41 I try this demo( pipeline_bt_source example ). I can connect to the sink successfully. Can u give me your complete log?

uched41, 2019-08-29:
Here is the full log: 
```MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6204
load:0x40078000,len:10180
load:0x40080400,len:6660
entry 0x40080764
I (29) boot: ESP-IDF v3.2-103-g4aa1058e8 2nd stage bootloader
I (29) boot: compile time 17:47:37
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 001e8480
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3cedc (249564) map
I (181) esp_image: segment 1: paddr=0x0004cf04 vaddr=0x3ffbdb60 size=0x0310c ( 12556) load
I (186) esp_image: segment 2: paddr=0x00050018 vaddr=0x400d0018 size=0xf22fc (991996) map
0x400d0018: _flash_cache_start at ??:?

I (535) esp_image: segment 3: paddr=0x0014231c vaddr=0x3ffc0c6c size=0x00288 (   648) load
I (535) esp_image: segment 4: paddr=0x001425ac vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/ched/esp-adf/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1779

I (542) esp_image: segment 5: paddr=0x001429b4 vaddr=0x40080400 size=0x163b8 ( 91064) load
I (601) boot: Loaded app from partition at offset 0x10000
I (601) boot: Disabling RNG early entropy source...
I (602) cpu_start: Pro cpu up.
I (605) cpu_start: Starting app cpu, entry point is 0x400811f0
0x400811f0: call_start_cpu1 at /home/ched/esp-adf/esp-adf/esp-idf/components/esp32/cpu_start.c:246

I (0) cpu_start: App cpu up.
I (616) heap_init: Initializing. RAM available for dynamic allocation:
I (623) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (629) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (635) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (641) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (647) heap_init: At 3FFC95D8 len 00016A28 (90 KiB): DRAM
I (653) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (659) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (666) heap_init: At 400967B8 len 00009848 (38 KiB): IRAM
I (672) cpu_start: Pro cpu start user code
I (19) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (67) BLUETOOTH_SOURCE_EXAMPLE: [ 1 ] Create Bluetooth service
I (67) BTDM_INIT: BT controller compile version [74c9f57]
I (77) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (157) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 0
I (747) BLUETOOTH_SERVICE: Starting device discovery...
I (747) BLUETOOTH_SOURCE_EXAMPLE: [1.1] Get Bluetooth stream
I (747) BLUETOOTH_SOURCE_EXAMPLE: [ 2 ] Create http stream to read data
I (757) BLUETOOTH_SOURCE_EXAMPLE: [ 3 ] Create mp3 decoder to decode mp3 file
I (767) STAGEFRIGHTMP3_DECODER: MP3 init
[... 116 lines cut ...]
0x400d28be: esp_vApplicationIdleHook at /home/ched/esp-adf/esp-adf/esp-idf/components/esp32/freertos_hooks.c:86

0x4008e3ed: prvIdleTask at /home/ched/esp-adf/esp-adf/esp-idf/components/freertos/tasks.c:3537

0x4008d15d: vPortTaskWrapper at /home/ched/esp-adf/esp-adf/esp-idf/components/freertos/port.c:403
```

I edited the bluetooth_service source file to accept the first Bluetooth speaker it detects by uncommenting the string compare .

```
/* search for device named "peer_bdname" in its extended inquiry response */
    if (eir) {
        get_name_from_eir(eir, (uint8_t *)&peer_bdname, NULL);
        /*if (strcmp((char *)peer_bdname, (char *)&g_bt_service->peer_bdname) != 0) {
            return;
        }*/

        ESP_LOGI(TAG, "Found a target device, address %s, name %s", bda_str, (uint8_t *)peer_bdname);
        g_bt_service->source_a2d_state = BT_SOURCE_STATE_DISCOVERED;
        memcpy(&g_bt_service->remote_bda, param->disc_res.bda, ESP_BD_ADDR_LEN);
        ESP_LOGI(TAG, "Cancel device discovery ...");
        esp_bt_gap_cancel_discovery();
    }
```
The bluetooth speaker is detected successfully and attempts to connect but fails after sometime.

mohnazemi, 2019-09-17:
I am facing the same exact issue with pipeline_bt_source example connecting to a Bluetooth speaker using LyraT v4.3 board. I am building the code using the IDF version 3.3 release branch and the master branch of ADF. The connection is established then it is dropped and fails. To me it seems like a bug in the IDF bluetooth handling the Bluedroid stack business but I have not yet been able to get an ideas on this as I am just a newbie starting with this platform.

Here is the full log from boot to fail:

```
I (29) boot: ESP-IDF v3.3 2nd stage bootloader
I (29) boot: compile time 00:52:17
I (31) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 40MHz
I (37) boot: SPI Mode       : DIO
I (41) boot: SPI Flash Size : 4MB
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 001e8480
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x49c78 (302200) map
I (198) esp_image: segment 1: paddr=0x00059ca0 vaddr=0x3ffbdb60 size=0x036f0 ( 14064) load
I (204) esp_image: segment 2: paddr=0x0005d398 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/xx/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (205) esp_image: segment 3: paddr=0x0005d7a0 vaddr=0x40080400 size=0x02870 ( 10352) load
I (218) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0x1125b8 (1123768) map
0x400d0018: _flash_cache_start at ??:?

I (617) esp_image: segment 5: paddr=0x001725d8 vaddr=0x40082c70 size=0x191fc (102908) load
0x40082c70: psram_gpio_config at /Users/xx/esp/esp-idf/components/esp32/spiram_psram.c:574 (discriminator 1)

I (676) boot: Loaded app from partition at offset 0x10000
I (676) boot: Disabling RNG early entropy source...
I (677) psram: This chip is ESP32-D0WD
I (682) spiram: Found 32MBit SPI RAM device
I (686) spiram: SPI RAM mode: flash 40m sram 40m
I (691) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (698) cpu_start: Pro cpu up.
I (702) cpu_start: Application information:
I (707) cpu_start: Project name:     bt_source_demo
I (712) cpu_start: App version:      1
I (717) cpu_start: Compile time:     Sep 17 2019 00:52:16
I (723) cpu_start: ELF file SHA256:  2f4ddb079ff99291...
I (729) cpu_start: ESP-IDF:          v3.3
I (733) cpu_start: Starting app cpu, entry point is 0x40081470
0x40081470: call_start_cpu1 at /Users/xx/esp/esp-idf/components/esp32/cpu_start.c:270

I (0) cpu_start: App cpu up.
I (1634) spiram: SPI SRAM memory test OK
I (1635) heap_init: Initializing. RAM available for dynamic allocation:
I (1635) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1641) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1647) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1653) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1659) heap_init: At 3FFC9DD0 len 00016230 (88 KiB): DRAM
I (1666) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1672) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1679) heap_init: At 4009BE6C len 00004194 (16 KiB): IRAM
I (1685) cpu_start: Pro cpu start user code
I (1690) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (145) cpu_start: Chip Revision: 1
W (146) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.
I (149) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (160) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (200) BLUETOOTH_SOURCE_EXAMPLE: [ 1 ] Create Bluetooth service
I (200) BTDM_INIT: BT controller compile version [e736f10]
I (200) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (300) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (660) BLUETOOTH_SERVICE: Starting device discovery...
I (660) BLUETOOTH_SOURCE_EXAMPLE: [1.1] Get Bluetooth stream
I (660) BLUETOOTH_SOURCE_EXAMPLE: [ 2 ] Create http stream to read data
[... 44 lines cut ...]
I (3200) BLUETOOTH_SOURCE_EXAMPLE: [5.2] Start Bluetooth peripheral
I (3210) BLUETOOTH_SOURCE_EXAMPLE: [ 6 ] Set up  event listener
I (3220) BLUETOOTH_SOURCE_EXAMPLE: [6.1] Listening event from all elements of pipeline
I (3220) BLUETOOTH_SOURCE_EXAMPLE: [6.2] Listening event from peripherals
I (3230) BLUETOOTH_SOURCE_EXAMPLE: [ 7 ] Start audio_pipeline
I (3240) AUDIO_ELEMENT: [http] Element task created
I (3240) AUDIO_ELEMENT: [mp3] Element task created
I (3250) AUDIO_ELEMENT: [bt] Element task created
I (3250) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:331, MEM Total:4212900 Bytes, Inter:119660 Bytes, Dram:102908 Bytes

I (3270) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (3280) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (3280) STAGEFRIGHTMP3_DECODER: MP3 opened
I (3280) AUDIO_PIPELINE: Pipeline started
I (3290) BLUETOOTH_SOURCE_EXAMPLE: [ 8 ] Listen for all pipeline events
I (12020) HTTP_STREAM: total_bytes=2994349
I (12040) STAGEFRIGHTMP3_DECODER: I2S setup: sample_rate = 44100 : channels = 2
I (12070) BLUETOOTH_SOURCE_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
W (38440) BT_SDP: SDP - Rcvd conn cnf with error: 0x8  CID 0x40

W (38440) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2

I (38440) BLUETOOTH_SERVICE: bt_a2d_source_cb state 4, evt 0x0

```

esp-xiaodong, 2019-09-17:
@mohnazemi  The master ADF should match  IDF3.2， not IDF3.3.

esp-xiaodong, 2019-09-17:
@uched41  Maybe you needn't revoke the code 。
        /*if (strcmp((char *)peer_bdname, (char *)&g_bt_service->peer_bdname) != 0) {
            return;
        }*/
The first device you found may not accept your connection. Just use the ADF master branch code, adjust the CONFIG_BT_REMOTE_NAME in menuconfig to the device you want to connect to.

mohnazemi, 2019-09-17:
@esp-xiaodong Many thanks for your help. I checked out IDF v3.2 but it was giving build errors with current ADF master branch however it was ok when I instead checked out IDF v3.2.2 and now the example started working on my Speaker. The issue is that the audio playback is very choppy and intermittent as it starts/stops every second or so.

jponko, 2019-09-26:
I'm unable to get the pipeline_bt_source to work as on a esp32-lyraT v4.3 board using the latest adf and idf v3.2.2. The console out put shows the following error:

E (135) BLUETOOTH_SERVICE: /home/jponko/esp/esp-adf/components/bluetooth_service/bluetooth_service.c:438 (bluetooth_service_start): enable controller failed
 
The same error occurs when compiling the pipeline_bt_sink example. I opened an issue with that example four days ago so the complete console log can be found there.

Jerry Ponko
AC9NM

mohnazemi, 2019-09-27:
And I am still having problems with playback and connection too. I cannot get it to connect to a Bose headset I have. When I get it connected to a speaker, the playback is intermittent.

esp-xiaodong, 2019-10-08:
> I'm unable to get the pipeline_bt_source to work as on a esp32-lyraT v4.3 board using the latest adf and idf v3.2.2. The console out put shows the following error:
>
> E (135) BLUETOOTH_SERVICE: [...]

@jponko  Please check the menuconfig. You need enable controller br/edr only.

esp-xiaodong, 2019-10-08:
> And I am still having problems with playback and connection too. I cannot get it to connect to a Bose headset I have. When I get it connected to a speaker, the playback is intermittent.

@mohnazemi Could you give me your log?

mohnazemi, 2019-10-08:
Here is my log, I managed to connect to a BOSE headset but the MP3 playback is intermittent (starts stops every second)

```
rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6236
load:0x40078000,len:10496
load:0x40080400,len:6660
entry 0x40080768
I (29) boot: ESP-IDF v3.2.3 2nd stage bootloader
I (29) boot: compile time 23:51:49
I (31) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 001e8480
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x434a8 (275624) map
I (189) esp_image: segment 1: paddr=0x000534d0 vaddr=0x3ffbdb60 size=0x03634 ( 13876) load
I (195) esp_image: segment 2: paddr=0x00056b0c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/mo/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (196) esp_image: segment 3: paddr=0x00056f14 vaddr=0x40080400 size=0x090fc ( 37116) load
I (220) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0x10a648 (1091144) map
0x400d0018: _flash_cache_start at ??:?

I (603) esp_image: segment 5: paddr=0x0016a668 vaddr=0x400894fc size=0x12968 ( 76136) load
0x400894fc: spi_flash_read at /Users/mo/esp/esp-idf/components/spi_flash/flash_ops.c:564

I (651) boot: Loaded app from partition at offset 0x10000
I (651) boot: Disabling RNG early entropy source...
I (652) psram: This chip is ESP32-D0WD
I (657) spiram: Found 32MBit SPI RAM device
I (661) spiram: SPI RAM mode: flash 40m sram 40m
I (666) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (673) cpu_start: Pro cpu up.
I (677) cpu_start: Starting app cpu, entry point is 0x40081390
0x40081390: call_start_cpu1 at /Users/mo/esp/esp-idf/components/esp32/cpu_start.c:250

I (0) cpu_start: App cpu up.
I (1578) spiram: SPI SRAM memory test OK
I (1579) heap_init: Initializing. RAM available for dynamic allocation:
I (1579) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1585) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1591) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1597) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1604) heap_init: At 3FFC9A88 len 00016578 (89 KiB): DRAM
I (1610) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1616) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1623) heap_init: At 4009BE64 len 0000419C (16 KiB): IRAM
I (1629) cpu_start: Pro cpu start user code
I (1634) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (89) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (91) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (131) BLUETOOTH_SOURCE_EXAMPLE: [ 1 ] Create Bluetooth service
I (131) BTDM_INIT: BT controller compile version [acab313]
I (141) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[... 55 lines cut ...]
I (2701) AUDIO_ELEMENT: [bt] Element task created
I (2701) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:331, MEM Total:4214204 Bytes, Inter:120720 Bytes, Dram:103960 Bytes

I (2721) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (2731) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (2731) STAGEFRIGHTMP3_DECODER: MP3 opened
I (2731) AUDIO_PIPELINE: Pipeline started
I (2741) BLUETOOTH_SOURCE_EXAMPLE: [ 8 ] Listen for all pipeline events
E (2791) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (2971) BT_APPL: new conn_srvc id:18, app_id:0
I (2971) BLUETOOTH_SERVICE: bt_a2d_source_cb state 4, evt 0x0
I (2971) BLUETOOTH_SERVICE: a2dp connected
I (2981) BLUETOOTH_SERVICE: a2dp media ready checking ...
I (2991) BLUETOOTH_SERVICE: bt_a2d_source_cb state 5, evt 0x3
I (2991) BLUETOOTH_SERVICE: a2dp media ready, starting ...
W (3001) BT_APPL: new conn_srvc id:18, app_id:1
I (3061) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x824b

I (3061) BLUETOOTH_SERVICE: bt_a2d_source_cb state 5, evt 0x3
I (3071) BLUETOOTH_SERVICE: a2dp media start successfully.
I (3071) BLUETOOTH_SERVICE: bt_a2d_source_cb state 5, evt 0x1
I (7741) HTTP_STREAM: total_bytes=2994349
I (7751) STAGEFRIGHTMP3_DECODER: I2S setup: sample_rate = 44100 : channels = 2
I (8251) BLUETOOTH_SOURCE_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
```

esp-xiaodong, 2019-10-09:
> Found a target device, address

@mohnazemi  
Your log seems ok, and i couldn't find any errors that may cause intermittent from it. You can check your wifi signal intensity and use other a2dp sink device to test it. When i use my speaker or esp32 a2dp sink to test, they all work well. Unfortunately, I don't have bose devices to test.

jponko, 2019-10-14:
I was able to get the pipeline_bt_source/sink examples to work with a pair
of esp32-lyraT boards with
Component config-->Bluetooth-->Bluetooth controller-->Bluetooth controller
mode:
Set to : *BR/EDR Only*

However, the sound is choppy because WiFi power saving is enabled in the
source example.
    periph_wifi_wait_for_connected(wifi_handle, portMAX_DELAY); // after
this line (line 89), add line the following line to disable WiFi power
saving
   * ESP_ERROR_CHECK (esp_wifi_set_ps(WIFI_PS_NONE)); // No WiFi power
saving*

My questions are:
1) Why does the Bluetooth controller mode need to be set to BR/EDR?
2) Why is WiFi power saving mode on by default. From my experience, the
ESP32's WiFi signal is fairly weak as is when power saving is off and
having it on by default doesn't make sense.
3) Will these changes make it to the github examples?

esp-xiaodong, 2019-10-15:
> I was able to get the pipeline_bt_source/sink examples to work with a pair of esp32-lyraT boards with Component config-->Bluetooth-->Bluetooth controller-->Bluetooth controller mode: Set to : [...]

1.a2dp is part of classical bt profile, not ble. So you need enable br/edr
2.power saving mode is initialized defaultly, but not started defaultly.
3.br/edr controller configuration will be pushed to github

mohnazemi, 2019-10-15:
I tried a different Bluetooth headset. The discovery can hardly find it. When it gets connected the connection fails again:

```
 (7630) event: sta ip: 192.168.1.163, mask: 255.255.255.0, gw: 192.168.1.1
I (7630) wifi: Set ps type: 0

I (7630) BLUETOOTH_SOURCE_EXAMPLE: [5.1] Create Bluetooth peripheral
I (7630) BLUETOOTH_SOURCE_EXAMPLE: [5.2] Start Bluetooth peripheral
I (7640) BLUETOOTH_SOURCE_EXAMPLE: [ 6 ] Set up  event listener
I (7650) BLUETOOTH_SOURCE_EXAMPLE: [6.1] Listening event from all elements of pipeline
I (7660) BLUETOOTH_SOURCE_EXAMPLE: [6.2] Listening event from peripherals
I (7660) BLUETOOTH_SOURCE_EXAMPLE: [ 7 ] Start audio_pipeline
I (7670) AUDIO_ELEMENT: [http] Element task created
I (7680) AUDIO_ELEMENT: [mp3] Element task created
I (7680) AUDIO_ELEMENT: [bt] Element task created
I (7690) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:331, MEM Total:4212448 Bytes, Inter:122428 Bytes, Dram:105668 Bytes

I (7700) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (7710) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (7710) STAGEFRIGHTMP3_DECODER: MP3 opened
I (7720) AUDIO_PIPELINE: Pipeline started
I (7720) BLUETOOTH_SOURCE_EXAMPLE: [ 8 ] Listen for all pipeline events
I (13430) BLUETOOTH_SERVICE: Device discovery failed, continue to discover...
I (13430) BLUETOOTH_SERVICE: Device discovery failed, continue to discover...
I (13440) BLUETOOTH_SERVICE: Discovery started.
I (13750) BLUETOOTH_SERVICE: Scanned device: 88:c6:26:de:3e:c4
I (13750) BLUETOOTH_SERVICE: --Class of Device: 0x240418
I (13750) BLUETOOTH_SERVICE: --RSSI: -52
I (13750) BLUETOOTH_SERVICE: --Name: Jaybird X3
I (13760) BLUETOOTH_SERVICE: Found a target device, address 88:c6:26:de:3e:c4, name Jaybird X3
I (13770) BLUETOOTH_SERVICE: Cancel device discovery ...
I (13770) BLUETOOTH_SERVICE: Device discovery stopped.
I (13780) BLUETOOTH_SERVICE: a2dp connecting to peer: Jaybird X3
E (13790) BT_APPL: reset flags
I (13790) BLUETOOTH_SERVICE: bt_a2d_source_cb state 4, evt 0x0
W (18930) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x40

W (18930) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2

I (18930) BLUETOOTH_SERVICE: bt_a2d_source_cb state 4, evt 0x0
I (37060) HTTP_STREAM: total_bytes=2994349
I (37070) STAGEFRIGHTMP3_DECODER: I2S setup: sample_rate = 44100 : channels = 2
I (37090) BLUETOOTH_SOURCE_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
```

maciekczwa, 2019-10-26:
I had many problems with chopped audio.

I tried putting file inside spiffs - worked OK.
Then I tried some web radio mp3 url - perfect audio. Looks like the url specified in example is the problem.

mohnazemi, 2019-10-27:
I gave up on this example and and went on to write an example myself with Bluetooth and MP3 file on the SD card instead of the wireless. It works perfectly fine. I also have to add that the issues with bluetooth connection I encountered _may_ have something to do interference between Bluetooth and WIFI, same antenna and same frequency at 2.5GHz

drewandre, 2020-03-17:
Any updates on this? I'm seeing the same issue but our application is dual mode BT classic/BLE so enabling controller br/edr only is not an option

fc92, 2020-06-01:
still quite unstable with ESP-IDF 4.1, last stable version

wtatarski, 2020-06-01:
I can confirm that this issue still exists. I also have problem with chopped audio after connecting to bluetooth speaker or headphones. Will it be fixed somewhen? Maybe this is a bug in WiFi and Bluetooth coexistence?

NikolayTach, 2020-07-11:
[image] remarkably thought out that it has the domino is not voltage in its inputs.

sdandade1, 2020-11-05:
> I gave up on this example and went on to write an example myself with Bluetooth and MP3 file on the SD card instead of the wireless. It works perfectly fine. I also have to add that the issues with Bluetooth connection I encountered _may_ be due to interference between Bluetooth and WIFI, same antenna and same frequency at 2.5GHz

Can you please share you code please?

mohnazemi, 2020-11-05:
> Can you please share you code please?

Sure.  Here it is: https://gist.github.com/mohnazemi/b7912c1750aa39d4d5d82df782903470

sdandade1, 2020-11-05:
> Sure. Here it is: https://gist.github.com/mohnazemi/b7912c1750aa39d4d5d82df782903470

Thank you very much @mohnazemi