espressif/esp-adf #236: AUDIO_NULL_CHECK error

Patsjemoe, 2019-08-19:
when making flash, get following error : (I did not change anything in the play_mp3 file).

C:/msys32/home/user/esp/esp-adf/components/display_service/display_service.c: In function 'display_service_create':
C:/msys32/home/user/esp/esp-adf/components/display_service/display_service.c:56:5: error: implicit declaration of function 'AUDIO_NULL_CHECK' [-Werror=implicit-function-declaration]
     AUDIO_NULL_CHECK(TAG, config, return NULL);
     ^
C:/msys32/home/user/esp/esp-adf/components/display_service/display_service.c:56:35: error: expected expression before 'return'
     AUDIO_NULL_CHECK(TAG, config, return NULL);
                                   ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/c/msys32/home/user/esp/esp-adf/esp-idf/make/component_wrapper.mk:274: display_service.o] error 1
make: *** [/c/msys32/home/user/esp/esp-adf/esp-idf/make/project.mk:450: component-display_service-build] error  2
 how to solve this ?
thanks in advance
best regards

jason-mao, 2019-08-19:
@Patsjemoe Which IDF version and Toolchain version you used?

Patsjemoe, 2019-08-19:
Hi,
toolchain :
xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0
esp-idf : v3.2.2

regards
Ludo

Patsjemoe, 2019-08-19:
Hi, 
I have made a clean install of 
 git clone --recursive https://github.com/donny681/esp-adf.git
and now, this problem is no longer active.

Basically, it looks as the problem started when I was trying to add a VOIP application into this same environment (ESP32 audio kit).

Patsjemoe, 2019-08-20:
Hi, 
does anyone know how to move a project from /esp/esp-adf/examples/advanced_examples to the one installed by git clone --recursive https://github.com/donny681/esp-adf.git ?
When I copy the application VOIP to the examples directory, I get errors like because of missing files...when I also copy these under components, I get the error like audio_null_check... is this because of different esp-idf versions ? or is it because I am just missing some files ?

nandibrenna, 2019-08-20:
Why are you trying to use such an old fork from esp-adf?
Are you using the ESP32-A1S?

Patsjemoe, 2019-08-20:
Can you please explain what you mean with old fork ? v3.2.2 is the latest
stable version ...?
I used esp-idf-tools-setup-2.0, but only had problems with missing files.
Version 4.0 is not released yet.

Yes, I am using ESP32-A1S

nandibrenna, 2019-08-20:
I thought because you are using https://github.com/donny681/esp-adf.git? And there were no updates since about 1 year...

Patsjemoe, 2019-08-20:
This version is for the ESP32 audio kit, and the play_mp3 sample works with
it.

If I use the latest standard git version, it does not work because there is
no option to choose the right board with the right codec.

So, I hope to find someone that can tell me how to import the ESP32 audio
kit in the board selection or how to import VOIP into
the donny681 git.

nandibrenna, 2019-08-20:
You can try use this patch for the esp-adf. I made it from donny681/esp-adf, it will certainly not be bug free but it works for me.
[patch.zip](https://github.com/espressif/esp-adf/files/3520239/patch.zip)

Patsjemoe, 2019-08-20:
hi, thanks.
sorry to ask, but how do you use the patch ?

nandibrenna, 2019-08-20:
Look at: https://docs.espressif.com/projects/esp-adf/en/latest/get-started/

git clone --recursive https://github.com/espressif/esp-adf.git
cd esp-adf
git submodule update --init
git am < esp32a1s.patch
then use make menuconfig and select your audio board

Patsjemoe, 2019-08-20:
Hi,
came across following when doing git am < esp32a1s.patch :
 I am not sure what this is for :

*** Please tell me who you are.
Run
  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"
to set your account's default identity.
Omit --global to set the identity only in this repository.

Patsjemoe, 2019-08-20:
Hi,
OK, did the git config mail....seems to be for git to check ....

it gave :
warning: squelched 6 whitespace errors
warning: 11 lines add whitespace errors.
I will now try to make

Patsjemoe, 2019-08-20:
Hi,
OK, menuconfig = ok,
can select the right board 😀

thanks a lot....now to test compilation ...

Patsjemoe, 2019-08-20:
Hi,
It compiled and loaded without any error.
Thanks a lot, this was a great suggestion.
If you ever come to belgium, I buy you a beer 😀....or 2 ...
regards
Ludo

Patsjemoe, 2019-08-20:
Thanks a lot for this solution.

closed 2019-08-20 by Patsjemoe

Patsjemoe, 2020-07-27:
Hi,
I am trying to use the git am < esp32a1s.patch for esp-adf-v2.1, but is not working
how can I add this board to v2.1 ?
I also installed ESP32-A1S-AudioKit and this uses ai_thinker_audio_kit_v2_2 instead of ai_audio board. Is this the same ? It looks as the ESP32-A1S-AudioKit directory has installed the same files as esp-adf. Are they the same as esp-adf-v2.1, or v2.0...or ?
best regards
Ludo