espressif/esp-adf #238: Can you set a Static IP address to ESP32 ?

Patsjemoe, 2019-08-22:
I am trying to get VOIP to run, but get upto start and wait for wifi network...then it seems to disconnect from my router....
I am trying to set the 5060 port in my router, but have no IP address of my board.
1) how can I find my IP address in esp-idf/esp-adf ?
2) can I set a static address ?

error : see attached file.
[esp_disconnects.txt](https://github.com/espressif/esp-adf/files/3529463/esp_disconnects.txt)

nandibrenna, 2019-08-22:
Did you set the SSID and password correctly?

Patsjemoe, 2019-08-22:
> Yes, I did, but will verify once more

Hi,
🙄...there was a silly error in my ssid....thank you to pointing me to
this...
now I get a SIP/2.0 401 Unauthorized error, which is probably caused by a
wrong setting of SIP.
IT is unclear to me what exactly I have to put in this field :
I have a sip identity user@sip.linphone.org...to I have to use then
tcp://user@sip.linphone.org:5060, or tcp://sip:user@sip.linphone.org:5060..or
??

regards
Ludo

Patsjemoe, 2019-08-22:
Hi,
I get a Core  1 panic'ed (LoadProhibited)...how to solve this ?

I (3863) SIP: Conecting...
W (3873) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1319
I Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was
unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060730  A0      : 0x800d6798  A1      :
0x3ffcb430
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      :
0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      :
0x00000008
A10     : 0x3ffc8500  A11     : 0x3f40f224  A12     : 0x3ffc8508  A13     :
0x3ffcd134
A14     : 0x00000005  A15     : 0x00000002  SAR     : 0x00000004  EXCCAUSE:
0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  :
0xffffffff

Backtrace: 0x400014fd:0x3ffcb430 0x400d6795:0x3ffcb440
0x400d482e:0x3ffcb750 0x400f460c:0x3ffcb810 0x400f4c40:0x3ffcb870 0x4
00f4d3a:0x3ffcb890 0x400f5d46:0x3ffcb8c0 0x4008956d:0x3ffcb8e0
0x400d6795: _svfprintf_r at
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../
../../.././newlib/libc/stdio/vfprintf.c:1529

0x400d482e: snprintf at
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../.
./.././newlib/libc/stdio/snprintf.c:116

0x400f460c: _sip_build_request at
/home/michael/work/esp/esp-adf-internal/components/esp_sip/esp_sip.c:1695

0x400f4c40: _sip_uac_request_register at
/home/michael/work/esp/esp-adf-internal/components/esp_sip/esp_sip.c:1695

0x400f4d3a: sip_register at
/home/michael/work/esp/esp-adf-internal/components/esp_sip/esp_sip.c:1695

0x400f5d46: _sip_task at
/home/michael/work/esp/esp-adf-internal/components/esp_sip/esp_sip.c:1695

0x4008956d: vPortTaskWrapper at
C:/msys32/home/ludok/esp/esp-idf/components/freertos/port.c:403

nandibrenna, 2019-08-22:
I think there is also a password for the user?
If so, then use: tcp://user:pass@sip.linphone.org:5060 
You can also try udp://user:pass@sip.linphone.org:5060 if tcp doesn't work.

Patsjemoe, 2019-08-22:
Thanks for this input.
I added the password, and got further in the program.
Then it was waiting for button pushes, but only showed which button was
pushed, nothing happened.
In the mean time, I get again some errors on authentication, need to check
why...
regards
Ludo

Patsjemoe, 2019-08-22:
Hi,
I now get a  SIP: Required authentication error .
I would assume this is only for TLS ?
I tried with TLS, but did not manage. Then I changed back to TCP...and get
this error.
I will do a make clean, as maybe there is something left behind ?
regards

Patsjemoe, 2019-08-22:
Hi,
make clean did not solve the problem.
I still get the authentication error when using tcp.
Is this some function that ESP32 voip SW forgot to implement, or how can I
solve this ?
regards
Ludo

Patsjemoe, 2019-08-23:
I will close this issue, and make a new one for the authentication error

closed 2019-08-23 by Patsjemoe