espressif/esp-adf #239: Voip - Authentication error

Patsjemoe, 2019-08-23:
I get an error which says : 

I (4233) SIP: Required authentication
...
SIP/2.0 401 Unauthorized

Is this an error because of missing/wrong setup in make menuconfig, or is this something missing in the VOIP SW ?
I read that it is supposed to read  nonce=xxx and respond, but how is this done ? I would assume this to be included in the VOIP SW ?

I have used tcp://userID:pass@sip.linphone.org, tried also UDP.

nandibrenna, 2019-08-23:
First check user and password for the SIP server. Have you tried another client to check if the credentials are ok and it works?

Patsjemoe, 2019-08-23:
Hi,
I am now looking in my set up of freepbx...this is sooooo unclear....wonder
why they make it so complex.
should be simple.

regadrs
Ludo

Patsjemoe, 2019-08-24:
Is there anyone that can explain simply what to put/do in freepbx, based on following 
information : linphone SIP identity, linphone username, ESP32 board IP address, ESP32 PSW....
 ?

Patsjemoe, 2019-08-24:
Hi,
I have now setup freepbx, and it starts to connect, then read again, and
then disconnects....any idea ?

(765963) SIP: Conecting...
W (765963) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1319
W (765963) SIP: [1970-01-01/00:12:22]=======WRITE 0600 bytes>>
I (765973) SIP:
.....
......then :
W (766023) SIP: [1970-01-01/00:12:22]=======================>>
W (770033) SIP: Read again
W (774033) SIP: Error read or timeout
E (774033) SIP: Error register device
W (774033) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1334

sosaucily, 2019-10-30:
See https://github.com/espressif/esp-adf/issues/240#issuecomment-547886026 perhaps it will help

sosaucily, 2019-10-31:
I also would like to be able to use something like linphone. Much easier than configuring a local SIP system.

Patsjemoe, 2019-11-01:
Hi Jesse,
I have stopped this project, as nobody that has ESP32 VOIP  up and running
is sharing the necessary info to properly setup freepbx.
I am not sure when I will restart my project, and  if I may use another
solution like raspberry pi.
I prefer to stay with ESP32, but if it is not working......
regards
Ludo

ahhfzhang, 2019-11-01:
Hi @Patsjemoe 
I introduce a simple server to you, it's Yate (acronym for Yet Another Telephony Engine) 

**Installation**: http://docs.yate.ro/wiki/Installation  （Support Window、Linux...）
**Setup :** http://docs.yate.ro/wiki/Beginners_in_Yate 
**Config :**   
1. Set the ports of server UDP, TCP and TLS separately in ysipchan.conf
[image]

2. Add User :
[image]

You can use the Yate client to test if the server is working: https://docs.yate.ro/wiki/SIP_Client
and then, you can setup ESP32 uri like： transport://user:pass@server:port

By the way,We has worked with Chinese telecommunications giant, China Mobile, and add Andlink to this VoIP solution, We also support GSM phone to call ESP32.

sosaucily, 2019-11-05:
@ahhfzhang Thank you for linking in the yate server. I was able to get the esp32 to call a softclient. I am using OS X and I don't see a Yate Server, so I was delaying trying it. But we tried it on Windows and it works great! 

Asterisk / FreePBX was working as well (Thanks to Dave C for his helpful docker https://hub.docker.com/r/tiredofit/freepbx). With this I was able to get the esp32 to connect via a SIP call to another soft client, but the audio didn't work. I think it was a networking problem to the docker network.

sosaucily, 2019-11-22:
@ahhfzhang Sorry to post this here, but do you have any experience with setting up a STUN or TURN server to work with Yate? Seems my NAT config won't allow for even Yate server to work well with RTP audio. Thanks

ahhfzhang, 2019-11-22:
I think you can use Freeswitch : https://freeswitch.org/confluence/display/FREESWITCH/Installation

Patsjemoe, 2019-11-22:
Hi Jesse,
no problem to ask.
But, cannot help you with this.  I hav no experience with stun.
 I am myself completely stuck with esp32 audio, as nobody seems to be
willing to show exactly how to set up a pbxserver to work with esp32.
I temporarily stopped the project as I had to solve some android problems.
regards
ludo

Patsjemoe, 2019-12-11:
Hi Michael,
I Just start to look again into te voip application.
Can this yate be installed properly on a raspberry pi, or
do  you need more performance from a pc ?
Regards
Ludo

ahhfzhang, 2019-12-12:
Hi Patsjemoe,

If you want to install the RPi PBX server, you can try **Asterisk for Raspberry Pi**  
URL: http://www.raspberry-asterisk.org/
Login, and create 2 extensions, one for ESP32, another one for the VoIP phone.

We have no performance requirements for the PBX server，just that Yate is easier to setup.

Patsjemoe, 2019-12-12:
Hi Michael,
I tried already with freepbx without succes,. I want my esp32 to call my
android phone, but nobody explains in detail what exactly to install. .
Looking into Yate, I faced directly an issue with the version for ubuntu.
It is very unclear what to install (it says it is outdated and then show 3
options... Not clear to me).
Can you show how/where exactly to create the 2 extensions?
Regards
Ludo

ahhfzhang, 2019-12-12:
modify regfile.conf 

format : 

> [username]
> password=something
> 

[image]

Patsjemoe, 2019-12-12:
It is not clear what the difference is between following (do you have an
idea ?) :
[image: image.png]
regards
Ludo

Patsjemoe, 2019-12-12:
Hi,
I have ubuntu 18.04.4, so do not understand they refer to old versions

Patsjemoe, 2019-12-12:
Hi Jesse,
I am now focussing on Asterisk on Ubuntu.
Have initiated a username and psw in the sip.conf file.
Try with username.psw@ipaddress:5060 on the ESP32 setup to call the pbx,
but get following :
[image: image.png]
so, it seems not to register at the pbx.(I always had this error also with
freepbx)
any idea ?
regards
Ludo

ahhfzhang, 2019-12-13:
I was wondering if you were using the latest sip lib?

Patsjemoe, 2019-12-13:
Hi,
Euhm....what do you mean ? Where/how to check ?
Have made a new attempt to install freepbx on ubuntu....and succeeded this
time....
Regards
Ludo

Patsjemoe, 2019-12-15:
Hi Jesse,
The videos were very good, I managed to register both esp32 and softphone
to the Asterisk.
Freepbx is still a mess, but focus now first on asterisk, because of the
many videos they made on the subject.
Now the next step, make a call, and make it outside my own network....but
made good progress.
One thing from the esp32 side : I had used user*.*pass instead of user*:*pass
, therefore the authentication issue.
But since the softphone worked, I knew asterisk was working well, and could
solve the esp32 too. 😀
thanks for the suggestion/referral to these videos.
regards
Ludo

closed 2019-12-21 by Patsjemoe

davidHuW, 2022-01-23:
> Hi @Patsjemoe I introduce a simple server to you, it's Yate (acronym for Yet Another Telephony Engine)
> 
> **Installation**: http://docs.yate.ro/wiki/Installation （Support Window、Linux...） **Setup :** http://docs.yate.ro/wiki/Beginners_in_Yate **Config :**
> 
> 1. Set the ports of server UDP, TCP and TLS separately in ysipchan.conf
>    [image]
> 2. Add user in regfile.conf :
>    [image]
> 
> You can use the Yate client to test if the server is working: https://docs.yate.ro/wiki/SIP_Client and then, you can setup ESP32 uri like： transport://user:pass@server:port
> 
> By the way,We has worked with Chinese telecommunications giant, China Mobile, and add Andlink to this VoIP solution, We also support GSM phone to call ESP32.

hi,
i have some problem about using voip case , loop register or disconnect from sip server after connected for a while.
could you tell me which version of adf and idf you used? 
thank you.

Patsjemoe, 2022-01-24:
> like： ***@***.***:port

Hi,
I did not use yate, did not get it right...
Regards
Ludo