espressif/esp-adf #240: VOIP: error register device

Patsjemoe, 2019-08-26:
hi,
can anyone being familiar with freepbx confirm what this error created ? seems to connect, reads again, but it looks as if freepbx or esp32 prgram do not respond

(765963) SIP: Conecting...
W (765963) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1319
W (765963) SIP: [1970-01-01/00:12:22]=======WRITE 0600 bytes>>
I (765973) SIP:
.....
......then :
W (766023) SIP: [1970-01-01/00:12:22]=======================>>
W (770033) SIP: Read again
W (774033) SIP: Error read or timeout
E (774033) SIP: Error register device
W (774033) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1334

Patsjemoe, 2019-08-28:
> Sometimes it connects immediately, sometimes it takes a long time to
> connect. Or it doesn't connect at all. No idea why.
> You can use wireshark to have a look at the sip data between esp32 and
> freepbx.

It connects, this is now visible in asterisk. But it looks as there is info
sent to esp, which is not answered and therefore disconnects

Patsjemoe, 2019-08-28:
If someone who used it, and got connected can just tell me exactly what is
put in freepbx, esp32 and in the sip app, it would help to understand if I
did not do anything stupid.

Patsjemoe, 2019-08-28:
Hi,
I start to be suspicious about the cause of the problem. It seems as freepbx is not getting response from the esp32.
Can anyone running this VOIP program confirm if it really works , so I know where to start looking for the problem ?

nandibrenna, 2019-08-28:
Sometimes it connects immediately, sometimes it takes a long time to connect. Or it doesn't connect at all. No idea why.  
You can use wireshark to have a look at the sip data between esp32 and freepbx.

sosaucily, 2019-10-30:
If you're still stuck on this, I might be able to help a bit. You need to have registered a SIP extension for this esp32 client before asterisk/freepbx will let it connect. I used this series of videos to help me, perhaps it will help you too. I think video 5 or 6 covers it, but it's not bad to start from the beginning. In FreePBX you go to "Applications" -> "Extensions" and add a new SIP extension

https://www.youtube.com/watch?v=3UEBclNDDUE&list=PLnzEbgyK52Gu9fdVDHburrsG3KBIntXFK&index=5

closed 2019-12-21 by Patsjemoe

davidHuW, 2022-01-23:
> If someone who used it, and got connected can just tell me exactly what is put in freepbx, esp32 and in the sip app, it would help to understand if I did not do anything stupid. Op wo 28 aug. 2019 08:42 schreef Ludo Kustermans <ludo.kustermans@gmail.com
> […](#)

hi,
i met the same problem that the esp32 will disconnect with my sip server, have you solved it ?
thanks.