espressif/esp-adf #242: A little bug and improvement to audio_board component..

arduino12, 2019-09-03:
I am adding support for the [SparkFun ESP32 Thing](https://www.sparkfun.com/products/13907) and the [M5StickC](https://docs.m5stack.com/#/en/core/m5stickc?id=m5stickc) boards that I just got,
I think we can improve the audio_board component structure and implementation:
1. Too many code duplication among the boards files.
2. A [bug](https://github.com/espressif/esp-adf/blob/df769620d25eb89dddcdcc729db15b9e796fd824/components/audio_board/lyrat_v4_3/board_pins_config.c#L58) in setting the GPIOs fields (int) with `memset` (bytes).
3. Some GPIOs are [defined](https://github.com/arduino12/esp-adf/blob/master/components/audio_board/lyrat_v4_3/board_def.h#L31L42) in `board_def.h` and some [hard-coded](https://github.com/arduino12/esp-adf/blob/master/components/audio_board/lyrat_v4_3/board_pins_config.c#L53L56) in `board_pin_config.c`,
I think they all should be defined in a header file, also add a common `#define INVALID_GPIO (-1)`.

I'll write my new boards files better, and can send a PR if it's OK to add non-official boards!
Anyway fixing 1,2 and 3 seems to me like a legitimate fix.

Thanks!

jason-mao, 2019-09-12:
@arduino12 Thank you for your report and suggestions.

alfem, 2019-09-17:
I am interested in esp-adf working in a M5StickC. Is @arduino12 PR incoming?

closed 2022-02-08 by jason-mao