espressif/esp-adf #245: Where are "esp_sip.h" and "g711.h" ?

e-music, 2019-09-05:
I'm trying to build this example on PlatformIO. Where can I find these headers:

#include "esp_sip.h"
#include "g711.h"

maxgerhardt, 2019-09-13:
I have a PlatformIO example here in which this library (and pretty much every other esp-adf library) is available: https://github.com/maxgerhardt/pio-esp-adf-example 
The files you are searching for are btw in the `esp-adf-libs` submodule: https://github.com/espressif/esp-adf-libs/tree/master/esp_sip

e-music, 2019-09-14:
Thank you very much for the helpful reply, as usual! I really needed those and have abandoned the project ever since. By the way, are you aware of something similar for the Arduino platform? I supposed it's going to be much easier for me to adapt to the SDK from Espressif since I'm more inclined to C syntax and FreeRTOS, but unfortunately, they changed almost everything which makes it tough to master the API.

maxgerhardt, 2019-09-15:
There are some game libraries with sound support (https://github.com/fdivitto/FabGL) but it all depends on your application and the needed functions. But there's no 'simple arduino equivalent' to ESP-ADF as far as I'm aware of.

closed 2021-02-23 by jason-mao