espressif/esp-adf #246: My attempt to build an example "pipeline_http_mp3"

pilnikov, 2019-09-08:
Hello all!

$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc.exe (crosstool-NG esp32-2019r1) 8.2.0

$ git describe --tags  // for adf
v2.0-beta1-47-gdf76962

$ git describe --tags // for idf
v4.1-dev-141-ga7e8d87d3

Build System: Make

Operating System: Windows

Board: Lyra 4.3

When I try to build an example, I get several errors:

///////////////1111111111111111111111111111111111///////////////////////

z:/esp-adf/components/audio_board/lyrat_v4_3/board_def.h:33:35: error: 'TOUCH_PAD_NUM9' undeclared (first use in this function)
 #define BUTTON_SET_ID             TOUCH_PAD_NUM9
                                   ^~~~~~~~~~~~~~
//////////////22222222222222222222222222222222222/////////////////////

z:/esp-adf/components/esp_peripherals/periph_wifi.c:102:46: error: unknown type name 'smartconfig_status_t'; did you mean 'smartconfig_type_t'?
 static void _wifi_smartconfig_event_callback(smartconfig_status_t status, void *pdata)
                                              ^~~~~~~~~~~~~~~~~~~~
                                              smartconfig_type_t
z:/esp-adf/components/esp_peripherals/periph_wifi.c: In function 'periph_wifi_config_start':
z:/esp-adf/components/esp_peripherals/periph_wifi.c:190:38: error: '_wifi_smartconfig_event_callback' undeclared (first use in this function); did you mean 'smartconfig_event_t'?
         err |= esp_smartconfig_start(_wifi_smartconfig_event_callback, 0);
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                      smartconfig_event_t
z:/esp-adf/components/esp_peripherals/periph_wifi.c:190:38: note: each undeclared identifier is reported only once for each function it appears in
z:/esp-adf/components/esp_peripherals/periph_wifi.c:190:16: error: too many arguments to function 'esp_smartconfig_start'
         err |= esp_smartconfig_start(_wifi_smartconfig_event_callback, 0);
                ^~~~~~~~~~~~~~~~~~~~~
In file included from z:/esp-adf/components/esp_peripherals/periph_wifi.c:37:
Z:/esp-idf/components/esp_wifi/include/esp_smartconfig.h:86:11: note: declared here
 esp_err_t esp_smartconfig_start(const smartconfig_start_config_t *config);
           ^~~~~~~~~~~~~~~~~~~~~
z:/esp-adf/components/esp_peripherals/periph_wifi.c: In function '_wifi_init':
z:/esp-adf/components/esp_peripherals/periph_wifi.c:293:9: error: implicit declaration of function 'esp_event_loop_get_queue'; did you mean 'esp_event_loop_delete'? [-Werror=implicit-function-declaration]
     if (esp_event_loop_get_queue() == NULL) {
         ^~~~~~~~~~~~~~~~~~~~~~~~

////////////////3333333333333333333333333////////////////////////////

z:/esp-adf/components/esp_peripherals/esp_peripherals.c:146:5: error: implicit declaration of function 'gpio_install_isr_service' [-Werror=implicit-function-declaration]
     gpio_install_isr_service(ESP_INTR_FLAG_LEVEL1);
     ^~~~~~~~~~~~~~~~~~~~~~~~
z:/esp-adf/components/esp_peripherals/esp_peripherals.c:146:30: error: 'ESP_INTR_FLAG_LEVEL1' undeclared (first use in this function); did you mean 'ESP_LOG_LEVEL'?
     gpio_install_isr_service(ESP_INTR_FLAG_LEVEL1);
                              ^~~~~~~~~~~~~~~~~~~~
                              ESP_LOG_LEVEL
z:/esp-adf/components/esp_peripherals/esp_peripherals.c:146:30: note: each undeclared identifier is reported only once for each function it appears in
z:/esp-adf/components/esp_peripherals/esp_peripherals.c: In function 'esp_periph_set_destroy':
z:/esp-adf/components/esp_peripherals/esp_peripherals.c:198:5: error: implicit declaration of function 'gpio_uninstall_isr_service' [-Werror=implicit-function-declaration]
     gpio_uninstall_isr_service();
     ^~~~~~~~~~~~~~~~~~~~~~~~~~

///////////////////////////////444444444444444444444///////////////////

z:/esp-adf/components/wifi_service/smart_config/smart_config.c:96:11: error: too many arguments to function 'esp_smartconfig_start'
     ret = esp_smartconfig_start(smartconfg_cb, 1);
           ^~~~~~~~~~~~~~~~~~~~~
In file included from z:/esp-adf/components/wifi_service/include/smart_config.h:28,
                 from z:/esp-adf/components/wifi_service/smart_config/smart_config.c:29:

apparently this is due to changes in the mechanism for connecting to wifi

The first problem was solved by adding a line
#include "driver / touch_pad.h"
to the esp-adf \ components \ audio_board \ lyrat_v4_3 \ board.h and esp-adf \ components \ audio_board \ lyrat_v4_3 \ board.h files

The third - commenting on lines ## 146 198,

Solving problems ## 2 4 requires more dramatic changes. My level of knowledge is not enough for this.

Thanks.

jason-mao, 2019-09-09:
Hi @pilnikov, ADF only support IDF 3.2 now. IDF 4.1 APIs and toolchain have some break changes. Please change your IDF and toolchain to IDF 3.2.

pilnikov, 2019-09-09:
https://github.com/pilnikov/adf/blob/master/esp-adf/components/esp_peripherals/periph_wifi.c
This is how it works. Only there are no messages about the process of connecting to WiFi

pilnikov, 2019-09-09:
I also replaced the file https://github.com/pilnikov/adf/blob/master/esp-adf/components/wifi_service/smart_config/smart_config.c
taking it from esp-idf

closed 2019-11-22 by jason-mao

adrian-fratila, 2019-12-03:
Hi, I have tried to build your adf version of the code with the latest idf 4.1, and it builds but for some reason the module doesn't connect to the WIFI network, although I've tried to connect to 2 different routers. Did you succeed to get your code working?
Any input would be highly appreciated. Thank you.

pilnikov, 2019-12-03:
Yes this is true! 
https://github.com/pilnikov/adf/commit/2a45acda3339c53bd3fb4adfd12070dfaa737b3a
I fixed the error. Give it a try.

adrian-fratila, 2019-12-03:
Good stuff, it works now :)
Thanks a lot!