espressif/esp-adf #249: pipeline_bt_sink not working -  enable controller failed

jponko, 2019-09-22:
Device is esp32 lyraT v4.3 board.
IDF v3.2.2

make erase_flash was run prior to upload.

console output :
jponko@jponko-ideapad:~/esp/pipeline_bt_sink$ make flash monitor
Toolchain path: /home/jponko/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/jponko/esp/esp-idf-v3.2.2/requirements.txt are satisfied.
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting........_____.....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: c4:4f:33:14:ef:bd
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 23136 bytes to 13712...
Wrote 23136 bytes (13712 compressed) at 0x00001000 in 0.1 seconds (effective 2402.3 kbit/s)...
Hash of data verified.
Compressed 852224 bytes to 491425...
Wrote 852224 bytes (491425 compressed) at 0x00010000 in 6.4 seconds (effective 1070.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 106...
Wrote 3072 bytes (106 compressed) at 0x00008000 in 0.0 seconds (effective 10369.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6196
load:0x40078000,len:10180
ho 0 tail 12 room 4
load:0x40080400,len:6660
entry 0x40080764
I (30) boot: ESP-IDF v3.2-103-g4aa1058e8-dirty 2nd stage bootloader
I (30) boot: compile time 10:17:13
I (31) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00124f80
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2acac (175276) map
I (157) esp_image: segment 1: paddr=0x0003acd4 vaddr=0x3ffbdb60 size=0x029ac ( 10668) load
I (161) esp_image: segment 2: paddr=0x0003d688 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/jponko/esp/esp-idf-v3.2.2/components/freertos/xtensa_vectors.S:1779

I (164) esp_image: segment 3: paddr=0x0003da90 vaddr=0x40080400 size=0x02580 (  9600) load
I (176) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x8ec5c (584796) map
0x400d0018: _flash_cache_start at ??:?

I (386) esp_image: segment 5: paddr=0x000cec7c vaddr=0x40082980 size=0x11460 ( 70752) load
0x40082980: coex_timer_ts_start_handler at ??:?

I (427) boot: Loaded app from partition at offset 0x10000
I (427) boot: Disabling RNG early entropy source...
I (428) cpu_start: Pro cpu up.
I (431) cpu_start: Starting app cpu, entry point is 0x4008112c
0x4008112c: call_start_cpu1 at /home/jponko/esp/esp-idf-v3.2.2/components/esp32/cpu_start.c:246

I (0) cpu_start: App cpu up.
I (442) heap_init: Initializing. RAM available for dynamic allocation:
I (448) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (454) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (461) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (467) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (473) heap_init: At 3FFCE278 len 00011D88 (71 KiB): DRAM
I (479) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (485) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (492) heap_init: At 40093DE0 len 0000C220 (48 KiB): IRAM
I (498) cpu_start: Pro cpu start user code
I (181) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (344) BLUETOOTH_EXAMPLE: [ 1 ] Create Bluetooth service
I (344) BTDM_INIT: BT controller compile version [74c9f57]
I (344) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (354) BLUETOOTH_SERVICE: /home/jponko/esp/esp-adf/components/bluetooth_service/bluetooth_service.c:438 (bluetooth_service_start): enable controller failed
I (364) BLUETOOTH_EXAMPLE: [ 2 ] Start codec chip
I (374) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (394) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (394) ES8388_DRIVER: init,out:02, in:00
I (404) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (404) BLUETOOTH_EXAMPLE: [ 3 ] Create audio pipeline for playback
I (414) BLUETOOTH_EXAMPLE: [3.1] Create i2s stream to write data to codec chip
I (424) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (424) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (434) I2S: APLL: Req RATE: 44100, real rate: 43945.238, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11249981.000, SCLK: 1406247.625000, diva: 1, divb: 0
I (444) LYRAT_V4_3: I2S0, MCLK output by GPIO0
I (454) BLUETOOTH_EXAMPLE: [3.2] Get Bluetooth stream
I (454) BLUETOOTH_EXAMPLE: [3.2] Register all elements to audio pipeline
I (464) BLUETOOTH_EXAMPLE: [3.3] Link it together [Bluetooth]-->bt_stream_reader-->i2s_stream_writer-->[codec_chip]
I (474) AUDIO_PIPELINE: link el->rb, el:0x3ffd2588, tag:bt, rb:0x3ffd28cc
I (484) BLUETOOTH_EXAMPLE: [ 4 ] Initialize peripherals
E (494) gpio: gpio_install_isr_service(394): GPIO isr service already installed
I (494) BLUETOOTH_EXAMPLE: [4.1] Initialize Touch peripheral
I (504) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (514) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (524) BLUETOOTH_EXAMPLE: [4.2] Create Bluetooth peripheral
I (524) BLUETOOTH_EXAMPLE: [4.2] Start all peripherals
W (544) PERIPH_TOUCH: _touch_init
I (544) BLUETOOTH_EXAMPLE: [ 5 ] Set up  event listener
I (554) BLUETOOTH_EXAMPLE: [5.1] Listening event from all elements of pipeline
I (554) BLUETOOTH_EXAMPLE: [5.2] Listening event from peripherals
I (564) BLUETOOTH_EXAMPLE: [ 6 ] Start audio_pipeline
I (574) AUDIO_ELEMENT: [bt] Element task created
I (584) AUDIO_ELEMENT: [i2s] Element task created
I (584) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:324, MEM Total:162640 Bytes

I (604) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (604) I2S_STREAM: AUDIO_STREAM_WRITER
I (604) AUDIO_PIPELINE: Pipeline started
I (604) BLUETOOTH_EXAMPLE: [ 7 ] Listen for all pipeline events

fronders, 2019-10-01:
hello! i had the same problem - the sdkconfig.defaults is erroneous - it enables BLE stack instead of Classic. In menuconfig you need to select "BR/EDR Only" for Bluetooth controller mode.

The sdkconfig.defaults that is provided with example is missing the line
```
CONFIG_BTDM_CONTROLLER_MODE_BR_EDR_ONLY=y
```

fronders, 2019-10-01:
same issue found in pipeline_bt_source example

manoj153, 2019-10-08:
yes, noticed this issue fixed it by menuconfig way 

Component config-->Bluetooth-->Bluetooth controller-->Bluetooth controller mode:
Set to : **BR/EDR Only**

jason-mao, 2019-10-22:
Fixed by 751fe30460fdb2e0741ea3e9201f813a60a78c73.

closed 2019-10-22 by jason-mao