espressif/esp-adf #253: demo REC_RAW_HTTP 只能运行一次

flfq, 2019-10-02:
```
Total bytes written: 201728
I (8096) REC_RAW_HTTP: [ * ] Stop pipeline
W (8096) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (8096) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (8096) AUDIO_PIPELINE: audio_pipeline state is not running, 1
Total bytes written: 202752
I (8136) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_POST_REQUEST, write end chunked marker
I (41126) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_FINISH_REQUEST
I (41126) REC_RAW_HTTP: Got HTTP Response = File 20191002T025850Z_16000_16_2.wav was written, size 202752
I (49246) REC_RAW_HTTP: [ * ] Resuming pipeline
Total bytes written: 8192
I (49316) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_POST_REQUEST, write end chunked marker
I (49326) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_FINISH_REQUEST
I (49326) REC_RAW_HTTP: Got HTTP Response = File 20191002T025858Z_16000_16_2.wav was written, size 8192
I (50796) REC_RAW_HTTP: [ * ] Stop pipeline
W (50796) AUDIO_ELEMENT: [i2s] Element already stopped
W (50796) AUDIO_ELEMENT: [http] Element already stopped
W (50796) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (50806) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (50806) AUDIO_PIPELINE: audio_pipeline state is not running, 1

```

flfq, 2019-10-02:
```
if (msg.cmd == PERIPH_BUTTON_PRESSED) {
            ESP_LOGI(TAG, "[ * ] Resuming pipeline");
            audio_element_set_uri(http_stream_writer, CONFIG_SERVER_URI);
            audio_pipeline_run(pipeline);
        } else if (msg.cmd == PERIPH_BUTTON_RELEASE || msg.cmd == PERIPH_BUTTON_LONG_RELEASE) {
            ESP_LOGI(TAG, "[ * ] Stop pipeline");
            audio_pipeline_stop(pipeline);
            audio_pipeline_wait_for_stop(pipeline);
            audio_pipeline_terminate(pipeline); 
```

fix

```
if (msg.cmd == PERIPH_BUTTON_PRESSED) {
            ESP_LOGI(TAG, "[ * ] Resuming pipeline");
            audio_element_set_uri(http_stream_writer, CONFIG_SERVER_URI);
            audio_pipeline_run(pipeline);
        } else if (msg.cmd == PERIPH_BUTTON_RELEASE || msg.cmd == PERIPH_BUTTON_LONG_RELEASE) {
            ESP_LOGI(TAG, "[ * ] Stop pipeline");
            audio_pipeline_stop(pipeline);
            audio_pipeline_wait_for_stop(pipeline);
            //audio_pipeline_terminate(pipeline); 
            http_stream_restart(http_stream_writer);
```

flfq, 2019-10-02:
但是不能获得服务器的返回值

flfq, 2019-10-03:
如果服务器返回信息超过2秒以上，就不能正常获得信息
http_stream 无法获得 RESPONSE FINISH这类关键事件
很多DEMO运行都有这类问题

flfq, 2019-10-06:
问题找到了，是DEMO里的HTTPSERVER在PYTHON2.7下有问题
在服务端close之前增加一个time.sleep(1)就可以了

cattleliu, 2021-08-16:
最后一个是在python程序里面改吗？可以贴出改的代码吗

closed 2022-02-08 by jason-mao