espressif/esp-adf #256: How could I replay the http stream ?

ninghuiy, 2019-10-07:
I tried "record and play" example,  after "play finished", I got some alert messages as below.
How could I replay the "http stream" file? like when I press a button, it will play the same http url. 
 How to manage the pipeline in this case?

I (4134) REC_AND_PLAY_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample
_rates=44100, bits=16, ch=2
W (8645) HTTP_STREAM: No more data,errno:0, total_bytes:139851
W (9765) REC_AND_PLAY_EXAMPLE: [ * ] Stop event received
I (9766) REC_AND_PLAY_EXAMPLE: [ 6 ] release all resources
W (9767) AUDIO_ELEMENT: IN-[filter] AEL_IO_ABORT
W (9772) AUDIO_ELEMENT: [filter] Element already stopped
W (9778) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (9785) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (9791) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (9798) AUDIO_ELEMENT: [http] Element already stopped
W (9803) AUDIO_ELEMENT: [mp3] Element already stopped
W (9809) AUDIO_ELEMENT: [filter] Element already stopped
W (9815) AUDIO_ELEMENT: [i2s] Element already stopped
W (9821) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (9829) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (9835) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (9841) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (9847) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (9854) AUDIO_PIPELINE: There are no listener registered
W (9860) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (9868) AUDIO_ELEMENT: [raw_read] Element has not create when AUDIO_ELEMENT_TERMINATE
W (9877) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERMINATE
W (9885) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (9892) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (9898) AUDIO_PIPELINE: There are no listener registered
W (9905) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE
W (9913) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (9921) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
W (9929) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERMINATE
W (9945) PERIPH_WIFI: Wi-Fi disconnected from SSID HTIC-WIFI, auto-reconnect disabled, reconnect after 1000 ms

Thanks

Ning

jason-mao, 2019-10-18:
@ninghuiy

I think you should call follow APIs after press button.
```
audio_pipeline_reset_ringbuffer(pipeline);
audio_pipeline_reset_elements(pipeline);
audio_pipeline_resume(pipeline);
```

jason-mao, 2019-12-11:
@ninghuiy Did you solved your problem ?

closed 2022-02-08 by jason-mao