espressif/esp-adf #257: Information about filtering low frequencies

WilliamTrdjmn, 2019-10-07:
Hi, 

I'd like to know how you filter low frequencies from wavfile. I work on a similar projet and I don't find any information about how making an efficient bandpass filter for low frequency bands (25Hz to 125 Hz for instance) without resampling. 
Because I'm doing audiostreaming block by block, and I need to keep initial conditions of every filter to keep continuity between each block. So if I resample my signal and filter it in low frequencies and then resample it again to keep a signal sampled at 44100 Hz, I loose initial conditions and so continuities. And that means ''clicks'' between each block (typically 2048 samples).

jason-mao, 2019-10-18:
Hi @WilliamTrdjmn，
I am not sure I have got your exactly means. Are you want to filter the 25-125Hz audio stream
?

WilliamTrdjmn, 2019-10-21:
Hi,

Yes it is. Actually, using scipy and pyaudio, I filter in third octave band
an input audio signal block by block with lfilter_zi to keep the delay
between each block. For medium and high frequencies, there's no issue of
continuity, no «clic». But when I filter low frequencies I need to
downsample the signal before filtering to keep a good filter response and
stability, then I filter with Fs=Fs/downsampling_factor, and then I need to
upsample this signal to sum it with over filtered signals to build the
entire filtered signal. So, this method is well for filtering but I also
need to stream it in real time and this doesn't keep continuity between
each block. I think it's resampling step but I'm not sure.
Do you know a good method to filter low frequencies (25Hz-125Hz) with
Fs=44100Hz audiostream using python/pyaudio ?

Thank you a lot !

Best regards,
William

houhaiyan, 2019-10-22:
I need confirming somthing.  first you need band-stop filter, then you need resample (upresample or downresample)?   Or  first you need  resample (upresample or downresample), then you need band-stop filter?
If it is ture, you can add band-stop filter element before or after resample elment.  And also, we can make special filter which include band-stop filter  for you.

houhaiyan, 2019-10-22:
Hi  William
I need confirming somthing. first you need band-stop filter, then you need resample (upresample or downresample)? Or first you need resample (upresample or downresample), then you need band-stop filter?
 If it is ture, you can add band-stop filter element before or after resample elment. And also, we can make special filter which include band-stop filter for you.
Best regards,

WilliamTrdjmn, 2019-10-22:
Hi @houhaiyan, 

In fact, I'm doing these steps (only for low frequencies 25Hz to 160Hz) :  
1) downsampling audiosignal with lowpass filter for anti aliasing (using scipy.signal.decimate())
2) filtering downsampled audiosignal (using IIR chebychev type 2 and scipy.signal.sosfilt() and scipy.signal.sosfilt_zi())
3) upsampling filtered audiosignal (using scipy.signal.resample())
Comment : there's some clics and discontinuity when I stream it (block by block using pyaudio and callback method)

and for higher frequencies (200Hz to 20kHz):
1) filtering audiosignal (using IIR chebychev type 2 and scipy.signal.sosfilt() and scipy.signal.sosfilt_zi())
Comment : there's no clics and discontinuity when I stream it (block by block using pyaudio and callback method)

Is it better for your understanding ?

houhaiyan, 2019-10-22:
Yes I see the step.  Is the `25Hz to 160Hz` data  important for you?  CB-II is low-pass filter. And it can make you `25Hz to 160Hz` data reserved.  If  you want reserved the `25Hz to 160Hz` data,  i think it is ok.
If you is to get rid of the 25Hz to 160Hz` data, I think your scheme  is error. you should use high-pass filter or band-stop filter

WilliamTrdjmn, 2019-10-22:
It's a third octave band bank filter so there're only bandpass filters

houhaiyan, 2019-10-23:
octave filter is to select band data which you wanted . And to deal with those data. And the center frequency designed as you wanted.  You can use down-resample to improve the precise.
@WilliamTrdjmn

WilliamTrdjmn, 2019-10-28:
And so ? I need to filter low frequencies during the audiostream and the down-resample step add a delay or something which induces «clic» between each block and in the output audiosignal ..

houhaiyan, 2019-10-29:
@WilliamTrdjmn
I am no sure.  I don't known your need (give up the 25Hz to 160Hz data? Or save the 25Hz to 160Hz data to deal with? You should answer my question.)  So I don't known whether you should filter the low frequencies. I think you should follow your need. But in audio stream low frequencies data is important. And octave filter is similar with band-pass filter.

`I need to filter low frequencies during the audiostream and the down-resample step add a delay or something which induces «clic» between each block and in the output audiosignal `
what is `clic`?  If your `clic` is `CLC`, in our project, the resample has included the `CLC`.

houhaiyan, 2019-10-29:
sorry, I made a writing error.  
If your clic is CIC, in our project, the resample has included the CIC.

WilliamTrdjmn, 2019-10-29:
What do you mean CIC ? And yes I need to deal with low frequencies data
(adjust magnitude in fact)

houhaiyan, 2019-10-31:
«clic» what's meaning?

vikramdattu, 2019-12-30:
@WilliamTrdjmn you might want to take a look at equalizer example.
https://github.com/espressif/esp-adf/tree/master/examples/audio_processing/pipeline_equalizer 

It is time domain IIR band pass filter. Setting appropriate gain values in `set_gain[]` array in `equalizer_example.c` should get you needed outcome.

closed 2022-02-08 by jason-mao