espressif/esp-adf #258: ESP32-A1s  ac101怎麼 沒有被放進audio_hal/driver/呢

kimouser64, 2019-10-08:
請問為什麼不放在一起而獨立另外
是要放生了嗎

kimouser64, 2019-10-08:
https://github.com/donny681/esp-adf

timotoots, 2019-10-23:
Could it be possible to add support in the main branch?

pilnikov, 2019-11-04:
https://github.com/pilnikov/adf 
Support is partially implemented, there are not enough functions for processing an analog keyboard, headphone detect and amp control. But the main functionality works. He plays music. Complies with the new management rules through HAL. Can an espressif add this board to the repository?

Flos, 2019-11-22:
Im also missing the support for the AC101 ("Audio Kit")

pilnikov, 2019-11-22:
try my bin files from https://github.com/pilnikov/adf/tree/master/esp-adf/examples/Ka_mod/bin
Several users have already successfully launched their aithinker a1s based boards. 
For myself, I use a very heavily modified version of ADF, which supports the new (4.1.xx) version of IDF with Cmake, ninja && e.t.c.
You can simply replace the components folder with mine and delete the idf folder (inside adf)

jason-mao, 2020-10-15:
Closed, please refer to https://github.com/espressif/esp-adf/issues/341

closed 2020-10-15 by jason-mao