espressif/esp-adf #262: 很多DEMO选择 MINI 语音板会显示 No board.h file

flfq, 2019-10-19:
esp-adf/components/esp_peripherals/periph_button.c:38:19: fatal error: board.h: No such file or directory
compilation terminated.

flfq, 2019-10-19:
```
if (CONFIG_ESP_LYRATD_MINI_V1_1_BOARD)
message(STATUS "Current board name is " CONFIG_ESP_LYRATD_MINI_V1_1_BOARD)
list(APPEND COMPONENT_ADD_INCLUDEDIRS ./lyratd_mini_v1_1)
set(COMPONENT_SRCS
./lyratd_mini_v1_1/board.c
./lyratd_mini_v1_1/board_pins_config.c
)
endif()
```
lyratd_mini_v1_1 应是 lyrat_mini_v1_1

jason-mao, 2019-10-21:
@flfq Thank you for your report.

closed 2019-10-22 by igrr

jalakamkiranadspl, 2021-03-24:
Hey I am getting this issue for the esp32 audio kit(from ai-thinker) too. 
[image]
Can you please tell how you resolved it? Or how can it be resolved?