espressif/esp-adf #264: 【BUG反馈】HFP模式中听筒噪音严重，MIC声音对方听不清

fengyue8029, 2019-10-31:
测试环境：
   1.程序代码: pipeline_a2dp_sink_and_hfp (master主分支)的官方例程,未更改.
   2.硬件环境: ESP32-LyraT v4.3

测试准备：
　在ESP32-LyraT v4.3开发板上，运行 pipeline_a2dp_sink_and_hfp 例程，并使ESP32-LyraT开发板与安卓手机通过蓝牙成功连接。（hfp over HCI）

现象描述：
   1. 直接在安卓手机端播放音乐，ESP32-LyraT开发板的喇叭输出声音，完美无杂音。
   2. 对着ESP32-LyraT开发板上的MIC讲话，并在手机端录音，录音回放完美无杂音。

   3. 在安卓手机上任意拨通一个电话，此时 ESP32-LyraT 开发板的喇叭中，会大概率听到非常明显的"噼噼啪啪"的噪音，十分刺耳.
       反复拨打电话多次, 没有找到噪音的规律。有时,拨打10通电话, 9通带有明显噪音. 有时拨打10通电话, 有５通有噪音, ５通无噪音.  
       问题重现概率：50%以上.

   4. 尝试通过ESP32-LyraT开发板的板载MIC, 与电话另一端的对方讲话,  对方无法听清楚ESP32-LyraT这端的讲话内容。感觉像有"丢帧"现象.  说出的每一个字, 对方都只能听到半个字的声音, 而且声音十分不清晰.  
　　问题重现概率：100%

jason-mao, 2019-11-08:
@fengyue8029 多谢反馈，我们调查一下。

dinasind, 2024-02-27:
```
2. 对着ESP32-LyraT开发板上的MIC讲话，并在手机端录音，录音回放完美无杂音。
```
@fengyue8029  这个是如何做到的？手机端用的什么app？
我想在手机端录音，使用例程pipeline_a2dp_sink_and_hfp 发现代码初始化时：
两个i2s，一个i2s1_cfg.type = AUDIO_STREAM_WRITER; 一个 i2s2_cfg.type = AUDIO_STREAM_READER;
i2s_stream_init(&i2s_cfg); 只有第一个能够初始成功，第二个在初始化时会报错：
E (1512) I2S: register I2S object to platform failed
好像只能有一个i2s设备注册，再注册时会报错。