espressif/esp-adf #266: FreeSWITCH不能通过originate呼通SIP并播放音乐

m8989, 2019-11-09:
FreeSWITCH最新版本。VOIP修改参数和自动应答后，能实现常规的接通SIP电话和按Rec拔出电话了。
但是，在FreeSWITCH服务器使用 originate user/1001  &playback(/12.wav) 不能建立联接。请问我该从哪里着手解决？
测试用到的命令:
originate user/1010 &playback(/12.wav)
originate user/1012 &conference(3000)

ahhfzhang, 2019-11-12:
我理解你的需求是希望服务器来接听客户端或者类似彩铃的功能，这个功能我们是支持的，对于客户端来说其实和常规的通话并没有什么区别，你可以使用一些开源的sip client来测试服务器配置是否正常，或者方便的话上传一些log方便我们来分析。

m8989, 2019-11-14:
感谢大家的关注。我今天特意重装新环境来测试和抓包，希望尽量能简单重现问题。

windows 10 + FreeSWITCH-1.10.1-Release-x64.msi + Wireshark-win64-3.0.6
ADF最新版本及VOIP例程
ESP32——LyraT 4.3开发板
华为手机装 linphone 4.0.1

安装FreeSWITCH软交换机的IP：172.16.16.98
华为手机装 linphone 4.0.1 的IP：172.16.0.223，SIP帐号1001
ESP32——LyraT 4.3开发板    的IP：172.16.0.139，SIP帐号1002

# 1.修改编译ESP32-VOIP
 esp-adf\examples\advanced_examples\voip修改，主要是设置sip帐号+一键拔号号码1001+自动应答
 SIP帐号 tcp://1002:1234@172.16.16.98:5060    //udp也试过，情况一样
在case SIP_EVENT_RINGING:事件中增加 
esp_sip_uas_answer(sip, true);

正常编译刷入后，可以正常的通过按rec键接通linphone.可以接听linphone打入的电话。

# 测试 服务器主动邀请linphone加入会议
打开FreeSWITCH或fs-cli命令终端，执行 originate user/1001 &conference(3000),linphone手机正常进入会议室并听到提示音乐
(有详细抓包)

# 测试 服务器主动邀请ESP32加入会议
打开FreeSWITCH或fs-cli命令终端，执行 originate user/1002 &conference(3000)
(有详细抓包)，ESP32不能进入会议，不会响应INVITE包,内容如下
```
ÄO3å+4oõEÜ¿ @,ð¬b¬ÄÄuINVITE sip:1002@172.16.0.139:5060 SIP/2.0
Via: SIP/2.0/UDP 172.16.16.98;rport;branch=z9hG4bKHDQpDa47BmatH
Max-Forwards: 70
From: "" <sip:0000000000@172.16.16.98>;tag=D2c3c5Z47X0yj
To: <sip:1002@172.16.0.139:5060>
Call-ID: afef09f8-8139-1238-a3ba-1f442d153298
CSeq: 12298984 INVITE
Contact: <sip:mod_sofia@172.16.16.98:5060>
User-Agent: FreeSWITCH-mod_sofia/1.10.1~64bit
Allow: INVITE, ACK, BYE, CANCEL, OPTIONS, MESSAGE, INFO, UPDATE, REGISTER, REFER, NOTIFY, PUBLISH, SUBSCRIBE
Supported: timer, path, replaces
Allow-Events: talk, hold, conference, presence, as-feature-event, dialog, line-seize, call-info, sla, include-session-description, presence.winfo, message-summary, refer
Content-Type: application/sdp
Content-Disposition: session
Content-Length: 742
X-FS-Support: update_display,send_info
Remote-Party-ID: <sip:0000000000@172.16.16.98>;party=calling;screen=yes;privacy=off

v=0
o=FreeSWITCH 1573681387 1573681388 IN IP4 172.16.16.98
s=FreeSWITCH
c=IN IP4 172.16.16.98
t=0 0
m=audio 24166 RTP/AVP 102 9 0 8 105 101
a=rtpmap:102 opus/48000/2
a=fmtp:102 useinbandfec=1; maxaveragebitrate=30000; maxplaybackrate=48000; ptime=20; minptime=10; maxptime=40
a=rtpmap:9 G722/8000
a=rtpmap:0 PCMU/8000
a=rtpmap:8 PCMA/8000
a=rtpmap:105 telephone-event/48000
a=fmtp:105 0-16
a=rtpmap:101 telephone-event/8000
a=fmtp:101 0-16
a=ptime:20
m=video 25808 RTP/AVP 103 104
b=AS:3072
a=rtpmap:103 H264/90000
a=rtpmap:104 VP8/90000
a=rtcp-fb:1
```
具体有什么深层次的原因我也不清楚。这里把我抓的包一起发出来。请求高人解答
[SIP-cap.zip](https://github.com/espressif/esp-adf/files/3844677/SIP-cap.zip)
# 更多测试
为了简化测试，可用换下面一些命令，同样不成功
```
originate user/1002 &echo
originate user/1002  &playback(http://mp3.9ku.com/hot/2004/07-13/10278.mp3)
originate user/1002  &playback(http://mp3.9ku.com/hot/2004/07-13/11271.mp3)

```

ahhfzhang, 2019-11-19:
@m8989  首先谢谢提供抓包分析，从抓包来看是协议的不同导致的，我们ESP32 VoIP目前同一时间只支持一个协议在运行，比如你sip config的地址protocol 是TCP，那么我们监听服务器的信令也是TCP的，而从抓包来看不论是服务器挂断还是服务器接听/会议邀请都是用的是UDP，所以这里我们就没有去处理这个信令，建议如下： 

可以在sip config时也使用UDP的URL 或者 修改服务器配置会议邀请或者信令都是用TCP。
主要是保证ESP32使用的协议和服务器信令协议保证一致。

m8989, 2019-11-19:
问题是我最初就让ESP32使用udp协议的，没有用以后才改用tcp试一试的。抓包时偷懒没再改回去。

ahhfzhang, 2019-11-19:
好的，有趣的功能，我们之前着重还是在P2P的交互上，电话会议我们也多测试一下。

m8989, 2019-11-25:
自己找出原因了。发在这里记录下，同时请官方兼容处理。
我在freeswitch中，禁用opus编码后测试，工作正常
```
ÄO3å+4oõEÜ©c @B¡¬b¬ÄÄýëINVITE sip:1002@172.16.0.138:5060 SIP/2.0
Via: SIP/2.0/UDP 172.16.16.98;rport;branch=z9hG4bKv2BrpB6y5jNZe
Max-Forwards: 70
From: "" <sip:0000000000@172.16.16.98>;tag=c3agUQ12NDm8Q
To: <sip:1002@172.16.0.138:5060>
Call-ID: 6dc568db-89d5-1238-9790-2bbf89c5bbf9
CSeq: 12772234 INVITE
Contact: <sip:mod_sofia@172.16.16.98:5060>
User-Agent: FreeSWITCH-mod_sofia/1.10.1~64bit
Allow: INVITE, ACK, BYE, CANCEL, OPTIONS, MESSAGE, INFO, UPDATE, REGISTER, REFER, NOTIFY, PUBLISH, SUBSCRIBE
Supported: timer, path, replaces
Allow-Events: talk, hold, conference, presence, as-feature-event, dialog, line-seize, call-info, sla, include-session-description, presence.winfo, message-summary, refer
Content-Type: application/sdp
Content-Disposition: session
Content-Length: 622
X-FS-Support: update_display,send_info
Remote-Party-ID: <sip:0000000000@172.16.16.98>;party=calling;screen=yes;privacy=off

v=0
o=FreeSWITCH 1574634353 1574634354 IN IP4 172.16.16.98
s=FreeSWITCH
c=IN IP4 172.16.16.98
t=0 0
m=audio 17700 RTP/AVP 102 9 0 8 104 101
a=rtpmap:102 opus/48000/2
a=fmtp:102 useinbandfec=1; maxaveragebitrate=30000; maxplaybackrate=48000; ptime=20; minptime=10; maxptime=40
a=rtpmap:9 G722/8000
a=rtpmap:0 PCMU/8000
a=rtpmap:8 PCMA/8000
a=rtpmap:104 telephone-event/48000
a=fmtp:104 0-16
a=rtpmap:101 telephone-event/8000
a=fmtp:101 0-16
a=ptime:20
m=video 19352 RTP/AVP 103
b=AS:3072
a=rtpmap:103 H264/90000
a=rtcp-fb:103 ccm fir
a=rtcp-fb:103 cc
```
** ^^^^^^^^^^^^^^^^^上面的包不能响应，是因为有这两行 ** 
```
a=rtpmap:102 opus/48000/2
a=fmtp:102 useinbandfec=1; maxaveragebitrate=30000; maxplaybackrate=48000; ptime=20; minptime=10; maxptime=40
```

猜想是源码里在处理这部分时出错了？ 
今天是重新看ESP32—LyraT 4.3开发板说明书时看到支持 OPUS,希望官方能支持opus编码，同时，也希望开放SIP能选择更多音质更好的格式，最少也要让上面这种包能正常通过解析.

sosaucily, 2020-08-11:
It does not appear to me that OPUS is supported for SIP/RTP audio currently. The encoder is included in the library, but not implemented for sip.

closed 2021-05-11 by espressif-bot