espressif/esp-adf #267: wifi and bluetooth simultaneously to stream audio to i2s

Loucotolo, 2019-11-15:
Hello
I'm using your audio framework
(https://github.com/espressif/esp-adf) on an ESP32-LyraT Development Board hadware.
I would like to know if it is possible to use it to output an i2s audio stream from either wifi and ou bluetooth in the same compilation. 

When running the code compiled with both wifi and bt, on the serial debug I get:

**[0m
[0;32mI (3899) AUDIO_PIPELINE: Pipeline started[0m
[0;32mI (4329) HTTP_STREAM: total_bytes=0[0m
[0;33mW (4599) STAGEFRIGHTMP3_DECODER: Free size[27820 bytes] not enough to allocate[27796 bytes], try to free 1 times[0m
[0;33mW (7599) STAGEFRIGHTMP3_DECODER: Free size[19084 bytes] not enough to allocate[27796 bytes], try to free 2 times[0m
[0;33mW (10599) STAGEFRIGHTMP3_DECODER: Free size[19084 bytes] not enough to allocate[27796 bytes], try to free 3 times[0m**

//--------------------------------------LOG------------------------------

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6328
load:0x40078000,len:10180
load:0x40080400,len:6712
entry 0x40080768
[0;32mI (29) boot: ESP-IDF v3.2-103-g4aa1058e8 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 09:14:35[0m
[0;32mI (29) boot: Enabling RNG early entropy source...[0m
[0;32mI (34) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (65) boot:  1 phy_init         RF data          01 01 0000d000 00001000[0m
[0;32mI (72) boot:  2 factory          factory app      00 00 00010000 00300000[0m
[0;32mI (80) boot: End of partition table[0m
[0;32mI (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2b8c8 (178376) map[0m
[0;32mI (155) esp_image: segment 1: paddr=0x0003b8f0 vaddr=0x3ffbdb60 size=0x033b4 ( 13236) load[0m
[0;32mI (161) esp_image: segment 2: paddr=0x0003ecac vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (163) esp_image: segment 3: paddr=0x0003f0b4 vaddr=0x40080400 size=0x00f5c (  3932) load[0m
[0;32mI (173) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0xea0f0 (958704) map[0m
[0;32mI (516) esp_image: segment 5: paddr=0x0012a110 vaddr=0x4008135c size=0x1574c ( 87884) load[0m
[0;32mI (565) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (566) boot: Disabling RNG early entropy source...[0m
[0;32mI (566) cpu_start: Pro cpu up.[0m
[0;32mI (570) cpu_start: Starting app cpu, entry point is 0x40081048[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (580) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (587) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (593) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (599) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (605) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[0m
[0;32mI (611) heap_init: At 3FFD2DD8 len 0000D228 (52 KiB): DRAM[0m
[0;32mI (617) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (624) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (630) heap_init: At 40096AA8 len 00009558 (37 KiB): IRAM[0m
[0;32mI (636) cpu_start: Pro cpu start user code[0m
[0;32mI (207) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (269) HTTP_SELECT_MP3_EXAMPLE: [ 1 ] Create Bluetooth service[0m
[0;32mI (269) BTDM_INIT: BT controller compile version [74c9f57][0m
[0;32mI (269) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (339) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 0[0m
[0;32mI (699) HTTP_SELECT_MP3_EXAMPLE: [2.0] Create audio pipeline for playback[0m
[0;32mI (699) HTTP_SELECT_MP3_EXAMPLE: [2.1] Create http stream to read data[0m
[0;32mI (709) STAGEFRIGHTMP3_DECODER: MP3 init[0m
[0;32mI (709) HTTP_SELECT_MP3_EXAMPLE: [2.2] Create mp3 decoder to decode mp3 file[0m
[... 45 lines cut ...]
[0;32mI (3799) HTTP_SELECT_MP3_EXAMPLE: [4.2] Listening event from peripherals[0m
[0;31mE (3809) AUDIO_EVT: Error remove listener[0m
[0;31mE (3809) AUDIO_EVT: C:/msys32/home/IOLine/esp/esp-adf/components/audio_pipeline/audio_event_iface.c:187 (audio_event_iface_set_listener): Error cleanup listener[0m
[0;32mI (3829) HTTP_SELECT_MP3_EXAMPLE: [ 5 ] Start audio_pipeline[0m
[0;32mI (3829) AUDIO_ELEMENT: [http] Element task created[0m
[0;32mI (3839) AUDIO_ELEMENT: [mp3] Element task created[0m
[0;32mI (3839) AUDIO_ELEMENT: [i2s] Element task created[0m
[0;32mI (3849) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:360, MEM Total:42028 Bytes

[0m
[0;32mI (3859) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (3869) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (3869) STAGEFRIGHTMP3_DECODER: MP3 opened[0m
[0;32mI (3869) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (3879) I2S_STREAM: AUDIO_STREAM_WRITER[0m
[0;32mI (3889) AUDIO_PIPELINE: Pipeline started[0m
[0;32mI (3889) AUDIO_ELEMENT: [bt] Element task created[0m
[0;32mI (3889) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:360, MEM Total:34748 Bytes

[0m
[0;32mI (3899) AUDIO_PIPELINE: Pipeline started[0m
[0;32mI (4329) HTTP_STREAM: total_bytes=0[0m
[0;33mW (4599) STAGEFRIGHTMP3_DECODER: Free size[27820 bytes] not enough to allocate[27796 bytes], try to free 1 times[0m
[0;33mW (7599) STAGEFRIGHTMP3_DECODER: Free size[19084 bytes] not enough to allocate[27796 bytes], try to free 2 times[0m
[0;33mW (10599) STAGEFRIGHTMP3_DECODER: Free size[19084 bytes] not enough to allocate[27796 bytes], try to free 3 times[0m

jason-mao, 2019-11-20:
@Loucotolo， I think your project out of memory. Did you used the PSRAM on you project?

api-monitor, 2019-11-26:
I have the same problem but in my case I turn off the bluetooth periphal when the system detect a station connect to the esp wifi.

My code to turn off is:

```
esp_periph_stop(bt_periph);
esp_periph_destroy(bt_periph);
audio_pipeline_unregister(pipeline, bt_stream_reader);
audio_element_deinit(bt_stream_reader);

bluetooth_service_destroy();
```

With the previous version of SDK "ESP-ADF Pre-release v2.0-beta1" the system worked perfectly

jason-mao, 2019-11-27:
@api-monitor Could you provide your whole log here?

api-monitor, 2019-11-28:
Here my log.

Note: My program setup both Bluetooth and Wifi initially, but when it detect a Wifi station connected the Bluetooth peripheral is turned off. 
I use the Wifi interface to control the esp board though an html interface hosted on my PC.
When the Wifi is active the pipeline stage are:
fatfs_stream -> mp3 decoder -> I2S.   

`I (0) cpu_start: App cpu up.
I (622) heap_init: Initializing. RAM available for dynamic allocation:
I (629) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (635) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (641) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (647) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (653) heap_init: At 3FFD3020 len 0000CFE0 (51 KiB): DRAM
I (659) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (666) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (672) heap_init: At 40099238 len 00006DC8 (27 KiB): IRAM
I (678) cpu_start: Pro cpu start user code
I (25) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
* NVS configured
I (106) INIT: Start codec chip
I (106) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (116) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (116) ES8388_DRIVER: init,out:02, in:00
I (126) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (136) INIT: Create i2s stream to write data to codec chip
I (136) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (136) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (146) I2S: APLL: Req RATE: 44100, real rate: 43945.238, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11249981.000, SCLK: 1406247.625000, diva: 1, divb: 0
I (156) LYRAT_V4_3: I2S0, MCLK output by GPIO0
I (166) INIT: Initialize peripherals
E (166) gpio: gpio_install_isr_service(394): GPIO isr service already installed
I (176) INIT: Create audio pipeline for playback
I (186) INIT: Set up  event listener
I (186) INIT: Listening event from all elements of pipeline
I (196) INIT: Listening event from peripherals
HTML PERIF
- TCP adapter initialized
- DHCP server stopped
- TCP adapter configured with IP 192.168.200.1/24
- DHCP server started
- Event loop initialized
I (216) wifi: wifi driver task: 3ffda148, prio:23, stack:3584, core=1
I (216) INIT: Bluetooth init service
I (216) wifi: wifi firmware version: 2af77cc
I (226) BTDM_INIT: BT controller compile version [74c9f57]
I (226) wifi: config NVS flash: enabled
I (236) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (236) wifi: config nano formating: disabled
I (246) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (256) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (286) wifi: Init dynamic tx buffer num: 32
I (286) wifi: Init data frame dynamic rx buffer num: 64
I (286) wifi: Init management frame dynamic rx buffer num: 64
I (286) wifi: Init management short buffer num: 32
I (296) wifi: Init static rx buffer size: 1600
I (296) wifi: Init static rx buffer num: 10
I (296) wifi: Init dynamic rx buffer num: 64
- Wifi adapter configured in SoftAP mode
- Wifi network settings applied
I (316) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 0
I (326) wifi: mode : softAP (24:0a:c4:9c:b2:c5)
I (326) wifi: Total power save buffer number: 16
I (326) wifi: Init max length of beacon: 752/752
I (326) wifi: Init max length of beacon: 752/752
- Wifi adapter starting...
I (816) WIFI: Wifi adapter started
- HTTP server started
I (926) INIT: Create Bluetooth peripheral
[... 24 lines cut ...]
I (113256) WIFI: Play/Stop JSON
I (113266) PIPELINE: [ * ] Starting audio pipeline
I (113266) AUDIO_ELEMENT: [i2s] Element task created
I (113266) AUDIO_ELEMENT: [file] Element task created
I (113276) AUDIO_ELEMENT: [mp3] Element task created
I (113276) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:360, MEM Total:31344 Bytes

I (113286) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (113296) I2S_STREAM: AUDIO_STREAM_WRITER
I (113296) AUDIO_ELEMENT: [file] AEL_MSG_CMD_RESUME,state:1
I (113306) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (113316) STAGEFRIGHTMP3_DECODER: MP3 opened
I (113316) AUDIO_PIPELINE: Pipeline started
I (113366) FATFS_STREAM: File size is 6536233 byte,pos:0
W (113456) STAGEFRIGHTMP3_DECODER: Free size[25244 bytes] not enough to allocate[27796 bytes], try to free 1 times
W (116456) STAGEFRIGHTMP3_DECODER: Free size[24784 bytes] not enough to allocate[27796 bytes], try to free 2 times
W (119456) STAGEFRIGHTMP3_DECODER: Free size[23888 bytes] not enough to allocate[27796 bytes], try to free 3 times
I (122456) STAGEFRIGHTMP3_DECODER: Func:mp3_decoder_open, Line:320, MEM Total:23432 Bytes

E (122456) STAGEFRIGHTMP3_DECODER: Allocate[27796 bytes] decoder buffer failed
E (122456) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN
I (122466) STAGEFRIGHTMP3_DECODER: Closed
W (122466) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
W (122476) AUDIO_ELEMENT: OUT-[file] AEL_IO_ABORT
`

jason-mao, 2019-12-11:
@api-monitor 
There is not enough memory for MP3 decoder. Refer to
```c
I (122456) STAGEFRIGHTMP3_DECODER: Func:mp3_decoder_open, Line:320, MEM Total:23432 Bytes

E (122456) STAGEFRIGHTMP3_DECODER: Allocate[27796 bytes] decoder buffer failed
```

api-monitor, 2019-12-14:
I noticed that, infact if I compile my code without the bluetooth part it runs perfectly. 
With the previous version of the SDK "alfa1" it works without out of memory. 
How do I clean up the memory from bluetooth garbage or I have to reboot my system to switch of?

vikramdattu, 2019-12-14:
Not all of the memory is released after bluetooth is disabled again.
The memory used for controller always stays in memory. So about 30K memory always stays. External memory is very much recommended.

closed 2021-02-23 by jason-mao