espressif/esp-adf #270: 编译play_mp3出错

dttlgotv, 2019-11-21:
编译idf没问题，但是编译adf中的play_mp3出错。

> idf.py build
Executing action: all (aliases: build)
Running cmake in directory f:\zhuweiyong\esp_git\esp-adf\examples\get-started\play_mp3\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=1 f:\zhuweiyong\esp_git\esp-adf\examples\get-started\play_mp3"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.11.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/guoxinghua/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/guoxinghua/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/guoxinghua/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works -- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/guoxinghua/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/guoxinghua/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v2.0-beta1-95-g94c68a7
-- Building ESP-IDF components for target esp32
warning: the default selection SR_RUN_WM6_CORE1 (undefined) of <choice SR_RUN_WN6_CORE> (defined at F:/zhuweiyong/esp_git/esp-adf/components/esp-sr/Kconfig.projbuild:83) is not contained in the choice
warning: the default selection CONFIG_MN1_MODEL_FLOAT (undefined) of <choice SR_MN_MODEL_SEL> (defined at F:/zhuweiyong/esp_git/esp-adf/components/esp-sr/Kconfig.projbuild:100) is not contained in the choice
warning: the default selection SR_MN1_CHINESE (undefined) of <choice SR_LANGUAGE_SEL> (defined at F:/zhuweiyong/esp_git/esp-adf/components/esp-sr/Kconfig.projbuild:115) is not contained in the choice
-- Found PythonInterp: C:/Users/guoxinghua/.espressif/python_env/idf4.1_py3.8_env/Scripts/python.exe (found version "3.8")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script F:/zhuweiyong/esp_git/esp-adf/examples/get-started/play_mp3/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script F:/zhuweiyong/esp_git/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script F:/zhuweiyong/esp_git/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script F:/zhuweiyong/esp_git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script F:/zhuweiyong/esp_git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script F:/zhuweiyong/esp_git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script F:/zhuweiyong/esp_git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script F:/zhuweiyong/esp_git/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
CMake Error at F:/zhuweiyong/esp_git/esp-adf/components/audio_hal/CMakeLists.txt:33 (target_link_libraries):
  Cannot specify link libraries for target "audio_hal" which is not built by
  this project.

-- Configuring incomplete, errors occurred!
See also "F:/zhuweiyong/esp_git/esp-adf/examples/get-started/play_mp3/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

DemoKK, 2019-11-21:
make clean 在编译 用最稳定的库

dttlgotv, 2019-11-21:
> 
> 
> make clean 在编译 用最稳定的库

还是如此

jason-mao, 2019-11-21:
@dttlgotv 你编译器是8.2.0 的版本。ADF的IDF目前是3.2  支持5.2.0的编译器，参考#251。

dttlgotv, 2019-11-21:
> 
> 
> @dttlgotv 你编译器是8.2.0 的版本。ADF的IDF目前是3.2 支持5.2.0的编译器，参考#251。

我是windows，麻烦问下怎么指定编译器呢？ 为什么单独idf 我可以编译呢？

dttlgotv, 2019-11-21:
> 
> 
> @dttlgotv 你编译器是8.2.0 的版本。ADF的IDF目前是3.2 支持5.2.0的编译器，参考#251。

 git branch
* (HEAD detached at 4aa1058e8)

这是我adf中idf的版本。。

jason-mao, 2019-11-21:
@dttlgotv Windows请下载这个
https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20181001.zip

dttlgotv, 2019-11-21:
> 
> 
> @dttlgotv Windows请下载这个
> https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20181001.

我系统是windows，但是没用msys2，就是cmd。麻烦告诉我一个下载链接。

另外，如果我下载了一个新的工具链，adf中idf 我checkout成release 4.0，修改什么地方可以指定这个工具呢，毕竟我已经有个最新的了。

dttlgotv, 2019-11-22:
> 
> 
> @dttlgotv Windows请下载这个
> https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20181001.zip

目前我用最新的adf，adf中idf我用release 4.0。
编译出错。。
-- Component directory F:/zhuweiyong/esp_git/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
CMake Error at F:/zhuweiyong/esp_git/esp-idf/tools/cmake/kconfig.cmake:28 (message):
  Windows requires a prebuilt mconf-idf for your platform on the PATH, or an
  MSYS2 version of gcc on the PATH to build mconf-idf.  Consult the setup
  docs for ESP-IDF on Windows.

DemoKK, 2019-11-22:
用esp-adf-v1.0 最稳定的库

dttlgotv, 2019-11-22:
> 
> 
> 用esp-adf-v1.0 最稳定的库
意思是 adf 用 1.0 ，idf 用 4.0？

jason-mao, 2019-11-22:
@dttlgotv IDF  的版本差别很大。目前ADF支持的IDF版本是3.2， 还不支持IDF4.0 。

DemoKK, 2019-11-22:
idf 3.3    adf1.0   这是量产的版本  其他版本bug多

dttlgotv, 2019-11-22:
> 
> 
> @dttlgotv IDF 的版本差别很大。目前ADF支持的IDF版本是3.2， 还不支持IDF4.0 。

idf 我到底用3.2 还是 3.3呢，要量产

dttlgotv, 2019-11-22:
> 
> 
> @dttlgotv IDF 的版本差别很大。目前ADF支持的IDF版本是3.2， 还不支持IDF4.0 。

adf 1.0,  idf 3.2, 后编译仍然出错。

$ make
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
GENCONFIG
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/spi_periph.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/sdio_slave_periph.o
CC build/bootloader/soc/esp32/rtc_wdt.o
CC build/bootloader/soc/esp32/sdmmc_periph.o
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
CC build/bootloader/soc/src//memory_layout_utils.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
esptool.py v2.8-dev
Python requirements from F:/zhuweiyong/esp_git/esp-adf/esp-idf/requirements.txt are satisfied.
Building partitions from /f/zhuweiyong/esp_git/esp-adf/esp-idf/components/partition_table/partitions_singleapp.csv...
CC build/adf_utils/json_utils.o
CC build/adf_utils/cloud_services/aws_sig_v4_signing.o
CC build/adf_utils/cloud_services/baidu_access_token.o
AR build/adf_utils/libadf_utils.a
CC build/app_trace/app_trace.o
CC build/app_trace/host_file_io.o
CC build/app_trace/app_trace_util.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_trace/libapp_trace.a
CC build/app_update/esp_ota_ops.o
AR build/app_update/libapp_update.a
CXX build/asio/asio/asio/src/asio.o
AR build/asio/libasio.a
CC build/audio_hal/audio_hal.o
CC build/audio_hal/driver/es8388/es8388.o
CC build/audio_hal/driver/es8388/headphone_detect.o
[... 145 lines cut ...]
CC build/esp_adc_cal/esp_adc_cal.o
AR build/esp_adc_cal/libesp_adc_cal.a
CC build/esp_event/esp_event_private.o
CC build/esp_event/esp_event.o
CC build/esp_event/default_event_loop.o
AR build/esp_event/libesp_event.a
CC build/esp_http_client/esp_http_client.o
CC build/esp_http_client/lib/http_utils.o
CC build/esp_http_client/lib/http_auth.o
CC build/esp_http_client/lib/http_header.o
AR build/esp_http_client/libesp_http_client.a
CC build/esp_http_server/src/httpd_sess.o
CC build/esp_http_server/src/httpd_parse.o
CC build/esp_http_server/src/httpd_uri.o
CC build/esp_http_server/src/httpd_txrx.o
CC build/esp_http_server/src/httpd_main.o
CC build/esp_http_server/src/util/ctrl_sock.o
AR build/esp_http_server/libesp_http_server.a
CC build/esp_https_ota/src/esp_https_ota.o
F:/zhuweiyong/esp_git/esp-adf/esp-idf/components/esp_https_ota/src/esp_https_ota.c: In function 'esp_https_ota':
F:/zhuweiyong/esp_git/esp-adf/esp-idf/components/esp_https_ota/src/esp_https_ota.c:38:37: error: 'esp_http_client_config_t {aka const struct <anonymous>}' has no member named 'use_global_ca_store'
     if (!config->cert_pem && !config->use_global_ca_store) {
                                     ^
make[1]: *** [/f/zhuweiyong/esp_git/esp-adf/esp-idf/make/component_wrapper.mk:286：src/esp_https_ota.o] 错误 1
make: *** [F:/zhuweiyong/esp_git/esp-adf/esp-idf/make/project.mk:530：component-esp_https_ota-build] 错误 2

jason-mao, 2019-11-22:
@dttlgotv ADF master 分支，IDF版本是3.2，请基于这个版本开发。

dttlgotv, 2019-11-22:
不用的错误，看不到吗！干嘛删除了

发自我的iPhone

dttlgotv, 2019-11-22:
> 
> 
> @dttlgotv ADF master 分支，IDF版本是3.2，请基于这个版本开发。

@jason-mao  又有别的错误

make
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
GENCONFIG
*
* Restart config...
*
*
* Audio HAL
*
Audio board
> 1. ESP32-Lyrat V4.3 (ESP_LYRAT_V4_3_BOARD)
  2. ESP32-Lyrat V4.2 (ESP_LYRAT_V4_2_BOARD)
  3. ESP32-LyraTD-MSC V2.1 (ESP_LYRATD_MSC_V2_1_BOARD)
  4. ESP32-LyraTD-MSC V2.2 (ESP_LYRATD_MSC_V2_2_BOARD)
  5. ESP32-Lyrat-Mini V1.1 (ESP_LYRAT_MINI_V1_1_BOARD) (NEW)
choice[1-5?]: 1
*
* Recorder Engine Configuration
*
Choose recorder engine functionality
  1. VAD ONLY (REC_ENG_ENABLE_VAD_ONLY) (NEW)
  2. VAD and WWE (REC_ENG_ENABLE_VAD_WWE) (NEW)
> 3. VAD, WWE and AMR encoding (REC_ENG_ENABLE_VAD_WWE_AMR) (NEW)
choice[1-3?]: 1
#
# configuration written to /f/zhuweiyong/esp_git/esp-adf/examples/get-started/play_mp3/sdkconfig
#
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/spi_periph.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/sdio_slave_periph.o
CC build/bootloader/soc/esp32/rtc_wdt.o
CC build/bootloader/soc/esp32/sdmmc_periph.o
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/src//memory_layout_utils.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
[... 114 lines cut ...]
CC build/driver/sigmadelta.o
CC build/driver/spi_master.o
CC build/driver/spi_common.o
AR build/driver/libdriver.a
CC build/dueros_service/dueros_service.o
CC build/dueros_service/esp_audio_device_info.o
EMBED build/dueros_service/duer_profile.txt.o
AR build/dueros_service/libdueros_service.a
CC build/esp-adf-libs/esp_codec/audio_sonic.o
CC build/esp-adf-libs/esp_codec/wav_decoder.o
CC build/esp-adf-libs/esp_codec/audio_forge.o
CC build/esp-adf-libs/esp_codec/downmix.o
CC build/esp-adf-libs/esp_codec/wav_encoder.o
CC build/esp-adf-libs/esp_codec/filter_resample.o
CC build/esp-adf-libs/esp_codec/audio_alc.o
CC build/esp-adf-libs/esp_codec/equalizer.o
CC build/esp-adf-libs/audio_misc/rec_eng_helper.o
AR build/esp-adf-libs/libesp-adf-libs.a
CC build/esp-sr/speech_command_recognition/mn_process_commands.o
AR build/esp-sr/libesp-sr.a
make[1]: *** 没有规则可制作目标“F:/zhuweiyong/esp_git/esp-adf/components/esp-tls/esp_tls.c”，由“esp_tls.o” 需求。 停止。
make: *** [F:/zhuweiyong/esp_git/esp-adf/esp-idf/make/project.mk:530：component-esp-tls-build] 错误 2

guoxinghua@DESKTOP-SGJ0K8P MINGW32 /f/zhuweiyong/esp_git/esp-adf/examples/get-started/play_mp3
$

dttlgotv, 2019-11-22:
@jason-mao  master 里面压根没有esp-adf/components/esp-tls这个文件夹

jason-mao, 2019-11-22:
@dttlgotv，esp-tls这个是IDF的组件， 您IDF PTAH  是怎么设置的？

dttlgotv, 2019-11-22:
@jason-mao 
export ADF_PATH="F:/zhuweiyong/esp_git/esp-adf"
export IDF_PATH="F:/zhuweiyong/esp_git/esp-idf"

jason-mao, 2019-11-22:
echo $IDF_PATH，看下路径是否正常。另外，在IDF目录下是否可以找到esp-tls文件夹。

dttlgotv, 2019-11-22:
@jason-mao   路径正常，idf 目录下有这个文件夹的。

jason-mao, 2019-11-22:
还有，ADF的根目录下就有对应版本的IDF，你IDF_PATH使用这个路径。IDF版本的切换，不要忘记submodule update.

dttlgotv, 2019-11-22:
@jason-mao 我正该下idf path 到adf下，正在编译。

idf版本切换后  我都用 git submodule --init

dttlgotv, 2019-11-22:
终于好了。多谢。

你们应该出个文档，不能让用户当小白鼠。。

jason-mao, 2019-11-22:
好的，这个我们正在改进中。

closed 2019-11-27 by jason-mao

vvhh2002, 2019-12-11:
I find a solution.
you should change CMakeList.txt  "target_link_libraries(${COMPONENT_NAME}" to "target_link_libraries(${COMPONENT_TARGET}"

may be someone could submit a pull request

jason-mao, 2019-12-12:
@vvhh2002 which version of IDF you used?

vvhh2002, 2019-12-12:
@jason-mao i have test with 3.3.1 and 3.3

jason-mao, 2019-12-13:
@vvhh2002 Thank you.
 Currently the IDF version of ADF is v3.2.3. `target_link_libraries ` should be `COMPONENT_NAME `. we are going to move IDF to v3.3.1, target_link_libraries(${COMPONENT_TARGET} will be modified.