espressif/esp-adf #272: example voip ，ringing ... press the key( play&Rec) do not pick up

xiangxud, 2019-11-21:
I (6049395) VOIP_EXAMPLE: ringing... RemotePhoneNum 1002
I (6050395) VOIP_EXAMPLE: ringing... RemotePhoneNum 1002
I (6050585) VOIP_EXAMPLE: [ * ] input key id is 3
I (6050585) VOIP_EXAMPLE: [ * ] [Play] input key event
I (6050595) VOIP_EXAMPLE: SIP_EVENT_REQUEST_NETWORK_IP
W (6050595) SIP: [1970-01-01/01:40:25]=======WRITE 0582 bytes>>
I (6050595) SIP:

SIP/2.0 200 OK
Via: SIP/2.0/UDP 39.98.198.244;branch=z9hG4bK73y21KUjacFyj;rport
Contact: <sip:1000@192.168.1.2:5060>
From: "Extension 1002"<sip:1002@172.16.0.149>;tag=F3prQc41D8t7j
To: <sip:1000@100.64.205.21:5060>;tag=-792157122
Call-ID: ea192204-870b-1238-b486-00163e0327ff
CSeq: 12619008 INVITE
Server: ESP32 SIP/2.0
Allow: ACK, INVITE, BYE, CANCEL, OPTIONS, INFO
Content-Length: 152
Content-Type: application/sdp

v=0
o=Z 0 3 IN IP4 192.168.1.2
s=Z
c=IN IP4 192.168.1.2
t=0 0
m=audio 9999 RTP/AVP 8 101
a=rtpmap:8 PCMA/8000
a=rtpmap:101 telephone-event/8000

W (6050645) SIP: [1970-01-01/01:40:25]=======================>>
I (6051455) VOIP_EXAMPLE: ringing... RemotePhoneNum 1002
I (6052365) VOIP_EXAMPLE: [ * ] input key id is 3
I (6052365) VOIP_EXAMPLE: [ * ] [Play] input key event
I (6052455) VOIP_EXAMPLE: ringing... RemotePhoneNum 1002
I (6052455) VOIP_EXAMPLE: SIP_EVENT_REQUEST_NETWORK_IP
W (6052455) SIP: [1970-01-01/01:40:26]=======WRITE 0582 bytes>>
I (6052455) SIP:

SIP/2.0 200 OK
Via: SIP/2.0/UDP 39.98.198.244;branch=z9hG4bK73y21KUjacFyj;rport
Contact: <sip:1000@192.168.1.2:5060>
From: "Extension 1002"<sip:1002@172.16.0.149>;tag=F3prQc41D8t7j
To: <sip:1000@100.64.205.21:5060>;tag=1506586951
Call-ID: ea192204-870b-1238-b486-00163e0327ff
CSeq: 12619008 INVITE
Server: ESP32 SIP/2.0
Allow: ACK, INVITE, BYE, CANCEL, OPTIONS, INFO
Content-Length: 152
Content-Type: application/sdp

v=0
o=Z 0 3 IN IP4 192.168.1.2
s=Z
c=IN IP4 192.168.1.2
t=0 0
m=audio 9999 RTP/AVP 8 101
a=rtpmap:8 PCMA/8000
a=rtpmap:101 telephone-event/8000

W (6052515) SIP: [1970-01-01/01:40:26]=======================>>
I (6053525) VOIP_EXAMPLE: ringing... RemotePhoneNum 1002

ahhfzhang, 2019-11-22:
@xiangxud Can you provide all the logs?

xiangxud, 2019-11-23:
hi；
  我用的是freeswitch搭建得服务器，板子拨出到x-lite可以接通，有语音， x-lite拨入到板子可接通，无语音，不知是服务器设置问题还是其他什么原因，整个串口log文件见附件，记事本可打开
另：我准备把本地识别整合到voip这个例程里面，用adf下esp-sr里的例程发现本地命令无法识别，skainet里面得例程很多库都没包含，而且版本好像不太一致，如果要这样整合有什么好的建议

xiangxud, 2019-11-23:
上封邮件log附件忘了，补上

closed 2021-05-11 by espressif-bot