espressif/esp-adf #274: Bugs after adding SR

pilnikov, 2019-11-22:
[image]

jason-mao, 2019-11-22:
@pilnikov Thanks for your report.

pilnikov, 2019-11-22:
[log.txt](https://github.com/espressif/esp-adf/files/3878409/log.txt)

[image]

file - esp-adf-libs/Cmakelists.txt

pilnikov, 2019-12-01:
[image]
[image]
in file /components/esp-sr/Kconfig.projbuild

pilnikov, 2019-12-02:
One more little remark: I am using cmake version 3.13.4
[image]

And every time I have to replace $ {COMPONENT_NAME) with $ {COMPONENT_LIB)
[image]

[image]

feizi, 2019-12-03:
When we use cmake version 3.5.1 and idfv3.2， $ {COMPONENT_NAME) can not be replaced with $ {COMPONENT_LIB).   We will try to find a way to be compatible with more versions.    
Thanks for your report.

pilnikov, 2020-02-26:
still not correct in file /components/esp-sr/Kconfig.projbuild

[image]

closed 2020-10-15 by espressif-bot