espressif/esp-adf #276: 编译play_mp3出错

dttlgotv, 2019-11-22:
$ make
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
esptool.py v2.8-dev
Python requirements from F:/zhuweiyong/esp_git/esp-adf/esp-idf/requirements.txt are satisfied.
Building partitions from /f/zhuweiyong/esp_git/esp-adf/esp-idf/components/partition_table/partitions_singleapp.csv...
CC build/adf_utils/json_utils.o
CC build/adf_utils/cloud_services/aws_sig_v4_signing.o
CC build/adf_utils/cloud_services/baidu_access_token.o
AR build/adf_utils/libadf_utils.a
CC build/app_trace/app_trace.o
CC build/app_trace/host_file_io.o
CC build/app_trace/app_trace_util.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_trace/libapp_trace.a
CC build/app_update/esp_ota_ops.o
AR build/app_update/libapp_update.a
CXX build/asio/asio/asio/src/asio.o
AR build/asio/libasio.a
CC build/audio_hal/audio_hal.o
CC build/audio_hal/driver/es8388/es8388.o
CC build/audio_hal/driver/es8388/headphone_detect.o
CC build/audio_hal/driver/es8374/es8374.o
CC build/audio_hal/driver/zl38063/zl38063.o
CC build/audio_hal/driver/zl38063/api_lib/vprocTwolf_access.o
CC build/audio_hal/driver/zl38063/api_lib/vproc_common.o
CC build/audio_hal/driver/zl38063/example_apps/tw_ldfwcfg.o
CC build/audio_hal/driver/zl38063/example_apps/tw_ldfw.o
CC build/audio_hal/driver/zl38063/example_apps/tw_spi_access.o
CC build/audio_hal/driver/zl38063/example_apps/tw_ldcfg.o
CC build/audio_hal/driver/zl38063/example_apps/tw_hal_verify.o
AR build/audio_hal/libaudio_hal.a
CC build/audio_pipeline/audio_element.o
CC build/audio_pipeline/audio_pipeline.o
CC build/audio_pipeline/ringbuf.o
CC build/audio_pipeline/audio_event_iface.o
AR build/audio_pipeline/libaudio_pipeline.a
CC build/audio_sal/audio_mem.o
AR build/audio_sal/libaudio_sal.a
CC build/audio_service/bluetooth_service.o
CC build/audio_service/bt_keycontrol.o
AR build/audio_service/libaudio_service.a
CC build/audio_stream/http_stream.o
CC build/audio_stream/spiffs_stream.o
CC build/audio_stream/raw_stream.o
CC build/audio_stream/fatfs_stream.o
CC build/audio_stream/i2s_stream.o
AR build/audio_stream/libaudio_stream.a
AR build/aws_iot/libaws_iot.a
CC build/bootloader_support/src/bootloader_random.o
CC build/bootloader_support/src/flash_encrypt.o
CC build/bootloader_support/src/bootloader_sha.o
CC build/bootloader_support/src/esp_image_format.o
CC build/bootloader_support/src/flash_partitions.o
CC build/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader_support/src/bootloader_clock.o
CC build/bootloader_support/src/secure_boot.o
CC build/bootloader_support/src/bootloader_common.o
CC build/bootloader_support/src/bootloader_utility.o
CC build/bootloader_support/src/efuse.o
CC build/bootloader_support/src/flash_qio_mode.o
CC build/bootloader_support/src/bootloader_flash.o
AR build/bootloader_support/libbootloader_support.a
CC build/bt/bt.o
AR build/bt/libbt.a
AR build/clouds/libclouds.a
CC build/coap/libcoap/src/address.o
CC build/coap/libcoap/src/async.o
CC build/coap/libcoap/src/block.o
CC build/coap/libcoap/src/coap_time.o
CC build/coap/libcoap/src/debug.o
CC build/coap/libcoap/src/encode.o
CC build/coap/libcoap/src/hashkey.o
CC build/coap/libcoap/src/mem.o
CC build/coap/libcoap/src/net.o
CC build/coap/libcoap/src/option.o
CC build/coap/libcoap/src/pdu.o
CC build/coap/libcoap/src/resource.o
CC build/coap/libcoap/src/str.o
CC build/coap/libcoap/src/subscribe.o
CC build/coap/libcoap/src/uri.o
CC build/coap/port/coap_io_socket.o
AR build/coap/libcoap.a
CC build/console/linenoise/linenoise.o
CC build/console/argtable3/argtable3.o
CC build/console/commands.o
CC build/console/split_argv.o
AR build/console/libconsole.a
CXX build/cxx/cxx_exception_stubs.o
CXX build/cxx/cxx_guards.o
AR build/cxx/libcxx.a
CC build/driver/rtc_module.o
CC build/driver/i2c.o
CC build/driver/sdmmc_host.o
CC build/driver/spi_slave.o
CC build/driver/can.o
CC build/driver/sdmmc_transaction.o
CC build/driver/mcpwm.o
CC build/driver/rmt.o
CC build/driver/sdspi_crc.o
CC build/driver/pcnt.o
CC build/driver/timer.o
CC build/driver/gpio.o
CC build/driver/sdspi_host.o
CC build/driver/ledc.o
CC build/driver/periph_ctrl.o
CC build/driver/i2s.o
CC build/driver/sdio_slave.o
CC build/driver/sdspi_transaction.o
CC build/driver/uart.o
CC build/driver/sigmadelta.o
CC build/driver/spi_master.o
CC build/driver/spi_common.o
AR build/driver/libdriver.a
CC build/esp-adf-libs/esp_codec/audio_sonic.o
CC build/esp-adf-libs/esp_codec/wav_decoder.o
CC build/esp-adf-libs/esp_codec/downmix.o
CC build/esp-adf-libs/esp_codec/wav_encoder.o
CC build/esp-adf-libs/esp_codec/filter_resample.o
CC build/esp-adf-libs/esp_codec/equalizer.o
AR build/esp-adf-libs/libesp-adf-libs.a
CC build/esp-sr/speech_command_recognition/mn_process_commands.o
AR build/esp-sr/libesp-sr.a
CC build/esp-tls/esp_tls.o
AR build/esp-tls/libesp-tls.a
CC build/esp32/restore.o
CC build/esp32/int_wdt.o
CC build/esp32/brownout.o
CC build/esp32/system_api.o
CC build/esp32/esp_himem.o
CC build/esp32/dport_access.o
CC build/esp32/esp_timer.o
CC build/esp32/hw_random.o
CC build/esp32/esp_err_to_name.o
CC build/esp32/pm_esp32.o
CC build/esp32/event_loop.o
CC build/esp32/pm_locks.o
CC build/esp32/fast_crypto_ops.o
CC build/esp32/coexist.o
CC build/esp32/core_dump.o
CC build/esp32/reset_reason.o
CC build/esp32/ets_timer_legacy.o
CC build/esp32/intr_alloc.o
CC build/esp32/ipc.o
CC build/esp32/cache_err_int.o
CC build/esp32/freertos_hooks.o
CC build/esp32/clk.o
CC build/esp32/spiram_psram.o
CC build/esp32/esp_timer_esp32.o
CC build/esp32/wifi_init.o
CC build/esp32/lib_printf.o
CC build/esp32/pm_trace.o
CC build/esp32/phy_init.o
CC build/esp32/crosscore_int.o
CC build/esp32/gdbstub.o
CC build/esp32/cache_sram_mmu.o
CC build/esp32/cpu_start.o
CC build/esp32/spiram.o
CC build/esp32/dbg_stubs.o
CC build/esp32/sleep_modes.o
CC build/esp32/task_wdt.o
CC build/esp32/event_default_handlers.o
CC build/esp32/esp_adapter.o
CC build/esp32/stack_check.o
CC build/esp32/panic.o
CC build/esp32/hwcrypto/aes.o
CC build/esp32/hwcrypto/sha.o
AS build/esp32/dport_panic_highint_hdl.o
AR build/esp32/libesp32.a
CC build/esp_adc_cal/esp_adc_cal.o
AR build/esp_adc_cal/libesp_adc_cal.a
CC build/esp_event/esp_event_private.o
CC build/esp_event/esp_event.o
CC build/esp_event/default_event_loop.o
AR build/esp_event/libesp_event.a
CC build/esp_http_client/esp_http_client.o
CC build/esp_http_client/lib/http_utils.o
CC build/esp_http_client/lib/http_auth.o
CC build/esp_http_client/lib/http_header.o
AR build/esp_http_client/libesp_http_client.a
CC build/esp_http_server/src/httpd_sess.o
CC build/esp_http_server/src/httpd_parse.o
CC build/esp_http_server/src/httpd_uri.o
CC build/esp_http_server/src/httpd_txrx.o
CC build/esp_http_server/src/httpd_main.o
CC build/esp_http_server/src/util/ctrl_sock.o
AR build/esp_http_server/libesp_http_server.a
CC build/esp_https_ota/src/esp_https_ota.o
F:/zhuweiyong/esp_git/esp-adf/esp-idf/components/esp_https_ota/src/esp_https_ota.c: In function 'esp_https_ota':
F:/zhuweiyong/esp_git/esp-adf/esp-idf/components/esp_https_ota/src/esp_https_ota.c:38:37: error: 'esp_http_client_config_t {aka const struct <anonymous>}' has no member named 'use_global_ca_store'
     if (!config->cert_pem && !config->use_global_ca_store) {
                                     ^
make[1]: *** [/f/zhuweiyong/esp_git/esp-adf/esp-idf/make/component_wrapper.mk:286：src/esp_https_ota.o] 错误 1
make: *** [F:/zhuweiyong/esp_git/esp-adf/esp-idf/make/project.mk:530：component-esp_https_ota-build] 错误 2

dttlgotv, 2019-11-22:
adf 1.0 , idf 3.2.

why error?

jason-mao, 2019-11-22:
Please don't create duplicate issue #270.

closed 2019-11-22 by jason-mao