espressif/esp-adf #278: Can do something before recording after wake up？

xiruilin, 2019-11-23:
Hi,
When waked up with the wake-up word "nihaoxiaoxin", rec_engine_cb() is called directly.
Can configure a custom callback function, call this function after wake-up to do something, 
and then call the function rec_engine_trigger_start() to start recording?
Thanks

jason-mao, 2019-11-27:
@xiruilin Could you  give more information for what want to do in the `custom callback function`?

xiruilin, 2019-11-27:
@jason-mao 

Maybe to control other devices, such as toy cars, cameras,
or play the voice prompt, or judge whether it is in microphone mute mode.
Anyway, I don't want to put these extra sounds into the microphone.
It will add a wake-up callback function.
After some extra processing is done in this callback function,
it is decided whether to call "rec_engine_trigger_start()" to turn on the microphone.
Thanks.

jason-mao, 2019-12-11:
@xiruilin
I think the current implementation meets your needs. 
The REC_EVENT_WAKEUP_START event posted after wake-up, then you can do what you want to do at the callback function, after that the ` rec_engine` detecting the voice by VAD, REC_EVENT_VAD_START means got active voice, silence a while (`vad_off_delay_ms`)  post REC_EVENT_VAD_STOP event. Out of the `wakeup_time_ms`post REC_EVENT_WAKEUP_END  event.

xiruilin, 2019-12-12:
Oh, thanks

closed 2019-12-12 by xiruilin