espressif/esp-adf #280: 编译esp-adf/examples/get-started/play_mp3 报错

woody5418, 2019-11-27:
/home/woody/esp/esp-adf/components/audio_stream/http_stream.c: In function '_http_open':
/home/woody/esp/esp-adf/components/audio_stream/http_stream.c:501:9: error: implicit declaration of function 'esp_http_client_set_redirection' [-Werror=implicit-function-declaration]
         esp_http_client_set_redirection(http->client);
         ^
cc1: some warnings being treated as errors
/home/woody/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'http_stream.o' failed
make[1]: *** [http_stream.o] Error 1
/home/woody/esp/esp-idf/make/project.mk:527: recipe for target 'component-audio_stream-build' failed
make: *** [component-audio_stream-build] Error 2
----------------------------------------------------------------
play_mp3文件夹想执行了make menuconfig  然后执行make 报错  又试了一下别的都是这个问题

jason-mao, 2019-11-27:
@woody5418 应该是你的IDF 版本和ADF不匹配。现在ADF支持的IDF版本是V3.2.3，你可以在ADF根目录下看到esp-idf目录，请用这个版本编译ADF。

woody5418, 2019-11-28:
我更换了IDF 3.2.3版本的IDF  我编译以后还是报错  报错内容如下：
home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:38:10: error: #include expects "FILENAME" or <FILENAME>
 #include CURSES_LOC
          ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:103:2: error: unknown type name ‘chtype’
  chtype atr; /* Color attribute */
  ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:200:16: error: unknown type name ‘WINDOW’
 int on_key_esc(WINDOW *win);
                ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:221:17: error: unknown type name ‘WINDOW’
 void attr_clear(WINDOW * win, int height, int width, chtype attr);
                 ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:221:54: error: unknown type name ‘chtype’
 void attr_clear(WINDOW * win, int height, int width, chtype attr);
                                                      ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:223:21: error: unknown type name ‘WINDOW’
 void print_autowrap(WINDOW * win, const char *prompt, int width, int y, int x);
                     ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:224:19: error: unknown type name ‘WINDOW’
 void print_button(WINDOW * win, const char *label, int y, int x, int selected);
                   ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:225:18: error: unknown type name ‘WINDOW’
 void print_title(WINDOW *dialog, const char *title, int width);
                  ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:226:15: error: unknown type name ‘WINDOW’
 void draw_box(WINDOW * win, int y, int x, int height, int width, chtype box,
               ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:226:66: error: unknown type name ‘chtype’
 void draw_box(WINDOW * win, int y, int x, int height, int width, chtype box,
                                                                  ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:227:8: error: unknown type name ‘chtype’
        chtype border);
        ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:228:18: error: unknown type name ‘WINDOW’
 void draw_shadow(WINDOW * win, int y, int x, int height, int width);
                  ^
/home/woody/esp/esp-idf/tools/kconfig/mconf.c: In function ‘conf’:
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:97:29: error: ‘KEY_MAX’ undeclared (first use in this function)
 #define ERRDISPLAYTOOSMALL (KEY_MAX + 1)
                             ^
/home/woody/esp/esp-idf/tools/kconfig/mconf.c:671:45: note: in expansion of macro ‘ERRDISPLAYTOOSMALL’
   if (res == 1 || res == KEY_ESC || res == -ERRDISPLAYTOOSMALL)
                                             ^
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:97:29: note: each undeclared identifier is reported only once for each function it appears in
 #define ERRDISPLAYTOOSMALL (KEY_MAX + 1)
                             ^
/home/woody/esp/esp-idf/tools/kconfig/mconf.c:671:45: note: in expansion of macro ‘ERRDISPLAYTOOSMALL’
   if (res == 1 || res == KEY_ESC || res == -ERRDISPLAYTOOSMALL)
                                             ^
/home/woody/esp/esp-idf/tools/kconfig/mconf.c: In function ‘show_help’:
/home/woody/esp/esp-idf/tools/kconfig/mconf.c:793:19: warning: implicit declaration of function ‘getmaxx’ [-Wimplicit-function-declaration]
  help.max_width = getmaxx(stdscr) - 10;
                   ^
/home/woody/esp/esp-idf/tools/kconfig/mconf.c:793:27: error: ‘stdscr’ undeclared (first use in this function)
  help.max_width = getmaxx(stdscr) - 10;
                           ^
In file included from /home/woody/esp/esp-idf/tools/kconfig/mconf.c:23:0:
/home/woody/esp/esp-idf/tools/kconfig/mconf.c: In function ‘conf_choice’:
/home/woody/esp/esp-idf/tools/kconfig/lxdialog/dialog.h:97:29: error: ‘KEY_MAX’ undeclared (first use in this function)
 #define ERRDISPLAYTOOSMALL (KEY_MAX + 1)
                             ^
/home/woody/esp/esp-idf/tools/kconfig/mconf.c:855:9: note: in expansion of macro ‘ERRDISPLAYTOOSMALL’
   case -ERRDISPLAYTOOSMALL:
         ^
Makefile:171: recipe for target 'mconf.o' failed
make[1]: *** [mconf.o] Error 1
make[1]: Leaving directory '/home/woody/esp/esp-idf/tools/kconfig'
make: *** No rule to make target '/home/woody/esp/esp-idf/tools/kconfig/conf-idf', needed by '/home/woody/esp/hello_world/build/include/config/auto.conf'.  Stop.

woody5418, 2019-11-28:
我的编译链的版本是xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz ADF是最新的  IDF是 ADF里面的esp-idf @ 90af68b这个版本

jason-mao, 2019-12-11:
@woody5418  这个问题怎么样了？

woody5418, 2019-12-11:
已经解决了就是IDF版本的问题 之前写camera的时候版本太高了 换到ADF编译不过 现在都解决了 。还有遇到一个问题是我现在采用的是按键按下去录音 然后松手录音完成。但是我想做到按一下就录音 然后我说完以后有没有一个函数可以检测什么时候录音结束，因为我把ASR历程的唤醒词加进去了 但是目前还做不到不用按键控制录音...有好的方法么recoder的例子我看都是用按键录音 或者触发以后录音10s  时间到了才结束 不是自动录音完事就结束的。

jason-mao, 2019-12-11:
我想你需要的是https://github.com/espressif/esp-adf-libs/blob/620689e1098e4f59d3156ecd8b445b2460f31aac/recorder_engine/include/recorder_engine.h。 这个函数集成了唤醒，VAD。按键可以通过`rec_engine_trigger_start`触发，然后VAD结束。
使用的例子参考https://github.com/espressif/esp-adf/blob/1d0f4d287b4b17b93b61e051382c09f58b2fc372/examples/dueros/main/dueros_app.c#L70

woody5418, 2019-12-11:
好的我看一下 非常感谢谢

woody5418, 2019-12-11:
可以加您的QQ么 因为ADF的问题貌似我现在加的一些安信可的群或者乐鑫的 都很少有人在用.遇到一些问题找不到人问

jason-mao, 2019-12-11:
@woody5418 有任何问题您可以随时在这里提出来。您是个人开发者，还是公司？

woody5418, 2019-12-11:
公司的不是个人

jason-mao, 2019-12-11:
@woody5418 您可以先提供相关项目和公司信息到sales@espressif.com，后续我们会联系您。

woody5418, 2019-12-11:
好的

closed 2019-12-11 by jason-mao