espressif/esp-adf #287: SIP required authentication

Patsjemoe, 2019-12-13:
Hi,
Making a new attempt on VOIP with ESP32 audio board and freepbx.
I get a 401 unauthorized, but as far as I understand, this is created by the pbx , and on this, the ESP32 should send a response with the right credentials. (realm, nonce...)
Then I get following : 
I (310803) SIP: Required authentication
E (310813) SIP: Error register device
So I assume now that the ESP32 is not answering to the 401 request....???
If it is now aswering, how can other people get this VOIP to work ?
regards
Ludo

Patsjemoe, 2019-12-17:
Am using now Asterisk, and am able to have the device registered (shown on screen in asterisk), but unreacheable.
In the SW from the ESP32, we have the following : 
 case SIP_EVENT_REGISTERED:
            ESP_LOGI(TAG, "SIP_EVENT_REGISTERED");
            break;
...but this SIP EVENT REGISTERED is not shown on the screen, togehter with a strange wording
SIP: Error register device Success
..is it an error, or a succes ...?
Regards
Ludo

kristianeurotronik, 2020-03-09:
Hi Patsjemoe, 

how did you fixed the original issue? I am also getting SIP: Required authentication. 

Thank you in advance

Best regards

Kristian

kristianeurotronik, 2020-03-09:
> Hi,
> Making a new attempt on VOIP with ESP32 audio board and freepbx.
> I get a 401 unauthorized, but as far as I understand, this is created by the pbx , and on this, the ESP32 should send a response [...]

Hi Patsjemoe,

how did you fixed the original issue? I am also getting SIP: Required authentication.

Thank you in advance

Best regards

Kristian

Patsjemoe, 2020-03-11:
Hi,
Due to lack of response, I froze temporarily my project.
If I get active again  on the application and have an answer, I will inform
you too.
Sorry I cannot do more now.
Best regards

ahhfzhang, 2020-04-09:
Hi, 
We've been testing a lot on the FreePBX recently, bugfix commit: https://github.com/espressif/esp-adf/commit/20b0827
hope you can try again.

many thanks.