espressif/esp-adf #288: speech_recognition/asr/ 无法运行

xpdrycry, 2019-12-15:
你好，根据官方提供的esp-adf/examples/speech_recognition/asr/，我们下载到v4.3的板子上面以后，发现出现了错误，没有办法检测到关键字，出现的信息如下；
I (214) boot: Loaded app from partition at offset 0x10000
I (214) boot: Disabling RNG early entropy source...
I (215) cpu_start: Pro cpu up.
I (218) cpu_start: Starting app cpu, entry point is 0x40080f98
0x40080f98: call_start_cpu1 at C:/msys32/home/Administrator/esp/esp-skainet/esp-idf/components/esp32/cpu_start.c:246

I (0) cpu_start: App cpu up.
I (229) heap_init: Initializing. RAM available for dynamic allocation:
I (236) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (242) heap_init: At 3FFB2FE0 len 0002D020 (180 KiB): DRAM
I (248) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (254) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (261) heap_init: At 4008B3A0 len 00014C60 (83 KiB): IRAM
I (267) cpu_start: Pro cpu start user code
I (60) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (61) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (70) example_asr_keywords: Initialize SR handle
wakeNet3
Quantized wakeNet3: wakeNet3_v1_hilexin_5_0.90_0.85, mode:0 (Nov 20 2019 14:34:44)
I (84) example_asr_keywords: keywords2: hilexin (index = 1)
Fail: index is out of range, the min index is 1, the max index is 1I (90) asr_event: keywords_num = 1, threshold = 0.000000, sample_rate = 16000, chunksize =
480, sizeof_uint16 = 2
I (107) asr_event: [ 1 ] Start codec chip
I (130) asr_event: [ 2.0 ] Create audio pipeline for recording
I (130) asr_event: [ 2.1 ] Create i2s stream to read audio data from codec chip
I (136) asr_event: [ 2.2 ] Create filter to resample audio data
I (140) asr_event: [ 2.3 ] Create raw to receive data
I (146) asr_event: [ 3 ] Register all elements to audio pipeline
I (152) asr_event: [ 4 ] Link elements together [codec_chip]-->i2s_stream-->filter-->raw-->[SR]
I (162) asr_event: [ 5 ] Start audio_pipeline
是不是这条语句有问题？
Fail: index is out of range, the min index is 1, the max index is 1I

withdra, 2019-12-19:
hi @xpdrycry 你可以尝试一下用 adf/example/recorder/pipeline_wav_sdcard 录一下音， 看是否可以录到数据， 怀疑是你的板子录音有问题导致的。 我用asr这个demo和你一样的环境是可以唤醒的

closed 2020-02-21 by jason-mao

bobo-ba, 2021-10-15:
你好 请问你最终是怎么解决这个问题的呢？