espressif/esp-adf #290: esp-adf的pipeline_spiffs_mp3例子，只能播放一次

woody5418, 2019-12-18:
按键按一次以后进入播放 ，播放以后结束 但是再次进入的时候无法播放。
audio_element_set_uri(spiffs_stream_reader, "/spiffs/enwozai_01.mp3");
            audio_pipeline_run(pipeline_s);  //error

            ESP_LOGI(TAG, "[ pw ] Get semaphore %d, %d", msg.source_type, msg.cmd);
            if (msg.source_type == AUDIO_ELEMENT_TYPE_ELEMENT && msg.source == (void *)mp3_decoder_s && msg.cmd == AEL_MSG_CMD_REPORT_MUSIC_INFO)
            {
                audio_element_info_t music_info_s = {0};
                audio_element_getinfo(mp3_decoder_s, &music_info_s);

                ESP_LOGI(TAG, "[ * ] Receive music info from mp3 decoder, sample_rates=%d, bits=%d, ch=%d",
                         music_info_s.sample_rates, music_info_s.bits, music_info_s.channels);

                audio_element_setinfo(i2s_stream_writer_s, &music_info_s);
                i2s_stream_set_clk(i2s_stream_writer_s, music_info_s.sample_rates, music_info_s.bits, music_info_s.channels);
                continue;
            }

            /* Stop when the last pipeline element (i2s_stream_writer in this case) receives stop event */
            if (msg.source_type == AUDIO_ELEMENT_TYPE_ELEMENT && msg.source == (void *)i2s_stream_writer_s && msg.cmd == AEL_MSG_CMD_REPORT_STATUS && (((int)msg.data == AEL_STATUS_STATE_STOPPED) || ((int)msg.data == AEL_STATUS_STATE_FINISHED)))
            {
                ESP_LOGW(TAG, "[ * ] Stop event received......");
                audio_pipeline_stop(pipeline_s);
                audio_pipeline_wait_for_stop(pipeline_s);
                audio_pipeline_reset_ringbuffer(pipeline_s);
                audio_pipeline_change_state(pipeline_s, AEL_STATE_INIT);
                audio_pipeline_reset_items_state(pipeline_s);
                audio_pipeline_run(pipeline_a);
            }
想知道使用spiffs结束MP3文件播放的时候我再次进入以后时候需要重新挂载或者使用一些其他的函数。

shootao, 2019-12-18:
你好, 
  1. 你的 ` audio_pipeline_run(pipeline_a)`的参数 `pipeline_a`  确定一下是不是为 `pipeline_s`;  
  2. 关于继续播放的调用可以看一下 [adf example](https://github.com/espressif/esp-adf/blob/master/examples/advanced_examples/dlna/main/audio_player.c) 里面 `audio_player_play` 函数的实现.

woody5418, 2019-12-18:
参数audio_pipeline_run(pipeline_a);是启用唤醒词的。目前的功能是唤醒词（pipeline_a）唤醒以后关闭唤醒，进入到播放（pipeline_s）然后播放结束以后开启唤醒词识别，现在是重启以后第一遍是可以的，，播放结束以后我需要关闭（pipeline_s）然后开启唤醒词的pipline，问题是可以多吃进去唤醒词判断，但是无法播放。

woody5418, 2019-12-18:
Welcome to minicom 2.7

OPTIONS: I18n 
Compiled on Nov 15 2018, 20:18:47.
Port /dev/ttyUSB0, 16:30:32

Press CTRL-A Z for help on special keys

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6604
load:0x40078000,len:10752                                                                                                                                                                                                   
load:0x40080400,len:6652                                                                                                                                                                                                    
entry 0x40080778                                                                                                                                                                                                            
I (61) boot: Chip Revision: 1                                                                                                                                                                                               
I (63) boot_comm: mismatch chip revision, expect 1, found 0                                                                                                                                                                 
I (38) boot: ESP-IDF v3.2.3-33-g90af68b 2nd stage bootloader                                                                                                                                                                
I (38) boot: compile time 11:18:43                                                                                                                                                                                          
I (52) boot: Enabling RNG early entropy source...                                                                                                                                                                           
I (52) boot: SPI Speed      : 80MHz                                                                                                                                                                                         
I (52) boot: SPI Mode       : DIO                                                                                                                                                                                           
I (54) boot: SPI Flash Size : 4MB                                                                                                                                                                                           
I (59) boot: Partition Table:                                                                                                                                                                                               
I (62) boot: ## Label            Usage          Type ST Offset   Length                                                                                                                                                     
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000                                                                                                                                                   
I (77) boot:  1 phy_init         RF data          01 01 0000d000 00001000                                                                                                                                                   
I (84) boot:  2 factory          factory app      00 00 00010000 00200000                                                                                                                                                   
I (92) boot:  3 storage          Unknown data     01 82 00300000 00100000                                                                                                                                                   
I (99) boot: End of partition table                                                                                                                                                                                         
I (103) boot_comm: mismatch chip revision, expect 1, found 0                                                                                                                                                                
I (110) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x48048 (294984) map                                                                                                                                   
I (205) esp_image: segment 1: paddr=0x00058070 vaddr=0x3ffb0000 size=0x02fac ( 12204) load                                                                                                                                  
I (209) esp_image: segment 2: paddr=0x0005b024 vaddr=0x40080000 size=0x00400 (  1024) load                                                                                                                                  
I (212) esp_image: segment 3: paddr=0x0005b42c vaddr=0x40080400 size=0x04be4 ( 19428) load                                                                                                                                  
[... cut ...]
W (39059) AUDIO_ELEMENT: OUT-[mp3] AEL_IO_ABORT                                                                                                                                                                             
W (39059) STAGEFRIGHTMP3_DECODER: output aborted -3                                                                                                                                                                         
W (39069) AUDIO_PIPELINE: audio_pipeline state is not running, 1

woody5418, 2019-12-18:
第二次唤醒的时候“msg.source == (void *)mp3_decoder_s”这个条件貌似不成立。所以没有播放

shootao, 2019-12-18:
你将 `msg.data == AEL_STATUS_STATE_STOPPED` 去掉试一下
```C
if (msg.source_type == AUDIO_ELEMENT_TYPE_ELEMENT && msg.source == (void *) i2s_stream_writer
            && msg.cmd == AEL_MSG_CMD_REPORT_STATUS
            &&  ((int)msg.data == AEL_STATUS_STATE_FINISHED)) {
            ESP_LOGW(TAG, "[ * ] Stop event received");
            audio_pipeline_stop(pipeline);
            audio_pipeline_wait_for_stop(pipeline);
            audio_element_reset_state(mp3_decoder);
            audio_element_reset_state(i2s_stream_writer);
            audio_pipeline_reset_ringbuffer(pipeline);
            audio_pipeline_reset_items_state(pipeline);
            audio_pipeline_run(pipeline);
            // break;
        }
```

woody5418, 2019-12-18:
取消“msg.data == AEL_STATUS_STATE_STOPPED”这个以后  第二次唤醒会提示
I (15059) main_app: Wake up                                                                                                                                                                                                 
W (15059) AUDIO_ELEMENT: IN-[filter] AEL_IO_ABORT                                                                                                                                                                           
W (15059) AUDIO_ELEMENT: [filter] Element already stopped                                                                                                                                                                   
W (15059) AUDIO_PIPELINE: audio_pipeline state is not running, 1                                                                                                                                                            
W (15079) SPIFFS_STREAM: No more data, ret:0                                                                                                                                                                                
I (15089) main_app: [ pw ] Get semaphore 131072, 8                                                                                                                                                                          
W (15089) main_app: [ * ] Stop event received......                                                                                                                                                                         
W (15089) AUDIO_ELEMENT: [spiffs] Element already stopped                                                                                                                                                                   
W (15099) AUDIO_ELEMENT: OUT-[mp3] AEL_IO_ABORT                                                                                                                                                                             
W (15099) STAGEFRIGHTMP3_DECODER: output aborted -3                                                                                                                                                                         
W (15109) AUDIO_PIPELINE: audio_pipeline state is not running, 1

shootao, 2019-12-18:
看你的log, 好像是刚  `wake up`  就抛了  `FINISHED` 事件, 这时候你的 `pipeline_s` 是 reset 了, 但是 `pipeline_a` 是什么状态? 
正常的 `spiffs` 连续播放是没有问题的

woody5418, 2019-12-19:
wakeup的时候 关闭了ASI的pipeline_a，然后开启pipeline_s，等待播放结束以后关闭pipeline_s，再开启pipeline_a

woody5418, 2019-12-19:
我第二次唤醒以后结束pipeline_a以后进入到pipeline_s的时候，这个条件“msg.source == (void *)i2s_stream_writer_s”一直成立，而且“(((int)msg.data == AEL_STATUS_STATE_STOPPED) || ((int)msg.data == AEL_STATUS_STATE_FINISHED))”也是成立的，是我结束上一次播放的时候，某些东西没有重置么？

shootao, 2019-12-19:
因为你上一次结束时调了 `audio_pipeline_stop(pipeline_s)` , 所有会有 `i2s_stream_writer_s` 的 `stop` 事件
看你发过来的 `Log` 就是因为一直有 *`stop`* *事件* 过来导致的

woody5418, 2019-12-19:
我一次结束的时候调用了
audio_pipeline_stop(pipeline_s);
audio_pipeline_wait_for_stop(pipeline_s);

woody5418, 2019-12-19:
再次唤醒的时候我调用的是
audio_element_set_uri(spiffs_stream_reader, "/spiffs/enwozai_01.mp3");
audio_pipeline_run(pipeline_s);

xiruilin, 2019-12-20:
@woody5418 

删除结束时添加的以下代码：
```
            audio_pipeline_stop(pipeline_s);
            audio_pipeline_wait_for_stop(pipeline_s);
            audio_pipeline_reset_ringbuffer(pipeline_s);
            audio_pipeline_change_state(pipeline_s, AEL_STATE_INIT);
            audio_pipeline_reset_items_state(pipeline_s);
```
再次唤醒时改为：
```
            audio_pipeline_stop(pipeline_s);
            audio_pipeline_wait_for_stop(pipeline_s);
            audio_element_set_uri(spiffs_stream_reader, "/spiffs/enwozai_01.mp3");
            audio_pipeline_run(pipeline_s);
```

升级esp-adf到[commit 40cb926](https://github.com/espressif/esp-adf/commit/40cb9267b9178a954dcccf7ceeae55cbb44ec2a1), 最新版本做了如下调整：
```
@@ -412,7 +421,6 @@ esp_err_t audio_pipeline_stop(audio_pipeline_handle_t pipeline)
            }
        }
    }
-    audio_pipeline_change_state(pipeline, AEL_STATE_INIT);
    return ESP_OK;
}
```

woody5418, 2019-12-23:
更新到最新版本的ADF以后原来可以运行的工程出现问题
E (5579) STAGEFRIGHTMP3_DECODER: Encountered error reading when MP3 init        
E (5579) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN    
之前可以播放的MP3文件打印错误

xiruilin, 2019-12-23:
@woody5418 

上面再次唤醒时的代码做些修改：

```
            audio_pipeline_stop(pipeline_s);
            audio_pipeline_wait_for_stop(pipeline_s);
            audio_event_iface_discard(evt);
            audio_element_set_uri(spiffs_stream_reader, "/spiffs/enwozai_01.mp3");
            audio_pipeline_run(pipeline_s);
```

woody5418, 2019-12-25:
audio_event_iface_discard(evt); 以后貌似就不会再监听事件了。我修改了我的代码。
我想知道我的监听设置为0“audio_event_iface_listen(evt, &msg, 0);”的时候，我的代码一直进入这个if，我如何不让他再进来这个if，并且，我再次播放的时候，直接提示“STAGEFRIGHTMP3_DECODER: output aborted -3 ”。
if (msg.source_type == AUDIO_ELEMENT_TYPE_ELEMENT &&
                msg.source == (void *)i2s_stream_writer_play &&
                msg.cmd == AEL_MSG_CMD_REPORT_STATUS &&
                (((int)msg.data == AEL_STATUS_STATE_STOPPED) ||
                 ((int)msg.data == AEL_STATUS_STATE_FINISHED))) {
                ESP_LOGW(TAG, "[ * ] Stop event ......");
                audio_pipeline_stop(pipeline_play);
                audio_pipeline_wait_for_stop(pipeline_play);
            }

woody5418, 2019-12-25:
while部分代码如下
while (1) {
        audio_event_iface_msg_t msg;
        audio_event_iface_listen(evt, &msg, 0);
        raw_stream_read(raw_read_asr, (char *)buff,
                        audio_chunksize * sizeof(short));
        int keyword = wakenet->detect(model_data, (int16_t *)buff);
        switch (keyword) {
            case 1: {
                ESP_LOGI(TAG, "Wake up");
                audio_pipeline_stop(pipeline_asr);
                audio_pipeline_wait_for_stop(pipeline_asr);
                ESP_LOGI(TAG,
                         "[ mp3 ]The path Settings --->'/spiffs/enwozai.mp3' ");
                audio_element_set_uri(spiffs_stream_reader_play,
                                      "/spiffs/enwozai.mp3");
                audio_pipeline_run(pipeline_play);  // error
                choose_type_flag = CHOOSE_STREAM_PLAY;
                break;
            }
        }

        if (msg.source_type == AUDIO_ELEMENT_TYPE_ELEMENT &&
            msg.source == (void *)mp3_decoder_play &&
            msg.cmd == AEL_MSG_CMD_REPORT_MUSIC_INFO) {
            audio_element_info_t music_info_s = {0};
            audio_element_getinfo(mp3_decoder_play, &music_info_s);

            ESP_LOGI(
                TAG,
                "[ * ] Receive music info from mp3 decoder, sample_rates=%d, "
                "bits=%d, ch=%d",
                music_info_s.sample_rates, music_info_s.bits,
                music_info_s.channels);

            audio_element_setinfo(i2s_stream_writer_play, &music_info_s);
            i2s_stream_set_clk(i2s_stream_writer_play,
                               music_info_s.sample_rates, music_info_s.bits,
                               music_info_s.channels);
            continue;
        }
        /* Stop when the last pipeline element (i2s_stream_writer_http_mp3 in
         * this case) receives stop event */
        if (choose_type_flag == CHOOSE_STREAM_PLAY) {
            if (msg.source_type == AUDIO_ELEMENT_TYPE_ELEMENT &&
                msg.source == (void *)i2s_stream_writer_play &&
                msg.cmd == AEL_MSG_CMD_REPORT_STATUS &&
                (((int)msg.data == AEL_STATUS_STATE_STOPPED) ||
                 ((int)msg.data == AEL_STATUS_STATE_FINISHED))) {
                ESP_LOGW(TAG, "[ * ] Stop event ......");
                audio_pipeline_stop(pipeline_play);
                audio_pipeline_wait_for_stop(pipeline_play);
                audio_pipeline_run(pipeline_asr);
                choose_type_flag = CHOOSE_STREAM_ASR;
            }
        }
    }
运行结果为：
第一次唤醒以后打印的LOGO 并且播放了音频文件，
I (3990) < app >: Wake up                                                                                                                                                                              
W (3990) AUDIO_ELEMENT: IN-[filter] AEL_IO_ABORT                                                                                                                                                       
W (3990) AUDIO_ELEMENT: [filter] Element already stopped                                                                                                                                               
W (3990) AUDIO_PIPELINE: audio_pipeline state is not running, 1                                                                                                                                        
I (4000) < app >: [ mp3 ]The path Settings --->'/spiffs/enwozai.mp3'                                                                                                                                   
W (4020) SPIFFS_STREAM: No more data, ret:0                                                                                                                                                            
I (4070) < app >: [ * ] Receive music info from mp3 decoder, sample_rates=16000, bits=16, ch=1                                                                                                         
W (6060) < app >: [ * ] Stop event ......                                                                                                                                                              
W (6060) AUDIO_ELEMENT: [spiffs] Element already stopped                                                                                                                                               
W (6060) AUDIO_ELEMENT: [mp3] Element already stopped                                                                                                                                                  
W (6060) AUDIO_ELEMENT: [i2s] Element already stopped                                                                                                                                                  
W (6070) AUDIO_PIPELINE: audio_pipeline state is not running, 1  

第二次的唤醒时候的LOGO，并且没有音频播放。
W (8570) AUDIO_ELEMENT: IN-[filter] AEL_IO_ABORT                                                                                                                                                       
W (8570) AUDIO_ELEMENT: [filter] Element already stopped                                                                                                                                               
W (8570) AUDIO_PIPELINE: audio_pipeline state is not running, 1                                                                                                                                        
[... cut ...]
W (8590) AUDIO_ELEMENT: [spiffs] Element already stopped                                                                                                                                               
W (8590) AUDIO_ELEMENT: [mp3] Element already stopped                                                                                                                                                  
W (8600) AUDIO_ELEMENT: [i2s] Element already stopped                                                                                                                                                  
W (8610) AUDIO_PIPELINE: audio_pipeline state is not running, 1

shootao, 2020-01-17:
在 `PLAY` 中代码如下试一下
```c
            audio_element_reset_state(http_stream_reader);
            audio_element_reset_state(i2s_stream_writer);
            audio_element_reset_state(mp3_decoder);
            audio_pipeline_reset_items_state(pipeline);
            audio_pipeline_reset_ringbuffer(pipeline);
            audio_element_set_uri(http_stream_reader, "https://dl.espressif.com/dl/audio/gs-16b-2c-44100hz.mp3");

            audio_pipeline_run(pipeline);
            audio_pipeline_resume(pipeline);
```

woody5418, 2020-01-17:
嗯嗯好的 我试一下这个

shootao, 2020-02-17:
@woody5418 问题解决了吗?

woody5418, 2020-02-19:
> @woody5418 问题解决了吗?

已经解决了 目前没有什么问题了非常感谢

closed 2021-02-23 by jason-mao