espressif/esp-adf #292: Error while Compiling Handsfree Profile code.

Tarun-Narain, 2019-12-22:
Just enables handsfree in pipeline_bt_sink_example

FAILED: play_bt_music.elf
cmd.exe /C "cd . && C:\Users\IDF3\.espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls  -nostdlib @CMakeFiles\play_bt_music.elf.rsp  -o play_bt_music.elf  && cd ."
`esp-idf/bt/libbt.a(btc_task.c.obj):(.rodata.profile_tab+0x78): undefined reference to 
 btc_hf_client_call_handler`
esp-idf/bt/libbt.a(btc_task.c.obj):(.rodata.profile_tab+0x7c): undefined reference to btc_hf_client_cb_handler'
esp-idf/bt/libbt.a(btc_dm.c.obj):(.literal.btc_in_execute_service_request+0xc): undefined reference to btc_hf_client_execute_service'
esp-idf/bt/libbt.a(btc_dm.c.obj): In function `btc_in_execute_service_request':
D:/IDF/esp-idf-v3.3/esp-idf-v3.3/components/bt/bluedroid/btc/core/btc_dm.c:514: undefined reference to `btc_hf_client_execute_service'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

esp-xiaodong, 2019-12-30:
How did you enable HFP? Could you give me your demo，or you can refer to "example/get-started/pipeline_a2dp_and_hfp" code.

Tarun-Narain, 2019-12-30:
Hi,
Same error on pipeline_a2dp_and_hfp, i enabled Handsfree from menuconfig>component>bluetooth the problem just comes in ADF codes. Same error when i enable handsfree in player/pipeline_btsink example. im using idf 3.3

esp-xiaodong, 2019-12-30:
hi，when you get adf code,"git pull" and then "git submodule update --init --incursive". Then enter IDF, "git submodule update --init --recursive". I guess that you didn't get the right lib.

closed 2020-02-21 by jason-mao

Tarun-Narain, 2020-03-20:
Anybody else if having the problem, please make sure you have IDF version 3.3.1.