espressif/esp-adf #294: E (121204) SIP_RTP: receive task, server addr=192.168.1.100:4002

niurenyige, 2019-12-25:
board:lyraT 4.3
VOIP-APP 没做任何修改
接听的时候产生错误。
LyraT SPEAKER 有声音出来，但是LyraT 的MIC好像没工作，对方听不到声音。

 (116754) SIP: [1970-01-01/00:01:28]=======================>>
W (116764) SIP: CHANGE STATE FROM 2, TO 16, :func: _sip_uas_process_req_invite:722
I (117874) VOIP_EXAMPLE: ringing... RemotePhoneNum 1001
I (118874) VOIP_EXAMPLE: ringing... RemotePhoneNum 1001
I (119874) VOIP_EXAMPLE: ringing... RemotePhoneNum 1001
I (120214) VOIP_EXAMPLE: [ * ] input key id is 1
I (120214) VOIP_EXAMPLE: [ * ] [Play] input key event
I (120274) VOIP_EXAMPLE: SIP_EVENT_REQUEST_NETWORK_IP
W (120274) SIP: [1970-01-01/00:02:00]=======WRITE 0587 bytes>>
I (120274) SIP:

SIP/2.0 200 OK
Via: SIP/2.0/UDP 192.168.1.100;rport;branch=z9hG4bKXD30vQv0Xpptg
Contact: <sip:1008@192.168.1.15:5060>
From: "Extension 1001" <sip:1001@192.168.1.100>;tag=6e3B4gZHHt45Q
To: <sip:1008@192.168.1.15:5060>;tag=-2012128518
Call-ID: a655cdcb-a192-1238-cb9c-f74c53e99b3b
CSeq: 14077307 INVITE
Server: ESP32 SIP/2.0
Allow: ACK, INVITE, BYE, CANCEL, OPTIONS, INFO
Content-Length: 154
Content-Type: application/sdp

v=0
o=Z 0 3 IN IP4 192.168.1.15
s=Z
c=IN IP4 192.168.1.15
t=0 0
m=audio 9999 RTP/AVP 8 101
a=rtpmap:8 PCMA/8000
a=rtpmap:101 telephone-event/8000

W (120324) SIP: [1970-01-01/00:02:00]=======================>>
I (120934) VOIP_EXAMPLE: ringing... RemotePhoneNum 1001
W (121144) SIP: [1970-01-01/00:02:00]<<=====READ 0374 bytes==
I (121144) SIP:

ACK sip:1008@192.168.1.15:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.1.100;rport;branch=z9hG4bKypvSyjD4tZcDc
Max-Forwards: 70
From: "Extension 1001" <sip:1001@192.168.1.100>;tag=6e3B4gZHHt45Q
To: <sip:1008@192.168.1.15:5060>;tag=-2012128518
Call-ID: a655cdcb-a192-1238-cb9c-f74c53e99b3b
CSeq: 14077307 ACK
Contact: <sip:mod_sofia@192.168.1.100:5060>
Content-Length: 0

W (121174) SIP: [1970-01-01/00:02:00]<<======================
I (121184) SIP_RTP: rtp local addr=192.168.1.15, port=9999
W (121194) SIP: CHANGE STATE FROM 16, TO 32, :func: sip_uas_process_req:774
E (121204) SIP_RTP: receive task, server addr=192.168.1.100:4002
I (121204) VOIP_EXAMPLE: SIP_EVENT_AUDIO_SESSION_BEGIN

ahhfzhang, 2020-01-06:
@niurenyige  
SIP_RTP 的 这行log并不是个错误, _sip_event_handler 里面的SIP_EVENT_READ_AUDIO_DATA 有一直在编码callback回去吗?  是不是被哪里卡住了.

RogerLaw, 2020-04-19:
same problem here. FreePBX 15.0.16.49
I am sure SIP_EVENT_READ_AUDIO_DATA callback is called.