espressif/esp-adf #296: Cannot specify link libraries for target "esp-sr" which is not built by  this project

flfq, 2019-12-26:
`git clone --recursive https://github.com/espressif/esp-adf.git
cd esp-adf/examples/get-started/play_mp3
make menuconfig`

`  /home/test/esp/esp-adf/esp-idf/tools/cmake/idf_functions.cmake:214 (crosstool_version_check)
  /home/test/esp/esp-adf/esp-idf/tools/cmake/idf_functions.cmake:373 (idf_verify_environment)
  /home/test/esp/esp-adf/esp-idf/tools/cmake/project.cmake:173 (idf_import_components)
  CMakeLists.txt:8 (project)

-- IDF_VER: v3.3.1
-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Found PythonInterp: /home/test/.pyenv/shims/python (found version "2.7.17")
-- Found Perl: /usr/bin/perl (found version "5.28.1")
-- Adding linker script /home/test/esp/play_mp3/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/test/esp/esp-adf/esp-idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script /home/test/esp/esp-adf/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/test/esp/esp-adf/esp-idf/components/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/test/esp/esp-adf/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
-- Current board name is CONFIG_ESP_LYRAT_V4_3_BOARD
CMake Error at /home/test/esp/esp-adf/components/esp-sr/CMakeLists.txt:15 (target_link_libraries):
  Cannot specify link libraries for target "esp-sr" which is not built by
  this project.

-- Configuring incomplete, errors occurred!
See also "/home/test/esp/play_mp3/build/CMakeFiles/CMakeOutput.log".
See also "/home/test/esp/play_mp3/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1`

jason-mao, 2019-12-27:
@flfq 谢谢你的反馈。你可以先到esp-sr目录下，拉最新的master，暂时修复这个问题。

jason-mao, 2019-12-27:
Fixed on cec2c10

closed 2019-12-27 by jason-mao