espressif/esp-adf #297: HTTP streaming is jerky

timotoots, 2019-12-27:
Board: LyraT
Example: Console example

Streaming from http is very jerky over wifi. It could be slow connection, but the stream plays well on a laptop. Is there a way to make the http stream buffer larger initially?

vikramdattu, 2019-12-28:
@timotoots using menuconfig, please set CPU frequency to 240MHz and SPI config to 80MHz and check again.

boarchuz, 2020-03-20:
I'm experiencing this too. It's apparent even with the play_http_mp3_example example. I've narrowed the issue down to the ringbuffer implementation though there's a lot to comprehend in one go. (I'm using all default configs for audio elements, and using recommended settings such as 240MHz CPU, 80MHz QIO, etc.)

The key problem is that the i2s element does not release the ringbuffer write lock until the entire buffer is exhausted. This means that there is often a noticeable hiccup in playback because the i2s stream has no data until the http element can load more from the server and begin refilling the buffer. Of course, all it takes is a few tens of milliseconds.

The http element is stuck waiting here: https://github.com/espressif/esp-adf/blob/60210eeeb892a148002de5d3567c46f8c22f9f89/components/audio_pipeline/ringbuf.c#L272

Sometimes, if the latency is low enough, I can't notice the hiccup audibly but I can see from the log that this behaviour is consistent for the entire stream. 
I'm sure this is not the intended behaviour as it makes for terrible audio playback and negates most of the benefits of buffering at all. The i2s and http elements should be coordinating so that the buffer is continuously topped up (eg. every 1KB or so). 

It's important that anyone from espressif using the play_http_mp3_example modifies the mp3 url to a server with high latency (eg. geographically distant) to create frequent/reliable pauses. The default url (an mp3 on espressif servers) may be such low latency that you won't detect an issue and think we're all doing something silly.

jason-mao, 2020-03-20:
@timotoots @spbkaizo Would you please provide your URL at here?

boarchuz, 2020-03-22:
I'm a little less sure now if it's the http-decoder or decoder-i2s (or both) where it's exhausting the buffer, more more certain that that's the fundamental problem. I put a bunch of debug logging in and can see the ogg decoder in my case running right through the buffer while the http element is waiting to write more; only when it's empty does the http element get a chance to begin refilling it, and what's worse is that the ogg decoder seems to wait until the buffer is completely full before it begins outputting more to i2s. 

These reproduce the problem 100% consistently for me, the hiccup is apparent every few seconds like clockwork:
MP3: https://file-examples.com/wp-content/uploads/2017/11/file_example_MP3_1MG.mp3
OGG: https://file-examples.com/wp-content/uploads/2017/11/file_example_OOG_1MG.ogg

jlunz, 2020-03-25:
On LyraT-Mini I am also experiencing this with https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_living_stream
I changed from sample url to mp3, both have small stops during playback.

maizezoidberg, 2020-11-09:
Anybody solve this problem? 
I am in Moscow, if I upload an mp3 or aac file to a server in Russia - everything works, without delays or interruptions. If I upload a file to a server in California, there are constant delays and interruptions during playback.

huybk213, 2020-12-03:
I have the same problem with http stream.
URL http://us4.internet-radio.com:8266/stream

GerryBriggs, 2021-02-16:
I am having the same problem.  To boarchuz's point that "the key problem is that the i2s element does not release the ringbuffer write lock until the entire buffer is exhausted" - well, if that's how they implemented a ring buffer then they have no idea of the fundamental principle of a ring buffer, which is to fill up behind the read pointer as much as possible but never go past the read pointer. That's literally the purpose of a ring buffer.  I used to build them in assembly language and you would have a write pointer and a read pointer. In this case the read pointer would be the output to I2S but the write pointer would be the input from the http download over wifi. The packet sizes would have to be in reasonably small chunks so that you only fill maybe up to 10% more of the ring buffer every time you download one. And for MP3, I think it takes chunks of 4096 bytes to decode before sending to I2S

huybk213, 2021-02-17:
@GerryBriggs How can I improve that? Could you please give me your suggestion?
Thanks

GerryBriggs, 2021-02-17:
> @GerryBriggs How can I improve that? Could you please give me your suggestion?
> Thanks

Well I havent got around to understanding the whole system yet, but there would have to be a pointer variable that is able to pass between the functionality of the data reader (from the web service) and the data writer (out to the mp3 / aac decoder). Then wherever the interrupt is that is triggered when the decoder asks for more data from the ring buffer to decode, a global flag would have to be set when it sees that the buffer has become ENOUGH partially empty that it can load in another packet from the internet source through the wifi. At that time a flag would be set back to true and then the primary code handler would, upon seeing that flag is true, go load another packet from the internet into the read pointer of the buffer (and then reset that flag to false).  I only started with ESP and LYRA-T yesterday and have got the examples to work so far (except the "living stream" which apparently requires a patch which I have but dont know what to do with).  But once I get into that ring buffer code hopefully in the next few weeks I will try to make it do what it is supposed to do. But Im from the assembly language world not high level languages so the interrupts and kernel may be to obfuscated by the code for me to do anything useful.

boarchuz, 2021-02-17:
@GerryBriggs I'm sure the underlying FreeRTOS ring buffer is rock solid, and I had trouble identifying the specific problem so please don't take my above comment as conclusive. It was difficult because this is a latency issue (which any debugging will significantly contribute to) and I'm not very familiar with ADF codebase.
Following this issue though so I can resume the project if anyone is able to make progress.

GerryBriggs, 2021-02-17:
In the old days we would call the problem a "buffer underrun error" because you gotta keep that ring buffer as full as possible without going over and if it falls way behind then eventually the read pointer catches up with the write pointer and that is when it gets catastrophic.  You really would want at least a three or four second buffer for music streaming over the web one-way. It's not two-way communications so the latency isnt that important. That way if there is packet delay up to three seconds on the wifi or over the web, there is no problem because the buffer is big enough to be still emptying out its four seconds worth of stored up data

huybk213, 2021-02-17:
Thanks @GerryBriggs

GerryBriggs, 2021-02-22:
My observations so far - there are a lot of streams that I find work perfectly and no hiccups.  Im using a jazz stream on mp3 that hasnt missed a single packet in five hours of play at 128 kbps :  http://east-mp3-128.streamthejazzgroove.com/stream.mp3

I had to modify the code of the pipeline_living_stream to switch from AAC over to MP3 stream but it just works really well.  FOr the other stream that was intermittent, I found that increasing various aspects of the ring buffers didnt help - and there are multiple ring buffers depending on the stage through the pipeline (which one would expect).  The best example to demonstrate the change was using the pipeline_http_stream which is not a live stream but a static MP3 file downloaded over the internet.  The default in the example is choppy ( https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.mp3 ) but when I changed to a different web-hosted source mp3 file ( http://www.amclassical.com/mp3/amclassical_bon_sonata_in_g_minor_mvt_2.mp3 ) then everything was fine.  As for the choppy MP3 that is the default in the example, increasing the size of the ring buffer in http_stream.h from its default 20 * 1024 bytes does not improve the choppiness, but bringing it down under 10 * 1024 does in fact start to make it worse, and at only 2 * 1024 it is much much worse.  I tried to make it as large as 120 * 1024 bytes and beyond this size I ran out of memory (so the example left about 100 KB of headroom to play around with).  I also tried messing around with another ring buffer - I think the I2S buffer - but again, there was no benefit. 
A lot of the quality problem seems to have to do with the quality of the source stream itself (or the connection quality of the static hosted mp3 file)

huybk213, 2021-02-22:
@GerryBriggs  I dont think problems is source stream server.
I'm from Vietnam, server is Amazone WebService S3, region is US.
When I am streaming file with http, file mp3 320kpbs,  streaming was jerky (but VLC software streaming was smooth at the same time esp32 playing).
But when i change server stream into my local host in Vietnam, streaming was smooth.

GerryBriggs, 2021-02-22:
OK I see what you mean.  My Win10 web browser also plays that stream just fine.  What does the more powerful machine do that the esp32 firmware currently does not, in order to keep the stream from stuttering?  May have to do with disordered packets ... or tiny packets that take less time to process on the large machine especially when they are unevenly sized

jason-mao, 2021-02-23:
Hi guys, In my opinion, the jerk or hiccups issue have follow reasons.
- The element buffer or I2S DMA buffer to small. Suggest to increase the buffer size.
- The element task  too heavy to go smoothly, such as all elements task on single core. Suggest balance the CPU loading.
- The stream server or Wi-Fi router have a lot of delay, the mean is that the download speed is too slowly. Suggest to increase the Wi-Fi, TCP and LwIP buffer. This is why the same stream good on PC, it's have a lot of memory to buffering. I provide some configurations as follow to improve the Wi-Fi，TCP and LwIP. BTW, these configurations based on PSRAM WROVER module, if the no PSRAM, it can be reduced as appropriate. 
```c
# 
# Wi-Fi
# 
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=512
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER=y
CONFIG_ESP32_WIFI_TX_BUFFER_TYPE=0
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM=8
CONFIG_ESP32_WIFI_CACHE_TX_BUFFER_NUM=32
CONFIG_ESP32_WIFI_CSI_ENABLED=
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP32_WIFI_TX_BA_WIN=16
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=32

#
# LWIP
# 
CONFIG_TCPIP_RECVMBOX_SIZE=512

#
# TCP
#
CONFIG_LWIP_MAX_ACTIVE_TCP=16
CONFIG_LWIP_MAX_LISTENING_TCP=16
CONFIG_TCP_MAXRTX=12
CONFIG_TCP_SYNMAXRTX=6
CONFIG_TCP_MSS=1436
CONFIG_TCP_MSL=60000
CONFIG_TCP_SND_BUF_DEFAULT=65535
CONFIG_TCP_WND_DEFAULT=512000
CONFIG_TCP_RECVMBOX_SIZE=512
CONFIG_TCP_QUEUE_OOSEQ=y
CONFIG_ESP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES=
CONFIG_TCP_OVERSIZE_MSS=y
CONFIG_TCP_OVERSIZE_QUARTER_MSS=
CONFIG_TCP_OVERSIZE_DISABLE=
CONFIG_LWIP_WND_SCALE=y
CONFIG_TCP_RCV_SCALE=3
````

huybk213, 2021-02-23:
Tks @jason-mao

GerryBriggs, 2021-02-23:
thanks Jason.  I will try this too

closed 2022-02-08 by jason-mao

abdoutech93, 2023-02-24:
I getting maye be the same error #955 
with hls aac stream encoded using ffmpeg with the folowing params 
`-thread_queue_size 1024 -f alsa -ac 1 -i hw:1,0  -strict -2 -c:a aac -b:a 128k   -ar 44100 -r 30 -g 8 -keyint_min 60 -b:v 2500k  -preset medium -bufsize 32k -maxrate 2500k -f flv  -flvflags no_duration_filesize`