espressif/esp-adf #298: ".pls" stream can't play

xiruilin, 2019-12-29:
Hi,
We want to play ".pls" stream, but got failed, below is the logs:

```
[0;32mI (1641645) DCS_AUDIO: Playing audio, offset: 0 url: http://playerservices.streamtheworld.com/pls/WMGKFMAACIHR.pls?at=1&fb_broadcast=false&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&callLetters=WMGK-FM&streamid=5297&clienttype=amazon&iheartradioversion=1.0&osversion=1.0&devicename=Indigo&terminalid=304&init_id=8169&profileid=1283982799[0m
[AudioPlayer - PlaybackStarted] Sending 414 bytes(buffer size: 16384)
[0;32mI (1641671) DISPATCHER: EXE IN, cmd type:2, index:4001, data:0x3f810f24, len:12[0m
[0;32mI (1641685) DCS_WRAPPER: duer_dcs_action_audio_play, 0x3f810f24, 8⸮?, 12[0m
[0;32mI (1641694) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe1704[0m
[0;32mI (1641701) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:540, MEM Total:3889588 Bytes, Inter:95556 Bytes, Dram:76292 Bytes

[0m
[0;32mI (1641712) ESP_AUDIO_TASK: It's a decoder[0m
[0;32mI (1641717) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[DEC_pls],RESAMPLE:[48000],OUT:[OUT_iis][0m
[0;32mI (1641726) ESP_AUDIO_TASK: 2.Handles,IN:0x3f80d2a4,CODEC:0x0,FILTER:0x3f8202d0,OUT:0x3f80dda4[0m
[0;31mE (1641735) ESP_AUDIO_TASK: DEC_pls does not supported[0m
[0;32mI (1641741) DUEROS_CPP: ESP_AUDIO_CALLBACK_FUNC, st: 5, err: 81003, src: 801[0m
[0;31mE (1641749) DUEROS_CPP: duer_dcs_audio_play_failed[0m
Duer dcs: got audio failed
```

How to solve it?
Thanks!

jason-mao, 2019-12-30:
@xiruilin What's the exact music data format?

xiruilin, 2019-12-30:
@jason-mao 
Contents of the WMGKFMAACIHR.pls:

```
[playlist]
File1=http://17413.live.streamtheworld.com:80/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File2=http://17413.live.streamtheworld.com:3690/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File3=http://17433.live.streamtheworld.com:80/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File4=http://17433.live.streamtheworld.com:3690/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File5=http://20593.live.streamtheworld.com:80/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File6=http://20593.live.streamtheworld.com:3690/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File7=http://13683.live.streamtheworld.com:80/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File8=http://13683.live.streamtheworld.com:3690/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File9=http://16963.live.streamtheworld.com:80/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File10=http://16963.live.streamtheworld.com:3690/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File11=http://15233.live.streamtheworld.com:80/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File12=http://15233.live.streamtheworld.com:3690/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File13=http://18063.live.streamtheworld.com:80/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File14=http://18063.live.streamtheworld.com:3690/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File15=http://17003.live.streamtheworld.com:80/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File16=http://17003.live.streamtheworld.com:3690/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File17=http://19003.live.streamtheworld.com:80/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
File18=http://19003.live.streamtheworld.com:3690/WMGKFMAACIHR_SC?callLetters=WMGK-FM&clienttype=amazon&streamid=5297&fb_broadcast=false&terminalid=304&init_id=8169&deviceid=amzn1.account.AHBVPGOGCVRV5AKKIRPTOZTBHUIA&at=1&profileid=1283982799&devicename=Indigo&osversion=1.0&iheartradioversion=1.0
Title1=WMGKFMAACIHR_SC
Title2=WMGKFMAACIHR_SC-Bak
Length1=-1
NumberOfEntries=18
Version=2```

vikramdattu, 2019-12-30:
@xiruilin we are in a process to add `pls` support. This should be added in couple of days.

xiruilin, 2019-12-30:
@vikramdattu 
Thanks

closed 2019-12-30 by xiruilin