espressif/esp-adf #299: pipeline_passthru example does not actually pass audio through the ESP32 for ES8388

Jeija, 2019-12-29:
My understanding from looking at the [passthru.c code from the passthru example](https://github.com/espressif/esp-adf/blob/master/examples/audio_processing/pipeline_passthru/main/passthru.c) is that this example is supposed to read the I²S data from the codec chip's ADC input and then write back to it's DAC output in real time. However, it seems that this is not actually what is happening. I have tested this on my ESP32-LyraT board with an ES8388 codec, but this issue might possibly also affect other boards.

Instead, the passthru code just enables the internal analog bypass mode of the ES8388. This is how it happens:

* The line `audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_LINE_IN, AUDIO_HAL_CTRL_START);` in `passthru.c` calls `audio_hal_ctrl_codec` in `audio_hal.c`, which calls `audio_hal->audio_codec_ctrl`, which points to the function `es8388_ctrl_state` in the case of a board with an ES8388 codec.
* Since the specified `audio_hal_codec_mode_t` was `AUDIO_HAL_CODEC_MODE_LINE_IN`, the function `es8388_ctrl_state` calls `es8388_start` with parameter `ES_MODULE_LINE`
* In `es8388_start`, the following configuration is written via I2C

https://github.com/espressif/esp-adf/blob/3438db90b6f66fd0cd6a52accc68619d12bd72db/components/audio_hal/driver/es8388/es8388.c#L175-L180

Note that `ES8388_DACCONTROL17` and `ES8388_DACCONTROL20` are set to `0x50`. When looking at the ES8388 datasheet, you will find that this configuration disables the "DAC to mixer" connection and enables the "LIN / RIN to mixer" connection. So the ADC and DAC are completely bypassed.

The original passthru example worked just fine for me (on my ESP32-LyraT it takes the microphone input, reads it via I²S, writes it back via I²S and outputs it through the headphone jack), I believe that it was this commit that broke it: https://github.com/espressif/esp-adf/commit/e05f405c61b73c9389241c1fec2c33009b628c87

Can you reproduce this problem?

I think a quick fix would be to just revert https://github.com/espressif/esp-adf/commit/e05f405c61b73c9389241c1fec2c33009b628c87.

espressif-zhanghu, 2020-01-08:
Hi Jeija
    Your discovery is great. Using audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_LINE_IN, AUDIO_HAL_CTRL_START)  is more suitable for [readme](https://github.com/espressif/esp-adf/blob/master/examples/audio_processing/pipeline_passthru/README.md) ，That's why we have this [commit](https://github.com/espressif/esp-adf/commit/e05f405c61b73c9389241c1fec2c33009b628c87).

jimruxton, 2020-01-26:
I don't think this issue has been fixed yet. This [commit](https://github.com/espressif/esp-adf/commit/e05f405c61b73c9389241c1fec2c33009b628c87) does as Jeija suggested above. It routes the audio around the ADC and DAC.  Looking at the registers of the ES8388 after the command ::
audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_LINE_IN, AUDIO_HAL_CTRL_START)
there are a few registers that are incorrectly set if the intent is to have AUX IN pass through ADC DAC to HEADPHONES. For example Register 10 – ADC Control 2, sets the input to ADC to LINE1 after this command is executed. AUX IN is on LINE2. Adding the following lines of code after this command sets the registers properly to have audio pass through the whole system as it did before the  [commit](https://github.com/espressif/esp-adf/commit/e05f405c61b73c9389241c1fec2c33009b628c87)
```
	es8388_write_reg(39, 144); // left DAC to left mixer enable
	es8388_write_reg(42, 144);// right DAC to right mixer enable
	es8388_write_reg(10, 80); // set LINE2 as input to ADC
	es8388_write_reg(43, 128); //DACLRC and ADCLRC same
```
es8388.h must be included for this to work.

chf2117, 2021-01-14:
The pipeline_passthru example did not work for me using the LyraT v4.3. Reverting e05f405 fixed the issue for me.

alamaral, 2023-01-09:
Still doesn't work properly a year later...

crosswick, 2024-01-14:
Issue is still present!! Stopping the pipeline immediately after starting it does not make any difference. Audio must be routed through the mixer of the codec, still. Please fix this.

shamanlim, 2024-10-09:
I use adf v2.4.1 (ESP_AUDIO-v1.7.1), and this sample did not work. Until I switch to adf v2.4(ESP_AUDIO-v1.7.0-63)