espressif/esp-adf #300: "ESP_AUDIO_CTRL: Play activity timeout" and "ESP_AUDIO_CTRL: Timeout(8s) on stop"

xiruilin, 2019-12-29:
Hi,
In the description of the "esp_audio.h", timeout is 8000ms,
but it was tested to be 800ms, so timeout occurs frequently, 
below is the logs:

play timeout (857ms):
```
[0;32mI (914307) DISPATCHER: EXE IN, cmd type:2, index:4001, data:0x3f810f14, len:12[0m
[0;32mI (914326) DCS_WRAPPER: duer_dcs_action_audio_play, 0x3f810f14, ~⸮?, 12[0m
[0;33mW (914337) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'[0m
[0;32mI (914345) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe1794[0m
[0;32mI (914353) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:540, MEM Total:3890596 Bytes, Inter:97160 Bytes, Dram:77896 Bytes[0m
[0;32mI (914364) ESP_AUDIO_TASK: It's a decoder[0m
[0;32mI (914369) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[NULL],RESAMPLE:[48000],OUT:[OUT_iis][0m
[0;32mI (914378) AUDIO_ELEMENT: [IN_http] Element task created[0m
[0;33mW (914383) ESP_AUDIO_TASK: Try to got muisc format by in stream[0m
[0;32mI (914383) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1[0m
[0;33mW (915164) ESP_AUDIO_CTRL: Play activity timeout[0m
[0;33mW (915164) AUDIO_PIPELINE: audio_pipeline state is not running, 1[0m
[0;33mW (915165) AUDIO_PIPELINE: audio_pipeline state is not running, 1[0m
[0;32mI (915171) DISPATCHER: EXE OUT,result type:2, index:4001, ret:8100b, data:0x0, len:0[0m
```

stop timeout (821ms):
```
[0;32mI (252734) DISPATCHER: EXE IN, cmd type:2, index:1004, data:0x0, len:0[0m
[0;32mI (252741) DCS_WRAPPER: duer_dcs_action_audio_stop[0m
[0;32mI (252747) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5, que:0x3ffe16e4[0m
[0;32mI (252755) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP, status:2[0m
[0;33mW (252761) AUDIO_PIPELINE: audio_element_stop has paused element[0m
[0;33mW (252767) AUDIO_ELEMENT: IN-[resample] AEL_IO_ABORT[0m
[0;33mW (252767) AUDIO_ELEMENT: IN-[DEC_mp3] AEL_IO_ABORT[0m
[0;32mI (252774) RSP_FILTER: rsp_filter_close[0m
[0;32mI (252779) STAGEFRIGHTMP3_DECODER: Closed[0m
[0;32mI (252784) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_PAUSE[0m
[0;33mW (253555) ESP_AUDIO_CTRL: Timeout(8s) on stop[0m
[0;32mI (253555) DISPATCHER: EXE OUT,result type:2, index:1004, ret:8100b, data:0x0, len:0[0m
```

resume timeout (841ms):
```
[0;32mI (119503) DISPATCHER: EXE IN, cmd type:2, index:4005, data:0x3f810f14, len:12[0m
[0;32mI (119514) DCS_WRAPPER: duer_dcs_action_audio_resume, 0x3f810f14, ⸮S⸮?46, 12[0m
[0;32mI (119525) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe1794[0m
[0;32mI (119533) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:540, MEM Total:3888904 Bytes, Inter:94596 Bytes, Dram:75332 Bytes[0m
[0;32mI (119544) ESP_AUDIO_TASK: It's a decoder[0m
[0;32mI (119548) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[DEC_mp3],RESAMPLE:[48000],OUT:[OUT_iis][0m
[0;32mI (119558) ESP_AUDIO_TASK: 2.Handles,IN:0x3f80d2a4,CODEC:0x3f80da8c,FILTER:0x3f851860,OUT:0x3f80dda4[0m
[0;33mW (119567) AUDIO_PIPELINE: audio_pipeline state is not running, 1[0m
[0;33mW (119574) AUDIO_PIPELINE: audio_pipeline state is not running, 1[0m
[0;33mW (119581) AUDIO_PIPELINE: audio_pipeline state is not running, 1[0m
[0;33mW (119587) AUDIO_PIPELINE: audio_pipeline state is not running, 1[0m
[0;32mI (119594) AUDIO_ELEMENT: [DEC_mp3] Element task destroyed[0m
[0;32mI (119600) AUDIO_ELEMENT: [OUT_iis] Element task destroyed[0m
[0;32mI (119606) AUDIO_ELEMENT: [resample] Element task destroyed[0m
[0;32mI (119613) AUDIO_PIPELINE: audio_pipeline_unlinked[0m
[0;33mW (119618) AUDIO_PIPELINE: There are no listener registered[0m
[0;33mW (119624) ESP_AUDIO_TASK: Destroy the old pipeline[0m
[0;32mI (119630) ESP_AUDIO_TASK: 2.2 Update all pipeline[0m
[0;32mI (119635) ESP_AUDIO_TASK: 2.3 Linked new pipeline[0m
[0;32mI (119641) AUDIO_PIPELINE: link el->rb, el:0x3f80d2a4, tag:IN_http, rb:0x3f820398[0m
[0;32mI (119649) AUDIO_PIPELINE: link el->rb, el:0x3f80da8c, tag:DEC_mp3, rb:0x3f817c50[0m
[0;32mI (119657) AUDIO_PIPELINE: link el->rb, el:0x3f851860, tag:resample, rb:0x3f817abc[0m
[0;32mI (119665) ESP_AUDIO_TASK: 3. Previous starting...[0m
[0;32mI (119671) AUDIO_ELEMENT: [IN_http] Element task created[0m
[0;32mI (119677) AUDIO_ELEMENT: [DEC_mp3] Element task created[0m
[0;32mI (119682) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (119685) AUDIO_ELEMENT: [resample] Element task created[0m
[0;32mI (119695) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (119695) AUDIO_ELEMENT: [OUT_iis] Element task created[0m
[0;32mI (119707) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (119714) I2S_STREAM: AUDIO_STREAM_WRITER[0m
[0;32mI (119702) RSP_FILTER: sample rate of source data : 24000, channel of source data : 1, sample rate of destination data : 48000, channel of destination data : 2[0m
[0;32mI (119689) STAGEFRIGHTMP3_DECODER: MP3 opened[0m
[0;32mI (119677) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1[0m
[0;33mW (120344) ESP_AUDIO_CTRL: Play activity timeout[0m
[0;33mW (120345) AUDIO_PIPELINE: audio_pipeline state is not running, 1[0m
[0;33mW (120346) AUDIO_PIPELINE: audio_pipeline state is not running, 1[0m
[0;32mI (122217) DISPATCHER: EXE OUT,result type:2, index:4005, ret:8100b, data:0x0, len:0[0m
```

jason-mao, 2019-12-30:
@xiruilin,  What is CONFIG_FREERTOS_HZ ? Could you given more description of the process?  I will reproduced.

xiruilin, 2019-12-30:
@jason-mao 
CONFIG_FREERTOS_HZ is 1000,
thanks

jason-mao, 2019-12-30:
@xiruilin Sure, it must be like that. How to reproduced your issue? could you provide your `sdkconfig` at here?

xiruilin, 2019-12-30:
@jason-mao 
If your CONFIG_FREERTOS_HZ is 100, 8000ms is 800 ticks, 
my CONFIG_FREERTOS_HZ  is 1000, 800 ticks is 800ms.
Is there any way to make timeout irrelevant to CONFIG_FREERTOS_HZ？
Or can you provide a function in the "libesp_audio.a" to set the timeout (ticks, not ms)?

xiruilin, 2019-12-30:
### sdkconfig

```
/*
 * Automatically generated file. DO NOT EDIT.
 * Espressif IoT Development Framework (ESP-IDF) Configuration Header
 */
#pragma once
#define CONFIG_IDF_TARGET_ESP32 1
#define CONFIG_IDF_CMAKE 1
#define CONFIG_IDF_TARGET "esp32"
#define CONFIG_IDF_FIRMWARE_CHIP_ID 0x0000
#define CONFIG_SDK_TOOLPREFIX "xtensa-esp32-elf-"
#define CONFIG_APP_BUILD_TYPE_APP_2NDBOOT 1
#define CONFIG_APP_BUILD_GENERATE_BINARIES 1
#define CONFIG_APP_BUILD_BOOTLOADER 1
#define CONFIG_APP_BUILD_USE_FLASH_SECTIONS 1
#define CONFIG_APP_COMPILE_TIME_DATE 1
#define CONFIG_BOOTLOADER_LOG_LEVEL_NONE 1
#define CONFIG_BOOTLOADER_LOG_LEVEL 0
#define CONFIG_BOOTLOADER_SPI_WP_PIN 7
#define CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V 1
#define CONFIG_BOOTLOADER_WDT_ENABLE 1
#define CONFIG_BOOTLOADER_WDT_TIME_MS 9000
#define CONFIG_BOOTLOADER_RESERVE_RTC_SIZE 0x0
#define CONFIG_ESPTOOLPY_BAUD_OTHER_VAL 115200
#define CONFIG_ESPTOOLPY_FLASHMODE_QIO 1
#define CONFIG_ESPTOOLPY_FLASHMODE "dio"
#define CONFIG_ESPTOOLPY_FLASHFREQ_80M 1
#define CONFIG_ESPTOOLPY_FLASHFREQ "80m"
#define CONFIG_ESPTOOLPY_FLASHSIZE_4MB 1
#define CONFIG_ESPTOOLPY_FLASHSIZE "4MB"
#define CONFIG_ESPTOOLPY_FLASHSIZE_DETECT 1
#define CONFIG_ESPTOOLPY_BEFORE_RESET 1
#define CONFIG_ESPTOOLPY_BEFORE "default_reset"
#define CONFIG_ESPTOOLPY_AFTER_RESET 1
#define CONFIG_ESPTOOLPY_AFTER "hard_reset"
#define CONFIG_ESPTOOLPY_MONITOR_BAUD_115200B 1
#define CONFIG_ESPTOOLPY_MONITOR_BAUD_OTHER_VAL 115200
#define CONFIG_ESPTOOLPY_MONITOR_BAUD 115200
#define CONFIG_PARTITION_TABLE_CUSTOM 1
#define CONFIG_PARTITION_TABLE_CUSTOM_FILENAME "partitions_dueros_example.csv"
#define CONFIG_PARTITION_TABLE_FILENAME "partitions_dueros_example.csv"
#define CONFIG_PARTITION_TABLE_OFFSET 0x8000
#define CONFIG_PARTITION_TABLE_MD5 1
#define CONFIG_ESP_SMARTCONFIG 1
#define CONFIG_LYRIC_ENABLE 1
#define CONFIG_HTMLVIEW_ENABLE 1
#define CONFIG_DIRECTIVE_DEBUG 1
#define CONFIG_ESP_LYRAT_V4_3_BOARD 1
#define CONFIG_REC_ENG_ENABLE_VAD_ONLY 1
#define CONFIG_SR_MODEL_WN5_QUANT 1
#define CONFIG_SR_WN5X3_NIHAOXIAOXIN 1
#define CONFIG_SR_MN1_MODEL_QUANT 1
#define CONFIG_SR_MN1_CHINESE 1
#define CONFIG_SPEECH_COMMANDS_NUM 20
#define CONFIG_CN_SPEECH_COMMAND_ID0 "da kai kong tiao"
#define CONFIG_CN_SPEECH_COMMAND_ID1 "guan bi kong tiao"
#define CONFIG_CN_SPEECH_COMMAND_ID2 "zeng da feng su"
#define CONFIG_CN_SPEECH_COMMAND_ID3 "jian xiao feng su"
#define CONFIG_CN_SPEECH_COMMAND_ID4 "sheng gao yi du"
#define CONFIG_CN_SPEECH_COMMAND_ID5 "jiang di yi du"
#define CONFIG_CN_SPEECH_COMMAND_ID6 "zhi re mo shi"
#define CONFIG_CN_SPEECH_COMMAND_ID7 "zhi leng mo shi"
#define CONFIG_CN_SPEECH_COMMAND_ID8 "song feng mo shi"
#define CONFIG_CN_SPEECH_COMMAND_ID9 "jie neng mo shi"
#define CONFIG_CN_SPEECH_COMMAND_ID10 "guan bi jie neng mo shi"
#define CONFIG_CN_SPEECH_COMMAND_ID11 "chu shi mo shi"
#define CONFIG_CN_SPEECH_COMMAND_ID12 "guan bi chu shi mo shi"
#define CONFIG_CN_SPEECH_COMMAND_ID13 "da kai lan ya"
[... 492 lines cut ...]
#define CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE
#define CONFIG_TASK_WDT CONFIG_ESP_TASK_WDT
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1
#define CONFIG_TASK_WDT_TIMEOUT_S CONFIG_ESP_TASK_WDT_TIMEOUT_S
#define CONFIG_TCPIP_RECVMBOX_SIZE CONFIG_LWIP_TCPIP_RECVMBOX_SIZE
#define CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY CONFIG_LWIP_TCPIP_TASK_AFFINITY_NO_AFFINITY
#define CONFIG_TCPIP_TASK_STACK_SIZE CONFIG_LWIP_TCPIP_TASK_STACK_SIZE
#define CONFIG_TCP_MAXRTX CONFIG_LWIP_TCP_MAXRTX
#define CONFIG_TCP_MSL CONFIG_LWIP_TCP_MSL
#define CONFIG_TCP_MSS CONFIG_LWIP_TCP_MSS
#define CONFIG_TCP_OVERSIZE_MSS CONFIG_LWIP_TCP_OVERSIZE_MSS
#define CONFIG_TCP_QUEUE_OOSEQ CONFIG_LWIP_TCP_QUEUE_OOSEQ
#define CONFIG_TCP_RECVMBOX_SIZE CONFIG_LWIP_TCP_RECVMBOX_SIZE
#define CONFIG_TCP_SND_BUF_DEFAULT CONFIG_LWIP_TCP_SND_BUF_DEFAULT
#define CONFIG_TCP_SYNMAXRTX CONFIG_LWIP_TCP_SYNMAXRTX
#define CONFIG_TCP_WND_DEFAULT CONFIG_LWIP_TCP_WND_DEFAULT
#define CONFIG_TIMER_QUEUE_LENGTH CONFIG_FREERTOS_TIMER_QUEUE_LENGTH
#define CONFIG_TIMER_TASK_PRIORITY CONFIG_FREERTOS_TIMER_TASK_PRIORITY
#define CONFIG_TIMER_TASK_STACK_DEPTH CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH
#define CONFIG_TIMER_TASK_STACK_SIZE CONFIG_ESP_TIMER_TASK_STACK_SIZE
#define CONFIG_TOOLPREFIX CONFIG_SDK_TOOLPREFIX
#define CONFIG_UDP_RECVMBOX_SIZE CONFIG_LWIP_UDP_RECVMBOX_SIZE
#define CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP
```

xiruilin, 2020-01-02:
This problem has been solved in "esp_audio v1.5.1", thanks!

closed 2020-01-02 by xiruilin

xiruilin, 2020-01-20:
@jason-mao 

It's nice to see that commit [d2b0ec6](https://github.com/espressif/esp-adf/commit/d2b0ec607e36ce7a5e8578e18cbc35f47b3e304b): ``Update esp_audio to ver1.5.2`` provides a function
``esp_audio_set_play_timeout`` to set the timeout.
But the definition of this function is not included in the header file ``esp_audio.h``.
Can you update it?
Thanks!

jason-mao, 2020-01-20:
@xiruilin Sorry and thank you mention this. Please refer to: 399f398191f7f553f72343dcb92018da42c25043.