espressif/esp-adf #303: ESP32 wroom32D no sound call VOIP example

Arto1994, 2019-12-30:
Hello everyone !

I'm testing the new adf , to make  a VOIP call. But I could not hear anything from a external client , I already debug and the callbacks of audio coding/decoding , trasmition and reception works , The sip was able to register to the server and handle the call the only error from SIP is  this one 

> **E (284027) SIP_RTP: receive task, server addr=192.168.0.114:4000**

For now I only  using the  received dad and doing the echo to trasmit again ,But I can hear anything. It seems like  an access error .

Also I used wireshark to view the sip protocol and I could see ICMP (unreachable port) error:

[image]

I'm tried with three servers  ,uSipServer Android app  and I install  Asterisk in raspeberry and  last using a Virutal machine with linux in amazon,  But in all of these there is no audio from esp32.

Also Wireshar have a tool to do a telephony (sip) diagram and this is what It captured:

192.168.0.124 is my PC calling the esp 192.168.0.122 trough server 192.1680.114 (android uSipServer )

[image]

As you see there is no audio data coming from esp32  :( , only there is audio from PC to esp32.

Other blocker for my using the adf , is that there is no source code to modify and this error is not clear printed from terminal or  at least I don't know it. 

Thanks for help!

ahhfzhang, 2020-01-07:
@Arto1994 Thanks for your report.
This seems to be a bug in the sip interaction, can you help provide the full log? 
and can you try Yate or Freeswitch server?

Arto1994, 2020-01-07:
Hello Espressif/Esp-Adf

For now I have the log that I put in the bottom of the page , Im using
asterisk and also I tried with uSipServer (app server) and happens the same
no sound , and I received the same error.

Tomorrow I can share the entire log , but meanwhile can you provide me more
error information , since you don't have released the source code  for me
I've been hard debug this  problem .

Other thing that I tried is running this asterisk server in a raspberry pi
as I said ,skipping the possible firewalls issues , but still the same :(

MrZANE42, 2020-03-25:
Any news on this? About to start a project possibly using this code

ahhfzhang, 2020-04-09:
Hi,
We've been testing a lot on the FreePBX recently, bugfix commit: https://github.com/espressif/esp-adf/commit/20b0827
hope you can try again.

many thanks.

closed 2020-10-15 by jason-mao