espressif/esp-adf #305: example DLNA连接网易云音乐播放出错

newstarqu, 2019-12-31:
用example中的DLNA例子直接编译，烧写到lyrat-4.3的开发板上，手机网易云音乐连接播放报错：
STAGEFRIGHTMP3_DECODER： Encountered error reading when MP3 init
这个错误是在库文件中打印的，没有源码无法调试。

jason-mao, 2019-12-31:
@newstarqu 帮把你的log提供一下

newstarqu, 2020-01-05:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6320
ho 0 tail 12 room 4
load:0x40078000,len:10180
load:0x40080400,len:6712
entry 0x40080768
I (31) boot: ESP-IDF v3.2-103-g4aa1058e8-dirty 2nd stage bootloader
I (31) boot: compile time 22:00:39
I (31) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00200000
I (82) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x37e24 (228900) map
I (176) esp_image: segment 1: paddr=0x00047e4c vaddr=0x3ffb0000 size=0x03160 ( 12640) load
I (181) esp_image: segment 2: paddr=0x0004afb4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /mnt/f/work/esp/esp32/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1779

I (183) esp_image: segment 3: paddr=0x0004b3bc vaddr=0x40080400 size=0x04c54 ( 19540) load
I (199) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size0x400d0018: _flash_cache_start at ??:?

=0xac344 (705348) map
I (447) esp_image: segment 5: paddr=0x000fc364 vaddr=0x40085054 size=0x125a0 ( 75168) load
0x40085054: _xt_lowint1 at /mnt/f/work/esp/esp32/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1154

I (492) boot: Loaded app from partition at offset 0x10000
I (492) boot: Disabling RNG early entropy source...
I (494) spiram: Found 64MBit SPI RAM device
I (497) spiram: SPI RAM mode: flash 40m sram 40m
I (503) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (510) cpu_start: Pro cpu up.
I (514) cpu_start: Starting app cpu, entry point is 0x40081458
0x40081458: call_start_cpu1 at /mnt/f/work/esp/esp32/esp-adf/esp-idf/components/esp32/cpu_start.c:246

I (0) cpu_start: App cpu up.
I (1405) spiram: SPI SRAM memory test OK
I (1406) heap_init: Initializing. RAM available for dynamic allocation:
I (1406) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1412) heap_init: At 3FFB9300 len 00026D00 (155 KiB): DRAM
I (1418) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1425) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1431) heap_init: At 400975F4 len 00008A0C (34 KiB): IRAM
I (1437) cpu_start: Pro cpu start user code
I (1442) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (121) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (123) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (183) wifi: wifi driver task: 3ffc92f4, prio:23, stack:3584, core=0
I (183) wifi: wifi firmware version: 2af77cc
I (183) wifi: config NVS flash: enabled
I (183) wifi: config nano formating: disabled
I (193) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (203) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (223) wifi: Init dynamic tx buffer num: 32
I (223) wifi: Init data frame dynamic rx buffer num: 32
I (223) wifi: Init management frame dynamic rx buffer num: 32
I (233) wifi: Init management short buffer num: 32
I (233) wifi: Init static tx buffer num: 16
I (243) wifi: Init static rx buffer size: 1600
[... 33 lines cut ...]
I (2323) AUDIO_PLAYER: [3.1] Listening event from all elements of pipeline
I (2343) DLNA_EXAMPLE: DLNA Started...
I (90063) DLNA_EXAMPLE: SetAVTransportURI=http://192.168.0.100:61323/mp3?filepath=%2Fstorage%2Femulated%2F0%2Fnetease%2Fcloudmusic%2FCache%2FMusic1%2F494645532-96000-1824437ac55592fc31d2c11f1dc47b31.mp3.uc%21&decode=true
I (90253) DLNA_EXAMPLE: Play with speed=1
I (90253) AUDIO_ELEMENT: [http] Element task created
I (90253) AUDIO_ELEMENT: [mp3] Element task created
I (90263) AUDIO_ELEMENT: [i2s] Element task created
I (90263) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:360, MEM Total:4289640 Bytes, Inter:162548 Bytes, Dram:127244 Bytes

I (90273) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (90283) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (90283) I2S_STREAM: AUDIO_STREAM_WRITER
I (90293) AUDIO_PIPELINE: Pipeline started
I (90303) STAGEFRIGHTMP3_DECODER: MP3 opened
I (90313) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (90373) HTTP_STREAM: total_bytes=3036766
I (91503) DLNA_EXAMPLE: get vol = 84
E (91783) STAGEFRIGHTMP3_DECODER: Encountered error reading when MP3 init
E (91783) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN
I (91783) STAGEFRIGHTMP3_DECODER: Closed
W (91793) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
I (91833) AUDIO_PLAYER: Stop pipeline
I (91833) DLNA_EXAMPLE: _avt_get_trans_state STOPPED
W (91833) AUDIO_ELEMENT: [i2s] Element already stopped
W (91833) AUDIO_PIPELINE: audio_pipeline state is not running, 1

jason-mao, 2020-01-06:
@newstarqu 我们更新了DLNA的example，您能先在最新的example（d28955d62bd68e670c45ad59bf23e9374b5b64e6）上测试一下吗？

newstarqu, 2020-01-13:
我更新了罪行的example，日志如下：
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6900
load:0x40078000,len:12072
load:0x40080400,len:6756
entry 0x4008077c
I (71) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v3.3.1 2nd stage bootloader
I (39) boot: compile time 21:37:24
I (39) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (48) boot: SPI Mode       : DIO
I (52) boot: SPI Flash Size : 4MB
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00200000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 1, min. application chip revision: 0
I (100) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x4ae00 (306688) map
I (217) esp_image: segment 1: paddr=0x0005ae28 vaddr=0x3ffb0000 size=0x0330c ( 13068) load
I (223) esp_image: segment 2: paddr=0x0005e13c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /mnt/f/work/esp/esp32/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1779

I (224) esp_image: segment 3: paddr=0x0005e544 vaddr=0x40080400 size=0x01acc (  6860) load
I (236) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xd38bc (866492) map
0x400d0018: _flash_cache_start at ??:?

I (546) esp_image: segment 5: paddr=0x001338dc vaddr=0x40081ecc size=0x15d90 ( 89488) load
0x40081ecc: esp_dport_access_stall_other_cpu_start at /mnt/f/work/esp/esp32/esp-adf/esp-idf/components/esp32/dport_access.c:216

I (597) boot: Loaded app from partition at offset 0x10000
I (597) boot: Disabling RNG early entropy source...
I (598) psram: This chip is ESP32-D0WD
I (603) spiram: Found 64MBit SPI RAM device
I (607) spiram: SPI RAM mode: flash 40m sram 40m
I (612) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (619) cpu_start: Pro cpu up.
I (623) cpu_start: Application information:
I (628) cpu_start: Project name:     dlna_example
I (633) cpu_start: App version:      1
I (638) cpu_start: Compile time:     Jan 13 2020 21:37:09
I (644) cpu_start: ELF file SHA256:  6a786da3ab841216...
I (650) cpu_start: ESP-IDF:          v3.3.1
I (654) cpu_start: Starting app cpu, entry point is 0x400814ec
0x400814ec: call_start_cpu1 at /mnt/f/work/esp/esp32/esp-adf/esp-idf/components/esp32/cpu_start.c:269

I (0) cpu_start: App cpu up.
I (1545) spiram: SPI SRAM memory test OK
I (1546) heap_init: Initializing. RAM available for dynamic allocation:
I (1546) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1552) heap_init: At 3FFB9580 len 00026A80 (154 KiB): DRAM
I (1559) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1565) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1572) heap_init: At 40097C5C len 000083A4 (32 KiB): IRAM
I (1578) cpu_start: Pro cpu start user code
I (1583) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (38) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (39) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (89) wifi: wifi driver task: 3ffc9678, prio:23, stack:3584, core=0
I (89) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (89) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (119) wifi: wifi firmware version: ac331d7
[... 169 lines cut ...]
E (108089) task_wdt: CPU 1: IDLE1
E (113089) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (113089) task_wdt:  - IDLE0 (CPU 0)
E (113089) task_wdt: Tasks currently running:
E (113089) task_wdt: CPU 0: el-DEC_auto
E (113089) task_wdt: CPU 1: IDLE1
I (115989) DLNA_EXAMPLE: Pause instance=0
I (115989) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:3, Available:5, que:0x3ffd65e0
I (115989) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PAUSE
I (116009) RSP_FILTER: rsp_filter_close
I (116019) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_PAUSE
I (116039) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_PAUSE
I (116039) ESP_AUDIO_CTRL: Total:3897931, get duration:243000 ms by bps:128000
I (116049) DLNA_EXAMPLE: _avt_get_duration 243000
I (116049) ESP_AUDIO_CTRL: Total:3897931, get duration:243000 ms by bps:128000
I (116059) DLNA_EXAMPLE: _avt_get_duration 243000
I (116059) DLNA_EXAMPLE: _avt_get_trans_state PAUSED_PLAYBACK
I (116209) ESP_AUDIO_CTRL: Total:3897931, get duration:243000 ms by bps:128000
I (116209) DLNA_EXAMPLE: _avt_get_duration 243000
I (116209) DLNA_EXAMPLE: _avt_get_time 00:01:33
I (116219) DLNA_EXAMPLE: _avt_get_pos 1390549
I (116529) ESP_AUDIO_CTRL: Total:3897931, get duration:243000 ms by bps:128000
I (116529) DLNA_EXAMPLE: _avt_get_duration 243000
I (116529) DLNA_EXAMPLE: _avt_get_time 00:01:33
I (116539) DLNA_EXAMPLE: _avt_get_pos 1390549

newstarqu, 2020-01-13:
我更新了最新的example和esp-adf代码，日志如下：
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6900
load:0x40078000,len:12072
load:0x40080400,len:6756
entry 0x4008077c
I (71) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v3.3.1 2nd stage bootloader
I (39) boot: compile time 21:37:24
I (39) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (48) boot: SPI Mode       : DIO
I (52) boot: SPI Flash Size : 4MB
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00200000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 1, min. application chip revision: 0
I (100) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x4ae00 (306688) map
I (217) esp_image: segment 1: paddr=0x0005ae28 vaddr=0x3ffb0000 size=0x0330c ( 13068) load
I (223) esp_image: segment 2: paddr=0x0005e13c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /mnt/f/work/esp/esp32/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1779

I (224) esp_image: segment 3: paddr=0x0005e544 vaddr=0x40080400 size=0x01acc (  6860) load
I (236) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xd38bc (866492) map
0x400d0018: _flash_cache_start at ??:?

I (546) esp_image: segment 5: paddr=0x001338dc vaddr=0x40081ecc size=0x15d90 ( 89488) load
0x40081ecc: esp_dport_access_stall_other_cpu_start at /mnt/f/work/esp/esp32/esp-adf/esp-idf/components/esp32/dport_access.c:216

I (597) boot: Loaded app from partition at offset 0x10000
I (597) boot: Disabling RNG early entropy source...
I (598) psram: This chip is ESP32-D0WD
I (603) spiram: Found 64MBit SPI RAM device
I (607) spiram: SPI RAM mode: flash 40m sram 40m
I (612) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (619) cpu_start: Pro cpu up.
I (623) cpu_start: Application information:
I (628) cpu_start: Project name:     dlna_example
I (633) cpu_start: App version:      1
I (638) cpu_start: Compile time:     Jan 13 2020 21:37:09
I (644) cpu_start: ELF file SHA256:  6a786da3ab841216...
I (650) cpu_start: ESP-IDF:          v3.3.1
I (654) cpu_start: Starting app cpu, entry point is 0x400814ec
0x400814ec: call_start_cpu1 at /mnt/f/work/esp/esp32/esp-adf/esp-idf/components/esp32/cpu_start.c:269

I (0) cpu_start: App cpu up.
I (1545) spiram: SPI SRAM memory test OK
I (1546) heap_init: Initializing. RAM available for dynamic allocation:
I (1546) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1552) heap_init: At 3FFB9580 len 00026A80 (154 KiB): DRAM
I (1559) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1565) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1572) heap_init: At 40097C5C len 000083A4 (32 KiB): IRAM
I (1578) cpu_start: Pro cpu start user code
I (1583) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (38) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (39) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (89) wifi: wifi driver task: 3ffc9678, prio:23, stack:3584, core=0
I (89) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (89) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (119) wifi: wifi firmware version: ac331d7
[... 169 lines cut ...]
E (108089) task_wdt: CPU 1: IDLE1
E (113089) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (113089) task_wdt:  - IDLE0 (CPU 0)
E (113089) task_wdt: Tasks currently running:
E (113089) task_wdt: CPU 0: el-DEC_auto
E (113089) task_wdt: CPU 1: IDLE1
I (115989) DLNA_EXAMPLE: Pause instance=0
I (115989) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:3, Available:5, que:0x3ffd65e0
I (115989) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PAUSE
I (116009) RSP_FILTER: rsp_filter_close
I (116019) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_PAUSE
I (116039) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_PAUSE
I (116039) ESP_AUDIO_CTRL: Total:3897931, get duration:243000 ms by bps:128000
I (116049) DLNA_EXAMPLE: _avt_get_duration 243000
I (116049) ESP_AUDIO_CTRL: Total:3897931, get duration:243000 ms by bps:128000
I (116059) DLNA_EXAMPLE: _avt_get_duration 243000
I (116059) DLNA_EXAMPLE: _avt_get_trans_state PAUSED_PLAYBACK
I (116209) ESP_AUDIO_CTRL: Total:3897931, get duration:243000 ms by bps:128000
I (116209) DLNA_EXAMPLE: _avt_get_duration 243000
I (116209) DLNA_EXAMPLE: _avt_get_time 00:01:33
I (116219) DLNA_EXAMPLE: _avt_get_pos 1390549
I (116529) ESP_AUDIO_CTRL: Total:3897931, get duration:243000 ms by bps:128000
I (116529) DLNA_EXAMPLE: _avt_get_duration 243000
I (116529) DLNA_EXAMPLE: _avt_get_time 00:01:33
I (116539) DLNA_EXAMPLE: _avt_get_pos 1390549

ahhfzhang, 2020-01-14:
@newstarqu  请帮忙修改或添加如下几个配置选项到sdkconfig 

CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240

再测试看是否还出现 watchdog 的问题

newstarqu, 2020-01-15:
上面的配置加了后没有出现watchdog的问题了，日志里还是会有别的错误，好像是M4A解码的问题
I (240810) ESP_AUDIO_CTRL: Total:8238455, bps:257762, duration:256047 ms
I (240810) DLNA_EXAMPLE: _avt_get_duration 256047
I (240810) DLNA_EXAMPLE: _avt_get_time 00:00:01
I (240810) DLNA_EXAMPLE: _avt_get_pos 80833
I (241050) ESP_AUDIO_CTRL: Total:8238455, bps:257546, duration:256047 ms
I (241050) DLNA_EXAMPLE: _avt_get_duration 256047
I (241050) DLNA_EXAMPLE: _avt_get_time 00:00:01
I (241060) DLNA_EXAMPLE: _avt_get_pos 88301
I (241310) ESP_AUDIO_CTRL: Total:8238455, bps:257614, duration:256047 ms
I (241310) DLNA_EXAMPLE: _avt_get_duration 256047
I (241320) DLNA_EXAMPLE: _avt_get_time 00:00:01
I (241320) DLNA_EXAMPLE: _avt_get_pos 96561
W (345640) HTTP_STREAM: No more data,errno:104, total_bytes:3444844
I (345640) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (345930) AUDIO_ELEMENT: IN-[DEC_auto] AEL_IO_DONE,-2
E (346050) AAC_DECODER: M4A decoder encountered error 340 20
E (346050) AUDIO_ELEMENT: [DEC_auto] ERROR_PROCESS, AEL_IO_FAIL
E (346050) ESP_AUDIO_TASK: Got element[DEC_auto] error:ESP_ERR_AUDIO_PROCESS, stop pipeline
W (346060) AUDIO_PIPELINE: audio_element_stop type is finished
I (346070) AUDIO_ELEMENT: OUT-[resample] AEL_IO_DONE,-2
I (346070) RSP_FILTER: rsp_filter_close
I (346080) AAC_DECODER: Closed
E (346080) AUDIO_ELEMENT: [DEC_auto] AEL_MSG_CMD_ERROR
I (346120) AUDIO_ELEMENT: IN-[OUT_iis] AEL_IO_DONE,-2
I (346160) ESP_AUDIO_TASK: Received last pos: 3444504 bytes
I (346160) ESP_AUDIO_TASK: Received last time: 106379 ms

closed 2020-07-09 by espressif-bot