espressif/esp-adf #309: Silent Microphone over Bluetooth HFP

nkizz, 2020-01-05:
Hi!
I'm currently trying to use the HFP example code here: https://github.com/espressif/esp-adf/tree/master/examples/get-started/pipeline_a2dp_sink_and_hfp. The module can sink A2DP and HFP audio and play it, but when a call is active the microphone is completely silent. This message is also constantly printing in the console whenever HFP is active:
```
W (1279166) BT_BTM: SCO xmit Q overflow, pkt dropped
W (1279166) BT_BTM: stat 10
W (1279186) BT_BTM: SCO xmit Q overflow, pkt dropped
W (1279186) BT_BTM: stat 10
W (1279206) BT_BTM: SCO xmit Q overflow, pkt dropped
W (1279206) BT_BTM: stat 10
```
A similar issue was encountered here: https://github.com/espressif/esp-idf/issues/3550, however increasing `BTM_SCO_DATA_SIZE_MAX` just increases the time between the messages, suggesting that no data is being sent and the queue is filling up regardless of the size. The message is from this file: https://github.com/espressif/esp-idf/blob/9dbb7526c5d140b22529a1cfb0b4b46a0ab288aa/components/bt/host/bluedroid/stack/btm/btm_sco.c#L447 and increasing `BTM_SCO_XMIT_QUEUE_THRS` also doesn't fix the issue. I'm using the ESP32-LyraT-Mini_V1.2 and the latest ADF version. 

Thanks!

esp-xiaodong, 2020-01-07:
@nkizz 
Please use this file for mini board, and test again.

[a2dp_sink_and_hfp_example.txt](https://github.com/espressif/esp-adf/files/4029875/a2dp_sink_and_hfp_example.txt)

nkizz, 2020-01-07:
The new code worked! I had to comment out `periph_bt_play(bt_periph);` since that function wasn't defined, but one I did that, the mic worked perfectly. The message about the queue overflow still prints to the console every 20 ms, so I commented out that print statement because it was causing the audio to cut out.

rahna803, 2020-12-28:
Hello,

I am facing the same problem but for LyraT 4.3 board. Is it possible to a working code?

pooyasa, 2021-06-02:
> Hello,
> 
> I am facing the same problem but for LyraT 4.3 board. Is it possible to a working code?

I fixed it by enabling the on board SPI RAM from Component Config > ESP32-specific > Support for external, SPI-Connected ram.

closed 2024-08-01 by jason-mao