espressif/esp-adf #314: LyraT I2c

timotoots, 2020-01-07:
Board: LyraT
Example: cli
Problem: Using Arduino wire library crashes the board.

I try to change this example to be usable with Arduino as component. It does work, but when I connect some I2C devices and use their libraries (that use wire library) and try to communicate with them, the board crashes.

Is it possible that the Wire.begin function and the audio playing could not be used simultaneously?

The planned Arduino-ADF release might help with that?

jason-mao, 2020-01-20:
@timotoots  I2C driver API does not thread safe. We replace by I2C_bus api on commit a91879f1144984ae7df8e6a6743bd08ca4c182eb. Please test again.

closed 2020-03-25 by jason-mao

rase1957, 2021-04-08:
Hello,

the issue Using "Arduino wire library using I2C crashes the board LyraT Board V4.3 " is still unsolved.
I am using:
idf: v3.3.5-14-g7c8602753
adf: v2.2-92-g59e5e02
esp32-arduino: V1.06 

I am using arduino u8g2 lib:

a) 
#define I2C_SCL 23
#define I2C_SDA 18
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE,I2C_SCL,I2C_SDA);

b)The call to u8g2.begin(); (which usues Wire) comes before call to 
audio_board_handle_t board_handle = audio_board_init();

Then:

Crash in  

audio_board_handle_t board_handle = audio_board_init();

I (4766) HTTP_MP3_WEBRADIO_PLAY: [ 1 ] Start audio codec chip
D:/esp32idf/esp-idf/components/esp_ringbuf/ringbuf.c:975 (xRingbufferReceiveUpToFromISR)- assert failed!
abort() was called at PC 0x4008afd3 on core 1
0x4008afd3: xRingbufferReceiveUpToFromISR at D:/esp32idf/esp-idf/components/esp_ringbuf/ringbuf.c:975 (discriminator 1)

ELF file SHA256: efb6c1b75581f6bc

Backtrace: 0x40089ecc:0x3ffbee10 0x4008a115:0x3ffbee30 0x4008afd3:0x3ffbee50 0x40088d9d:0x3ffbee80 0x4008384d:0x3ffbeeb0 0x4017bf93:0x3ffbc540 0x400d3de2:0x3ffbc560 0x4008d05d:0x3ffbc580 0x4008b4a6:0x3ffbc5a0
0x40089ecc: invoke_abort at D:/esp32idf/esp-idf/components/esp32/panic.c:715

0x4008a115: abort at D:/esp32idf/esp-idf/components/esp32/panic.c:715

0x4008afd3: xRingbufferReceiveUpToFromISR at D:/esp32idf/esp-idf/components/esp_ringbuf/ringbuf.c:975 (discriminator 1)

0x40088d9d: i2c_isr_handler_default at D:/esp32idf/esp-idf/components/driver/i2c.c:1134

0x4008384d: _xt_lowint1 at D:/esp32idf/esp-idf/components/freertos/xtensa_vectors.S:1154

0x4017bf93: esp_pm_impl_waiti at D:/esp32idf/esp-idf/components/esp32/pm_esp32.c:492

0x400d3de2: esp_vApplicationIdleHook at D:/esp32idf/esp-idf/components/esp32/freertos_hooks.c:108

0x4008d05d: prvIdleTask at D:/esp32idf/esp-idf/components/freertos/tasks.c:3507

0x4008b4a6: vPortTaskWrapper at D:/esp32idf/esp-idf/components/freertos/port.c:403

c) Deactivation u8g2 usage in my code no crash in audio_board_init();, rest works fine.

Do I oversee something?

Regards 

Rainer