espressif/esp-adf #316: Ineffective configuration field for `filter_resample`

FlorianDenis, 2020-01-07:
In the filter configuration structure `rsp_filter_cfg_t` there is the following field: 
```
int out_len_bytes;    /*!< The buffer length of the output stream data. This parameter must be configured in encoding mode. */
```

This seems to indicate the user has the ability to configure the length of the output of the filter.
So with the following code

```
rsp_filter_cfg_t rsp_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
rsp_cfg.mode = RESAMPLE_ENCODE_MODE;
rsp_cfg.out_len_bytes = 240;

audio_element_handle_t resampler = rsp_filter_init(&rsp_cfg);
audio_element_set_write_cb(resampler, write_cb, ctx);

```

I would expect `write_cb` to be called with a `len` parameter always equal to  `240`.

In reality, the value of this field is overwritten in  `rsp_filter_open`, specifically:

```
if (filter->resample_info->mode == RESAMPLE_DECODE_MODE) {
    resample_info->out_len_bytes = 0;
} else if (filter->resample_info->mode == RESAMPLE_ENCODE_MODE) {
    resample_info->out_len_bytes = RESAMPLING_POINT_NUM / 4;
}
```

meaning the value passed via the configuration structure has no effect on the behaviour of the resampler. 

Either the structure should be updated to reflect this behaviour is not customisable, or the passed configuration should be properly handled.

closed 2020-01-13 by espressif-bot

DeltaEvo, 2020-01-13:
It seems that the commit closing this issue close https://github.com/espressif/esp-adf/issues/317 instead

houhaiyan, 2020-01-15:
@DeltaEvo Thank you for the ISSUE,  The out_len_bytes could configure in  `RESAMPLE_ENCODE_MODE`. we will fix it.