espressif/esp-adf #317: Confusion between `rsp_filter_cfg_t.type` and `rsp_filter_cfg_t.mode`

FlorianDenis, 2020-01-07:
I am confused as to the usage of the following fields in `rsp_filter_cfg_t`: 

```
int mode;             /*!< The resampling mode (the encoding mode or the decoding mode). For decoding mode, input PCM length is constant; for encoding mode, output PCM length is constant. */
int type;             /*!< The resampling type (Automatic, Upsampling and Downsampling) */
``` 

Since no types besides `int` are specified in the struct definition, one might read the documentation and conclude that:
-  `mode` should be a `esp_resample_mode_t` *i.e.* either `RESAMPLE_DECODE_MODE` or `RESAMPLE_ENCODE_MODE`
- `type` should be a `esp_resample_type_t` *i.e.* one of: `ESP_RESAMPLE_TYPE_AUTO`, `ESP_RESAMPLE_TYPE_DECIMATE`, `ESP_RESAMPLE_TYPE_INTERP`, `ESP_RESAMPLE_TYPE_RESAMPLE`, `ESP_RESAMPLE_TYPE_BYPASS`

Which makes sense when looking at the code. 

That being said, the `rsp_filter_init`documentation states the following: 
```
*   - If the audio_codec_type_t is `AUDIO_CODEC_TYPE_DECODER`, `src_rate` and `src_ch` will be fetched from `audio_element_getinfo`.
 *   - If the audio_codec_type_t is `AUDIO_CODEC_TYPE_ENCODER`, `src_rate`, `src_ch`, `dest_rate` and `dest_ch` must be configured.
 ```
which seems to directly contradict the above.

In addition,  a few of your examples use the following code: 
```
rsp_filter_cfg_t rsp_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
rsp_cfg.src_rate = 48000;
rsp_cfg.src_ch = 2;
rsp_cfg.dest_rate = 16000;
rsp_cfg.dest_ch = 1;
rsp_cfg.type = AUDIO_CODEC_TYPE_ENCODER;
audio_element_handle_t filter = rsp_filter_init(&rsp_cfg);
```
See for instance the `esp_dispatcher_dueros` or `voip`  examples.

By looking strictly at the documentation, it seems the mode of the resulting `filter` should be the default mode, ie `DECODE` instead of what the code seems to suggest ? 

Can you please clarify the relationship between those 2, and the meaning of the `rsp_filter_init` documentation ?

houhaiyan, 2020-06-19:
pleae update adf project. the documents has been updated.

FlorianDenis, 2020-06-19:
As @DeltaEvo mentioned there has been a confusion - issue #316 was mistakenly closed instead of this one

Agreed, the documentation has been corrected in January after this issue was submitted. 
This issue can be marked as resolved.

houhaiyan, 2020-06-19:
@FlorianDenis  About   #316 , do you have any confusion?

FlorianDenis, 2020-06-19:
I just looked at the code again and I think the problem that was initially reported in #316 has been fixed now; so it all good 👍

closed 2020-10-15 by jason-mao