espressif/esp-adf #318: Support i2s_write_expand

pad52, 2020-01-07:
Please add a configuration to the I2S driver for using the i2s_write_expand function instead of the i2s_write

jason-mao, 2020-01-08:
@pad52 Hi，there is already support override the I2S_stream write function. Please refer to `
```c
int my_i2s_write(audio_element_handle_t self, char *buffer, int len, TickType_t ticks_to_wait, void *context)
{
    i2s_stream_t *i2s = (i2s_stream_t *)audio_element_getdata(self);
    audio_element_info_t info;
    size_t bytes_written = 0;
    audio_element_getinfo(self, &info);
    if (info.channels == 1) {
        i2s_mono_fix(info.bits, (uint8_t *)buffer, len);
    }
    if ((i2s->config.i2s_config.mode & I2S_MODE_DAC_BUILT_IN) != 0) {
        i2s_dac_data_scale(info.bits, (uint8_t *)buffer, len);
    }
    i2s_write_expand(xx);
    audio_element_getinfo(self, &info);
    info.byte_pos += bytes_written;
    audio_element_setinfo(self, &info);
    return bytes_written;
}

////////////////////////////////////////////////////////////////////////////////////////

audio_element_handle_t my_i2s = i2s_stream_init(default_cfg);
audio_element_set_write_cb(my_i2s, my_i2s_write, NULL);

```

pad52, 2020-01-13:
Thanks @jason-mao 
do I need to change also the _i2s_process function and in general everywhere I found a call to i2s_write()?

jason-mao, 2020-01-19:
I  think no need that.  But there is not good code to clear DMA buffer use `i2s_write`, should be call `audio_element_output`.

pad52, 2020-01-20:
Dear @jason-mao, I just pulled the last esp-adf version and the code you suggested worked flawlessy.
I just had to remove the "static" declaration from i2s_mono_fix, i2s_dac_data_scale and i2s_stream_clear_dma_buffer functions and add the prototypes in the i2s_stream.h
In case you would implement this modification I can finally work without overriding the esp-adf components. 
We can close the issue.

Maybe it's OT but Can I ask you what difference are there between the pipeline_a2dp_sink_stream example and the pipeline_bt_sink example?

jason-mao, 2020-01-21:
Hi Pad52,
> Maybe it's OT but Can I ask you what difference are there between the pipeline_a2dp_sink_stream example and the pipeline_bt_sink example?

The difference between the two example is the way they are implemented. `pipeline_a2dp_sink_stream` use `a2dp_stream`, `pipeline_bt_sink` use `bluetooth_service`.

closed 2020-02-20 by jason-mao