espressif/esp-adf #320: SIP source files

szucsi23, 2020-01-13:
Hi!
I'm working with your SIP libraries, and it would be really helpful, if I could see and add my own additions to the source files.
Is there any way, to make them open source, or some other way to make it accessible for the users?
It would be really great!
Thanks in advance!

iscapla, 2020-04-14:
Hello @szucsi23, I'd like to take a look on it too because I think that there is a memory leak problem on the sync periodically messages . Have you got any news?

szucsi23, 2020-04-15:
Sadly no, still waiting.

jason-mao, 2020-06-19:
Hi guys, I'm sorry that we cannot provide source code.  We will continue to improve the sip design. If you are have any issues, please let me know.

closed 2020-06-19 by jason-mao