espressif/esp-adf #322: pipeline_sdcard_mp3_control 播放失败

lxdemon, 2020-01-17:
log如下：
I (38) SDCARD_MP3_CONTROL_EXAMPLE: [1.0] Initialize peripherals management
I (38) SDCARD_MP3_CONTROL_EXAMPLE: [1.1] Initialize and start peripherals
W (68) PERIPH_TOUCH: _touch_init
I (548) SDCARD_MP3_CONTROL_EXAMPLE: [1.2] Set up a sdcard playlist and scan sdcard music save to it
E (548) PLAYLIST_SDCARD: open file error, line: 112, have you mounted sdcard, set the long file name and UTF-8 encoding configuration ?
E (558) PLAYLIST_SDCARD: /home/lixin/workspace/esp-adf/components/playlist/playlist_operator/sdcard_list.c:197 (sdcard_list_show): Got NULL Pointer
I (568) SDCARD_MP3_CONTROL_EXAMPLE: [ 2 ] Start codec chip
E (578) gpio: gpio_install_isr_service(410): GPIO isr service already installed
I (618) SDCARD_MP3_CONTROL_EXAMPLE: [ 3 ] Create and start input key service
I (618) SDCARD_MP3_CONTROL_EXAMPLE: [4.0] Create audio pipeline for playback
I (638) SDCARD_MP3_CONTROL_EXAMPLE: [4.1] Create i2s stream to write data to codec chip
I (638) SDCARD_MP3_CONTROL_EXAMPLE: [4.2] Create mp3 decoder to decode mp3 file
I (638) SDCARD_MP3_CONTROL_EXAMPLE: [4.3] Create resample filter
I (648) SDCARD_MP3_CONTROL_EXAMPLE: [4.4] Create fatfs stream to read data from sdcard
E (658) PLAYLIST_SDCARD: /home/lixin/workspace/esp-adf/components/playlist/playlist_operator/sdcard_list.c:281 (sdcard_list_current): Got NULL Pointer
I (668) SDCARD_MP3_CONTROL_EXAMPLE: [4.5] Register all elements to audio pipeline
I (688) SDCARD_MP3_CONTROL_EXAMPLE: [4.6] Link it together [sdcard]-->fatfs_stream-->mp3_decoder-->resample-->i2s_stream-->[codec_chip]
I (688) SDCARD_MP3_CONTROL_EXAMPLE: [5.0] Set up  event listener
I (698) SDCARD_MP3_CONTROL_EXAMPLE: [5.1] Listen for all pipeline events
W (708) SDCARD_MP3_CONTROL_EXAMPLE: [ 6 ] Press the keys to control music player:
W (718) SDCARD_MP3_CONTROL_EXAMPLE:       [Play] to start, pause and resume, [Set] next song.
W (738) SDCARD_MP3_CONTROL_EXAMPLE:       [Vol-] or [Vol+] to adjust volume.

SD卡中并未生成__playlist, 难道是SD卡的问题，但是使用读卡器可正常读写

lxdemon, 2020-01-17:
补充硬件信息：ESP32-LyraT v4.3

jason-mao, 2020-01-19:
@lxdemon 

> E (548) PLAYLIST_SDCARD: open file error, line: 112, have you mounted sdcard, set the long file name and UTF-8 encoding configuration ?

这个信息提示你的menuconfig 配置不对.请参考下图：
[image]

lxdemon, 2020-01-20:
按你给的配置现在可以了，谢谢！

closed 2020-01-20 by jason-mao