espressif/esp-adf #324: hfp_stream.c addition to bluetooth causing build to fail

domnoble, 2020-01-19:
looking at the repository i can see that the hfp_stream.c was added to the bluetooth CMakeList 11 days ago, somehow, when building firmware using idf.py i get this error:

`/esp-adf/components/bluetooth_service/hfp_stream.c:32:24: fatal error: raw_stream.h: No such file or directory`

removing hfp_stream gets a successful compilation. im not sure if its related, i dont think so but im using a slightly modified version of esp-adf where ive swapped out the audio_board and audio_hal stuff so it supports the AC101 / Ai Thinkers ESP32-A1S

eagle-lk, 2020-01-20:
I look into the CMakeLists of component bluetooth_service, and find line 4 set the requirement of the component. Because hfp_stream.c use the header file raw_stream.h,  which is found in the include dir of component autio_stream,  I think hfp_stream.c require component audio_stream. so I modified line 4 to add audio_stream in the requirement list. and build OK.

domnoble, 2020-01-22:
compiling with gcc make works fine, i think this was just user error on my part once i got everything in the right place and on my path, and tried gcc make it compiled successfully

closed 2020-01-22 by domnoble