espressif/esp-adf #325: esp_audio_play with starting position by bytes > 0 fails decode audio

cbontoz, 2020-01-20:
Hi, 

I have set-up the esp_audio library based on the example code, and playing AAC audio from SD card works fine. When I try to start the audio from a specific start position in bytes using `esp_audio_play` function the decoder fails as in the log below:

```
I (38337) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffd0c10
I (38347) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:486, MEM Total:4035552 Bytes, Inter:64172 Bytes, Dram:48404 Bytes
I (38347) ESP_AUDIO_TASK: It's a decoder
I (38357) ESP_AUDIO_TASK: 1.CUR IN:[IN_file],CODEC:[DEC_auto],RESAMPLE:[0],OUT:[OUT_iis]
I (38367) ESP_AUDIO_TASK: 2.Handles,IN:0x3f8080cc,CODEC:0x3f8086d0,FILTER:0x0,OUT:0x3f808804
W (38377) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (38377) AUDIO_PIPELINE: audio_pipeline state is not running, 1
I (38387) AUDIO_PIPELINE: audio_pipeline_unlinked
W (38387) AUDIO_PIPELINE: There are no listener registered
W (38397) ESP_AUDIO_TASK: Destroy the old pipeline
I (38397) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (38407) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (38417) AUDIO_PIPELINE: link el->rb, el:0x3f8080cc, tag:IN_file, rb:0x3f82dc60
I (38417) AUDIO_PIPELINE: link el->rb, el:0x3f8086d0, tag:DEC_auto, rb:0x3f82dc88
I (38427) ESP_AUDIO_TASK: 3. Previous starting...
I (38437) AUDIO_ELEMENT: [IN_file] Element task created
I (38437) AUDIO_ELEMENT: [DEC_auto] Element task created
I (38447) AUDIO_ELEMENT: [DEC_auto] AEL_MSG_CMD_RESUME,state:1
I (38447) ESP_DECODER: Ready to do audio type check (line 121)
**E (38457) AUDIO_ELEMENT: [DEC_auto] AEL_STATUS_ERROR_OPEN**
I (38457) AUDIO_ELEMENT: [OUT_iis] Element task created
I (38467) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (38467) I2S_STREAM: AUDIO_STREAM_WRITER
W (38477) AUDIO_ELEMENT: IN-[OUT_iis] AEL_IO_ABORT
W (38477) AUDIO_PIPELINE: audio_pipeline state is not running, 1
E (38487) ESP_AUDIO_TASK: media_ctrl_task error:ESP_ERR_AUDIO_OPEN
I (38487) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffd0c10
I (38497) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:5
I (38507) AUDIO_ELEMENT: [IN_file] AEL_MSG_CMD_RESUME,state:1
I (38507) PLAY_AUDIO: _play(), i:0x3f8080cc, c:0x3f8086d0, f:0x0, o:0x3f808804,status:5
I (38517) PLAY_AUDIO: bytes: 204800
W (38517) ESP_AUDIO_CTRL: Already stoped, status[5]
```

Some code snippets. Player setup code:
```
static void _setup_player(void)
{
    if (player) {
        return;
    }
    esp_audio_cfg_t cfg = DEFAULT_ESP_AUDIO_CONFIG();
    board_handle = audio_board_init ();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_DECODE, AUDIO_HAL_CTRL_START);
    cfg.vol_handle = board_handle->audio_hal;
    cfg.vol_set = (audio_volume_set)audio_hal_set_volume;
    cfg.vol_get = (audio_volume_get)audio_hal_get_volume;
    cfg.prefer_type = ESP_AUDIO_PREFER_MEM;
    cfg.cb_func = NULL;
    cfg.cb_ctx = NULL;
    player = esp_audio_create(&cfg);

    // Create readers and add to esp_audio
    http_stream_cfg_t http_cfg = HTTP_STREAM_CFG_DEFAULT();
    http_cfg.type = AUDIO_STREAM_READER;
    http_cfg.task_core = 0;
    http_cfg.out_rb_size = 20 * 1024;
    http_cfg.event_handle = _http_stream_event_handle;
    http_cfg.type = AUDIO_STREAM_READER;
    http_cfg.enable_playlist_parser = true;

    fatfs_stream_cfg_t fatfs_cfg = FATFS_STREAM_CFG_DEFAULT();
    fatfs_cfg.type = AUDIO_STREAM_READER;

    esp_audio_input_stream_add(player, fatfs_stream_init(&fatfs_cfg));
    esp_audio_input_stream_add(player, http_stream_init(&http_cfg));

    // Add decoders and encoders to esp_audio
// #ifdef ESP_AUDIO_AUTO_PLAY
    audio_decoder_t auto_decode[] = {
        // DEFAULT_ESP_AMRNB_DECODER_CONFIG(),
        // DEFAULT_ESP_AMRWB_DECODER_CONFIG(),
        // DEFAULT_ESP_FLAC_DECODER_CONFIG(),
        // DEFAULT_ESP_OGG_DECODER_CONFIG(),
        // DEFAULT_ESP_OPUS_DECODER_CONFIG(),
        DEFAULT_ESP_MP3_DECODER_CONFIG(),
        DEFAULT_ESP_AAC_DECODER_CONFIG(),
        DEFAULT_ESP_M4A_DECODER_CONFIG(),
        // DEFAULT_ESP_TS_DECODER_CONFIG(),
    };
       esp_decoder_cfg_t auto_dec_cfg = DEFAULT_ESP_DECODER_CONFIG();
       auto_dec_cfg.task_core = 0;
       auto_dec_cfg.task_prio = 25;
       auto_dec_cfg.out_rb_size = 20 * 1024;
       esp_audio_codec_lib_add(player, AUDIO_CODEC_TYPE_DECODER,   esp_decoder_init(&auto_dec_cfg, auto_decode, sizeof(auto_decode) / sizeof(audio_decoder_t)));

        i2s_stream_cfg_t i2s_writer = I2S_STREAM_CFG_DEFAULT();
        i2s_writer.type = AUDIO_STREAM_WRITER;
	i2s_writer.task_core = 0;
	i2s_writer.task_prio = 25;
	// i2s_writer.task_stack = 20 * 1024;
	i2s_writer.i2s_config.dma_buf_len = 1024;
	i2s_writer.i2s_config.dma_buf_count = 4;
	i2s_writer.i2s_config.mode = I2S_MODE_MASTER | I2S_MODE_TX;
	i2s_writer.i2s_config.channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT;
	i2s_writer.out_rb_size = 50 * 1024;

	i2s_stream_writer = i2s_stream_init(&i2s_writer);
        esp_audio_output_stream_add(player, i2s_stream_writer);

        // Set default volume
        esp_audio_vol_set(player, 45);
        enable_amps(true);
        AUDIO_MEM_SHOW(TAG);
        ESP_LOGI(TAG, "esp_audio instance is:%p\r\n", player);
}
```

```
esp_err_t _play(int volume, enum audio_track_id track_id, bool mng_amps, char *uri) {
	esp_err_t ret = ESP_FAIL;

	if (uri != NULL){
		ESP_LOGE(TAG, "playing %s", uri);
                // Random position (2 * 1024 * 2000) confirmed inside the track size.
		esp_audio_play(player, AUDIO_CODEC_TYPE_DECODER, uri, 2 * 1024 * 2000);
	}

	esp_audio_info_t                backup_info;

	esp_audio_info_get(player, &backup_info);
	ESP_LOGI(TAG, "%s, i:%p, c:%p, f:%p, o:%p,status:%d", __func__,
				backup_info.in_el,
				backup_info.codec_el,
				backup_info.filter_el,
				backup_info.out_el, 
				backup_info.st.status);
	return ESP_OK;
}
```

Can you spot what I might be missing? I didn't find an example to demonstrate how we can start audio track with position offset. (esp-idf v3.2.2)

Thank you,
Christos

jason-mao, 2020-01-21:
@cbontoz Hi Christos, please disable `ESP_AUDIO_AUTO_PLAY ` for use manual decoder.

cbontoz, 2020-01-23:
Thank you @jason-mao 

I will confirm the fix soon. Unfortunately, esp_audio doesn't support EQ so we cannot use it in our project. We will need to control the start position using audio pipelines and elements directly.

closed 2020-01-25 by cbontoz

michaelwittig, 2020-05-31:
@cbontoz can you share some details on how you implemented starting a file from a specific position with audio pipelines?

oliverjc, 2020-08-06:
@michaelwittig I'm also interested in this. Did you find a way?

michaelwittig, 2020-08-07:
I do:
```
// prepare pipeline
audio_element_set_uri(fatfs_stream_reader, "/sdcard/sound.mp3");
audio_pipeline_reset_ringbuffer(pipeline);
audio_pipeline_reset_elements(pipeline);

// offset
audio_element_info_t info = {0};
audio_element_getinfo(fatfs_stream_reader, &info);
info.byte_pos = position;
audio_element_setinfo(fatfs_stream_reader, &info);

// start mp3
audio_pipeline_run(pipeline);
```

oliverjc, 2020-08-10:
Thanks for sharing @michaelwittig, I came to a similar solution. My goal is to achieve looping of wav files, whereby I can rewind the positions of the file reader and it outputs a continuous stream of data to the pipeline. 
I've managed to rewind the file reader without issue, but I'm getting `WAV_DECODER: Error parse wav header` errors from the wav decoder. Is this something you've experienced?

michaelwittig, 2020-08-10:
Hi @oliverjc . No, I have not experienced similar issues (I play mp3s... not sure if that makes a difference)