espressif/esp-adf #326: i2c not in peripherals list

stasartcode, 2020-01-20:
Can You add i2c to esp peripheral /like other peripherals/ to easy esp_periph_start() ? Now is only driver (?) ... \esp_peripherals\driver\i2c_bus\
And where are: 
periph_FLASH.c (ID +5)
periph_BLUETOOTH.c (ID +9)
periph_ADC.c (ID +7)
???

\esp_peripherals\include\esp_peripherals.h

`
/**
 * @brief Peripheral Identify, this must be unique for each peripheral added to the peripherals list
 */
typedef enum {
    PERIPH_ID_BUTTON     = AUDIO_ELEMENT_TYPE_PERIPH + 1,
    PERIPH_ID_TOUCH      = AUDIO_ELEMENT_TYPE_PERIPH + 2,
    PERIPH_ID_SDCARD     = AUDIO_ELEMENT_TYPE_PERIPH + 3,
    PERIPH_ID_WIFI       = AUDIO_ELEMENT_TYPE_PERIPH + 4,
    PERIPH_ID_FLASH      = AUDIO_ELEMENT_TYPE_PERIPH + 5,
    PERIPH_ID_AUXIN      = AUDIO_ELEMENT_TYPE_PERIPH + 6,
    PERIPH_ID_ADC        = AUDIO_ELEMENT_TYPE_PERIPH + 7,
    PERIPH_ID_CONSOLE    = AUDIO_ELEMENT_TYPE_PERIPH + 8,
    PERIPH_ID_BLUETOOTH  = AUDIO_ELEMENT_TYPE_PERIPH + 9,
    PERIPH_ID_LED        = AUDIO_ELEMENT_TYPE_PERIPH + 10,
    PERIPH_ID_SPIFFS     = AUDIO_ELEMENT_TYPE_PERIPH + 11,
    PERIPH_ID_ADC_BTN    = AUDIO_ELEMENT_TYPE_PERIPH + 12,
    PERIPH_ID_IS31FL3216 = AUDIO_ELEMENT_TYPE_PERIPH + 13,
} esp_periph_id_t;
`

jason-mao, 2020-02-25:
@stasartcode Sorry, I think I have not got your exact means.Could you give more explanation?

stasartcode, 2020-02-26:
Look for peripherals list in file : \esp_peripherals\include\esp_peripherals.h

part of code:
@brief Peripheral Identify, this must be unique for each peripheral added to the peripherals list
*/
typedef enum {
PERIPH_ID_BUTTON = AUDIO_ELEMENT_TYPE_PERIPH + 1,
PERIPH_ID_TOUCH = AUDIO_ELEMENT_TYPE_PERIPH + 2,
PERIPH_ID_SDCARD = AUDIO_ELEMENT_TYPE_PERIPH + 3,
PERIPH_ID_WIFI = AUDIO_ELEMENT_TYPE_PERIPH + 4,
**PERIPH_ID_FLASH = AUDIO_ELEMENT_TYPE_PERIPH + 5,**
PERIPH_ID_AUXIN = AUDIO_ELEMENT_TYPE_PERIPH + 6,
**PERIPH_ID_ADC = AUDIO_ELEMENT_TYPE_PERIPH + 7,**
PERIPH_ID_CONSOLE = AUDIO_ELEMENT_TYPE_PERIPH + 8,
**PERIPH_ID_BLUETOOTH = AUDIO_ELEMENT_TYPE_PERIPH + 9,**
PERIPH_ID_LED = AUDIO_ELEMENT_TYPE_PERIPH + 10,
PERIPH_ID_SPIFFS = AUDIO_ELEMENT_TYPE_PERIPH + 11,
PERIPH_ID_ADC_BTN = AUDIO_ELEMENT_TYPE_PERIPH + 12,
PERIPH_ID_IS31FL3216 = AUDIO_ELEMENT_TYPE_PERIPH + 13,
} esp_periph_id_t;

compare this list with list of files in \esp\esp-adf\components\esp_peripherals\include 

You can find for for example periph_button.h
but not :
**periph_FLASH.h
periph_BLUETOOTH.h
periph_ADC.h**

My idea:
i2c should be added like next peripheral for easy pipeline (example: esp_periph_start() funcion) ..

jason-mao, 2022-02-08:
@stasartcode  Thank you for your suggestion, we will discuss this.

closed 2022-02-08 by jason-mao