espressif/esp-adf #327: Add custom board support without modifying ADF tree

qberthet, 2020-01-21:
Hi,

I am designing a custom esp32 board that I plan to use with esp-adf. I have understood that I can add a new board in "esp-adf/components/audio_board", but I would like to keep the esp-adf tree clean of modifications and keep the specific board files in my project tree.

Is there a "clean" way to do this?

I have tried to replicate the "audio_board" folder in the components folder of my project, but the "Audio HAL" menuconfig entry appear twice.. 

This isn't an issue in the code but as I can't find an easy way to do it an example in the documentation would be nice.

Using the current stable version 

Thanks in advance!

jason-mao, 2020-02-12:
@qberthet, I will provide a custom method to add new board in yourself project.

jason-mao, 2020-02-18:
@qberthet Please checked on f38a7d240c40ad5a6cba4c91dce8308373b170f1. And refer to `https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3/components/my_board`

xiruilin, 2020-02-19:
@jason-mao 
Thank you for your update, it's also what we need.

It turns out that many "Cmakelists.txt" need to include "audio_board". Can you add "my_board" to cmake(or convention a name)? For example:
```
# Edit following two lines to set component requirements (see docs)
set(COMPONENT_REQUIRES audio_hal audio_board my_board audio_pipeline)
set(COMPONENT_PRIV_REQUIRES )
```
So we don't need to modify the files in ADF.

jason-mao, 2020-02-19:
@xiruilin Yes. Thank you mention this.

jason-mao, 2020-02-21:
@xiruilin Cmake issue fixed on 2a95e1f2c603d07f98735518295243cbdb2958ed.

closed 2020-02-21 by jason-mao

ihouses, 2020-12-21:
Hi, I have installed the last version of IDF and ADF and I want to use the custom board for the example sdcard_mp3

Now I am confused since there are two ways for adding a custom board, by selecting My_audio_board component or selecting  Audio Hal with Custom audio board. What is the correct way?

[image]