espressif/esp-adf #328: Voip example crackle on LyraT V4.3

bkbme, 2020-01-21:
When running the voip example on a LyraT V4.3 I get crackles in the downstream direction (from the SIP server towards the ESP32).
Things I checked:
1. Board/headphone seems working fine, since play_mp3 example has no issues
2. RTP datastream is fine, since I recorded it with wireshark, extracted the G711a data from the packet payload and played it with sox/play. No crackle there.
3. Since the G711 encode/decode functions are not available as source code, I compiled in a alternate implementation for G711 decode. Same issue.

I am  really out of ideas what I could be looking at next. Any pointers?

ahhfzhang, 2020-01-22:
hi @bkbme  
You can try to set g711dec pipeline resample filter complexity to 0,
`rsp_cfg.complexity = 0;`
hope it can help you.

bkbme, 2020-01-22:
Hi, 
I can confirm, that helps indeed! Awesome!

Can you explain why that removes the crackling?
As per documentation, the complexity parameter I just explained in terms of "speed" (i.e. computational load I assume) and "accuracy".
I have no idea what a low accuracy for a resamping filter would mean.

My assumption: Has it to do with the computational load that the resampling filter causes? So maybe the system was not "fast" enough to decode the datastream?
But in this case I should have heared some corruption in the upstream direction (from the ESP32 to the sip server) which is fine.

vikramdattu, 2020-01-22:
@bkbme The example uses 160MHz CPU frequency by default. Could you please try changing it to 240MHz. Also, please change SPIRAM frequency to 80MHz and check if example works fine as it is!

Changing mode to release mode (Compiler Options > -Os) would also help to get some considerable optimization.

bkbme, 2020-01-24:
Hi guys,

I tried all of the above,
- 240Mhz CPU frequency
- Release mode -Os
- SPI and Flash Frequency = 80MHz

But the only thing, that removes the crackling is the filter complexity (setting it to 0).

closed 2020-04-09 by espressif-bot